rust-embedded/wg #2: Support more family of microcontrollers

japaric, 2016-09-29:
The "Arduino" sub-ecosystem is probably the biggest as it contains several devices from different
architectures. Right now, Rust pretty much only supports, albeit not officially, microcontrollers
based on the ARM Cortex-M processor. This family is big but there are a lot of other widely used
architectures, to name a few:

- AVR
- PIC
- MSP430
- Xtensa (see [ESP8266])

[ESP8266]: http://www.esp8266.com/

To ease adoption, eventually, most of these architectures should be officially supported by the
compiler. By "oficially", I mean that not only the Rust compiler should be able to generate programs
for the target but there should also be some minimal testing of the target in the rust-lang/rust
repo.

Here's the current state of Rust support for all the architectures listed above (as of 2016-09-28):

(Note that, because `rustc` is LLVM based, adding support for any of these architectures depends on
LLVM also supporting the architecture.)

- **ARM Cortex-M**. Fully supported but not built into `rustc`. Can be used as a "custom" target by
  creating a JSON file that describes (specifies) the target.

- **AVR**. The [avr-llvm] fork supports this architecture. This fork is in the process of being
  [merged back] into upstream LLVM.

- **PIC**. Some (all?) 32-bit PIC micros are based on the MIPS32 M4K architecture so it may be
  possible to leverage existing LLVM support for the MIPS architecture to create targets for these.
  I think LLVM doesn't support the architecture of 8-bit and 16-bit variants though.

- **MSP430**. There is a MSP430 backend in official LLVM. Success building a Rust program for MSP430
  was reported back in [2014] but I haven't seen anyone pushing for or requesting MSP430 support
  recently.

- **Xtensa**. No LLVM support AFAIK.

[avr-llvm]: https://github.com/avr-llvm/llvm
[merged back]: https://reviews.llvm.org/search/query/Pb8J1sLVKkbr/#R
[2014]: https://github.com/rust-lang/rust/issues/16831#issuecomment-60413097

alexchandel, 2017-02-11:
I've successfully compiled Rust programs to IR, then compiled that to MIPS32 assembly and assembled it with LLVM, and linked it with the PIC toolchain's linker (although I imagine LLD would do it too). I've done something similar with MSP430 too.

So it's possible, someone just has to create bare-bones target files for rustc for those architectures.

thejpster, 2017-02-11:
How different were the LLVM versions in Rust and in the MIPS32 version you
used to compile the IR? I recall AVR support for example was in one fork
and Rust used a different fork, so someone had the fun job of trying to
merge it all together.

pftbest, 2017-02-11:
I don't get the message. MIPS32r2 is officially supported "Tier 2" target, of course it will work :)

whitequark, 2017-02-11:
> Xtensa. No LLVM support AFAIK.

The Espressif people are actively working on Xtensa support, from what I know.

whitequark, 2017-05-24:
Honestly I can't imagine many Rust people putting up with the kind of code you have to write to get a GCC frontend... An Xtensa backend isn't that scary either, it's only a few months of work.

rafaeldelucena, 2017-05-24:
Regardless of imagination, what the reason to not doing it?

The reason to do is enabling Rust on all available architectures which GCC is present (Taking advantage of all the expertise that exists, all the tests and validations that have been done in real world, and not try to do something new and have to deal with this kinda stuff again)

With a GCC frontend

redbrain, 2017-05-24:
I reckon it would be worth the effort i want to restart the project some day.

whitequark, 2017-05-24:
> Regardless of imagination, what the reason to not doing it?

To rephrase my earlier comment: the reason is that now you have to work with gcc. If my choice would have been working on gcc or not writing a compiler, I would not have written a compiler. There is a reason no new language (that I'm aware of) uses gcc as its middle/backend. gcc often has many more architecture-specific microoptimizations than LLVM for more obscure architectures, but they are, aptly said, well hidden.

rafaeldelucena, 2017-05-24:
> There is a reason no new language (that I'm aware of) uses gcc as its middle/backend.

This is okay, mainly because new languages as a first step, they intend to initially run on more common architectures.

But when you want to have an infrastructure for embedded systems, I think you have to think about a greater plurality of vendors and chips, and at this point GCC is way ahead of LLVM.

whitequark, 2017-05-24:
> I think you have to think about a greater plurality of vendors and chips, and at this point GCC is way ahead of LLVM.

Or we could write more backends for LLVM, and benefit every LLVM-based language, of which there are far more. The argument goes both ways, and IME (I maintain an LLVM backend) this way is easier.

rafaeldelucena, 2017-05-24:
> IME (I maintain an LLVM backend) this way is easier.

As you mentioned, clearly this is the easiest way, since you are already an LLVM developer.

In my view, everything that involves hardware and the physical world takes a lot more time to validate and test, It's not just code. This time has already been spent by the GCC developers and their users (embedded software developers) and of course, the product consumers.

> Or we could write more backends for LLVM, and benefit every LLVM-based language

One thing does not make the other unfeasible, which I questioned from the beginning was the reason for not wanting to do this.

Anyway thanks for your answers

jamesmunns, 2019-02-03:
I believe that the topics of this issue are now covered by other efforts, such as issue #3 and #20. Marking this for a cleanup sweep. If we would like this to stay open, please provide an update to what this issue should be focused on.

caemor, 2019-02-05:
@whitequark do you know of any news for an open source llvm backend for xtensa besides the paid one for cadence/tensilica customers?

jamesmunns, 2019-02-14:
@caemor you might want to take a look at https://twitter.com/EspressifSystem/status/1095849676572520453.

Please feel free to open a new issue to discuss any specific topics.

closed 2019-02-14 by jamesmunns

n8henrie, 2019-05-26:
@caemor you may have already seen https://github.com/espressif/llvm-xtensa but if not it looks promising.

caemor, 2019-05-26:
yes already saw. it's looking good although it will still take some time until its integrated/merged into llvm