rust-embedded/wg #4: Support more OpenWRT devices

japaric, 2016-09-29:
Currently, the Rust compiler supports MIPS OpenWRT devices via these targets:

- `mips(el)-unknown-linux-uclibc`. OpenWRT 15.05. There is no binary release of `std` for this
  target though. Having binary releases for this target is at least blocked by rust-lang/libc#361
- `mips(el)-unknown-linux-musl`. OpenWRT Trunk. As a Tier 3 (?) target.

OpenWRT not only supports MIPS devices but also ARM devices and other architectures. We should
investigate what's required to support those other targets.

For OpenWRT 15.05 ARM devices, we'll probably need an `arm-unknown-linux-uclibc` target.

OpenWRT Trunk for ARM seems more tricky. In theory, the `arm-unknown-linux-musleabi` target is the
right target but that produces statically linked binaries but users of these devices will probably
want dynamically linked binaries to keep binary sizes small. Hence something like
rust-lang/rfcs#1721 to support these devices.

jamesmunns, 2019-02-03:
We have not made any particular progress on this issue in 2 or so years, so I would like to propose that we close this issue. Marking this for a cleanup sweep. If we would like this to stay open, please provide an update to what this issue should be focused on.

jamesmunns, 2019-02-04:
CC @aep, who might have some insight of what is and isn't possible/reasonable in this space. I'm interested in keeping this issue open if we think it is still valuable, or if there is progress to be made here!

Also CC @cr1901, who commented about this on IRC.

cr1901, 2019-02-04:
@jamesmunns To clarify, I made a passing comment on IRC.

I'm interested in this topic to the extent that it's another target to play w/, and there does still seem to be binary size issues (some OpenWRT distros fit on a 4MB flash!) if you're using anything beyond `libcore`.

But I don't have any OpenWRT hardware and I don't have the bandwidth to contribute; I simply appreciate someone trying to use Rust w/ MIPS :).

aep, 2019-02-04:
uclibc is pretty dead. I'd not recommend adding any targets with it. openwrt will use musl where possible.

arm-unknown-linux-musleabi is broken and will stay broken. rustup comes with  a libc, including a crt, but they're shipping the wrong crt for the link mode so it just crashes. 

https://github.com/rust-lang/rust/issues/54294

mips(el)-unknown-linux-musl  works fine with the usual caveats that most of libstd is never tested on mips/el so for example backtrace doesn't compile, so you need to avoid any library that uses backtrace, including failure.

(re @cr1901): we're have nearly thirty thousand devices out there with most of its  userspace in rust. we replaced tokio with a reasonably sized alternative (i just happened to blog about that), but libstd is an open problem.

jamesmunns, 2019-02-14:
I am closing this issue, please feel free to open another issue if you would like this discussed further.

closed 2019-02-14 by jamesmunns