rust-embedded/wg #5: Programs linked to `std` are too big

japaric, 2016-09-29:
The biggest problem at the lower spectrum of this sub-ecosystem is binary size. Rust programs tend
to be very big in size compared to the common storage sizes (a few MBs) of these devices.

Possible solutions are:

## Dynamic linking

Currently, Rust doesn't have a stable ABI/metadata so it may change from version to version or even
between nightlies. Still, it is possible to use dynamic linking to get a very small executable but
the dynamic libraries (dylibs) are still rather big (due to metadata?).

## Top down approach: strip down the binaries

Quite a few things can be done to the reduce the binary size of general Rust programs. This [blog
post] does an excellent of covering them. 

[blog post]: https://lifthrasiir.github.io/rustlog/why-is-a-rust-executable-large.html

The results look encouraging in the "Hello world" example: from 600K to 8K but these results "don't
scale" to more complex programs. For instance, a program that depends on clap are still around 1 MB
in size, even with all these tricks.

## Bottom up approach: Bring your own standard library

[lrs] is a re-implementation of Rust standard library that claims to produce very small binaries
with fast compile times but only supports Linux (all OpenWRT devices run Linux though). Sadly, it
appears to no longer be maintained, doesn't support MIPS and it's not compatible with the crates on
crates.io because lrs API is different from `std`'s so we can't tell if complex program result in
significantly smaller binaries if lrs is used instead of `std`.

[lrs]: https://github.com/lrs-lang/lib

kevinmehall, 2016-09-29:
> Currently, Rust doesn't have a stable ABI/metadata so it may change from version to version or even between nightlies.

OTOH, this is much less of a blocker in the embedded Linux space than on desktop OSs. It's common for the build process to produce a read-only squashfs with the entire system, and updates rebuild and replace that entire filesystem. But, because of this issue, dynamic linking hasn't seen much attention in the ecosystem and isn't well-supported in Cargo.

jamesmunns, 2019-02-03:
At the moment, we don't have a particular focus on the areas most affected by this, such as embedded linux targets with extremely limited flash space. Because of this, I'd like to propose that we close this issue.

Marking this for a cleanup sweep. If we would like this to stay open, please provide an update to what this issue should be focused on.

jamesmunns, 2019-02-14:
I am closing this issue, please feel free to open another issue if you would like this discussed further.

closed 2019-02-14 by jamesmunns