rust-embedded/wg #9: Cargo doesn't work out of the box for Cortex-M micros

japaric, 2016-09-29:
The Rust compiler supports the Cortex-M architecture but this doesn't work out of the box:

```
$ cargo build --target thumbv6m-none-eabi
```

for two reasons:

- `thumbv6m-none-eabi` is not a "built-in" target so you have to supply your own target definition.
- `cargo build` doesn't compile the standard crates and there's no binary release of the standard
  crates for this target.
  
The upstream solution is to either:

- Add the thumb targets to the compiler *and* also provide binary releases of the standard crates
  for them. Or,
- Just add the targets and teach Cargo how to build the standard libraries "on the fly".

This is being worked on. Seems like the Rust team is leaning towards the second solution. You can
track the upstream status in:

- rust-lang/rfcs#1133 Teach Cargo how to build the standard crates on the fly
- rust-lang/rfcs#1645 Add the thumb targets

japaric, 2016-10-05:
Update: The target definitions have landed upstream in rust-lang/rust#36874.

thejpster, 2016-10-15:
This upstream change is excellent. I'll go and delete the custom targets from primer immediately.

Is it possible to set a default target for a crate? If we have a one-board-per-crate model then specifying the crate seems redundant.

japaric, 2016-10-16:
>  Is it possible to set a default target for a crate?

Not using a crate, no. You can use .cargo/config but that's "local" to the current project and doesn't get passed up to the top crate if set in one of the dependencies.

This "feature" has been mentioned before in the issue tracker before though (can't find the comment right now).

japaric, 2017-05-18:
In the short term I hope to make binary releases of the core, compiler-builtins, etc. crates for the thumb targets available and installable via rustup; that should let us drop the dependency on Xargo for those targets.

japaric, 2017-06-03:
> In the short term I hope to make binary releases of the core, compiler-builtins, etc. crates for the thumb targets available and installable via rustup; that should let us drop the dependency on Xargo for those targets.

One blocker to do this is landing the [compiler-builtins] crate into rust-lang/rust as we require a libcompiler_builtins crate that contains weak memcpy et al symbols in the sysroot to make all this work. 
If someone wants to help with that [this comment] explains what needs to be done.

[compiler-builtins]: https://github.com/rust-lang-nursery/compiler-builtins
[this comment]: https://github.com/rust-lang-nursery/compiler-builtins/issues/66#issuecomment-305943313

jamesmunns, 2019-02-03:
I believe now that some embedded targets have landed, we can close this issue. Marking this for a cleanup sweep. If we would like this to stay open, please provide an update to what this issue should be focused on.

jamesmunns, 2019-02-14:
I am closing this issue, please feel free to open another issue if you would like this discussed further.

closed 2019-02-14 by jamesmunns