rust-embedded/wg #12: Survey: List of crates you use in embedded development

japaric, 2016-10-05:
Ultimately, we'd like to have a list of curated crates that are nicely categorized in a presentable format (a website with search function, etc.). But, for starters, let's list here all the crates we use in embedded development.

Just add a comment here with the list of crates that you use The crate can be a library or a tool (binary). Try to stick to the following format:

```
[`$crate_name`]($crates_io_url_or_github_repo_url) - $description
Tags: `$tag1` `$tag2` ...
```

I'll be collecting them here in the top comment. Try to not to list a crate that has already been mentioned unless you'd like to propose more tags for it.

The tags at this point are free form and can be anything you think will be useful to someone that has to decide whether they can use that crate or not (is it for my target architecture/device? Is it no_std? etc.).

This is also a good opportunity to figure out what are common tags that we could use for #11.

phil-opp, 2016-10-06:
[`bitflags`](https://crates.io/crates/bitflags) - A macro for C-like bit flags, pretty useful for modeling hardware registers.
Tags: `no_std` `macro` `bitflags`

[`lazy_static`](https://crates.io/crates/lazy_static) - A macro that allows to initialize a static using arbitrary code.
Tags: `no_std` `marco` static` `lazy`

[`rustc_builtins`](https://github.com/japaric/rustc-builtins) - Current state of the Rust port of `compiler-rt`. Provides various intrinsics that are required by the compiler.
Tags: `no_std` `compiler-rt` `instrinsics`

[`xargo`](https://github.com/japaric/xargo) - A drop-in replacement for xargo that compiles the sysroot for custom targets on the fly.
Tags: `cargo` `sysroot` `core` `libcore` `liballoc` `libcollections`

Amanieu, 2016-10-06:
[`intrusive-collections`](https://crates.io/crates/intrusive-collections) - Intrusive (non-allocating) singly/doubly linked lists and red-black trees.
Tags: `no_std` `intrusive` `rbtree` `collections`

[`log`](https://crates.io/crates/log) - Supports logging in `no_std` environments, as long as you define your own logger type.
Tags: `no_std` `log`

japaric, 2016-10-06:
> xargo

:heart: 

> rustc_builtins

if you use the `rustc_builtins` crate, you are morally obliged to contribute back to it by porting at least one compiler-rt intrinsic to Rust. :smiling_imp:. (I should totally go and modify that crate to show that message as a warning everytime it gets compiled.)

---

[`cortex-m`](https://crates.io/crates/cortex-m) - Low level access to Cortex-M processors
Tags: `arm` `cortex-m` `hardware` `no_std` `register` `peripheral`

Inspired by the [x86](https://crates.io/crates/x86) crate, I created the `cortex-m` crate with the hope that everyone who works with Cortex-M microcontrollers can build upon it and help me improve it!

nastevens, 2016-10-16:
[`meta-rust`](https://github.com/meta-rust/meta-rust) - Yocto layer providing from-source version of the Rust compiler and Cargo
Tags: `build` `yocto`

[`meta-rust-bin`](https://github.com/rust-embedded/meta-rust-bin) - Yocto layer that uses pre-built Rust and Cargo from rust-lang.org
Tags: `build` `yocto`

[`fixedvec`](https://github.com/rust-embedded/fixedvec-rs) - `Vec`-like interface to a pre-sized buffer of memory. Allows using something like `Vec` when doing heapless programming.
Tags: `no_std `

[`sysfs-gpio`](https://github.com/rust-embedded/rust-sysfs-gpio) - A Rust Interface to the Linux sysfs GPIO interface
Tags: `hardware` `linux` `gpio`

[`i2cdev`](https://github.com/rust-embedded/rust-i2cdev) - Rust library for interfacing with i2c devices under Linux
Tags: `hardware` `linux` `i2c`

[`spidev`](https://github.com/rust-embedded/rust-spidev) - Rust library providing access to spidev devices under Linux
Tags: `hardware` `linux` `spi`

whitequark, 2016-10-16:
[log_buffer](https://github.com/whitequark/rust-log_buffer) - Rust library for logging into a circular string buffer
Tags: `no_std`

dpc, 2016-10-16:
[`slog`](https://crates.io/crates/slog) -  Supports logging in `no_std` environments, as long as you implement your own `Drain`
Tags: `log` `no_std`

SimonSapin, 2016-10-16:
[teensy3-rs](https://github.com/jamesmunns/teensy3-rs/) - bindings for the official [C/C++ stack](https://github.com/PaulStoffregen/cores) for [Teensy 3.x](https://www.pjrc.com/teensy/teensy31.html) boards.

flosse, 2016-10-16:
here are some links to related posts: https://github.com/flosse/rust-os-comparison#embedded-systems

Geal, 2016-10-16:
I don't know if you'd do much parsing in embedded dev, but [nom](https://github.com/Geal/nom) can build in `no_std` (and if it fails for you, I'm willing to fix it and support no_std correctly).

emk, 2016-10-16:
I have three very minor crates that were all building on nightly as of a couple of weeks ago:

- [cpuio](https://crates.io/crates/cpuio). Bare metal (no_std) inb, outb, inw, outw, inl, outw instructions, with a Rust-friendly API.
- [alloc_buddy_simple](https://crates.io/crates/alloc_buddy_simple). Simple, drop-in replacement allocator for Rust running on bare metal. This intended for small, known-size heaps that contain simple collections, such as lists of USB devices discovered on the bus.
- [pic8259_simple](https://crates.io/crates/pic8259_simple). Kernel-space interface to the 8259 and 8259A interrupt controllers.

japaric, 2016-10-16:
If you are working with Cortex-M microcontrollers, need register maps for their peripherals but
don't feel like writing an API that involves thousands of LoCs, you can use:

[`svd2rust`](https://crates.io/crates/svd2rust) - Generate Rust register maps (`struct`s) from SVD files
Tags: `tool` `arm` `cortex-m` `registers` `peripherals`

to save yourself lots of time. Many thanks to @brandonedens for porting my initial version that used
quasi/aster to quote/syn! :heart:

---

@flosse that link seems more appropriate for #15

@Geal Some embedded applications must not use any heap memory. Can `nom` produce parsers that don't
rely on the `alloc` / `collections` crates? Some bare-metal targets, like the `thumbv6m-none-eabi`,
can't even compile the `alloc` crate because they don't support atomics and the `alloc` crate uses
atomics for its oom handler :-(.

@emk nice, an allocator! I've been looking for one. Is this one x86 specific? Do you know if it will
run on a Cortex-M processor? (`usize = 32`, single core, no MMU, no virtual memory, no memory pages, etc.)

aochagavia, 2016-10-16:
I guess "xargo - A drop-in replacement for xargo..." should be "xargo - A drop-in replacement for cargo"

thejpster, 2016-10-16:
My [launchpad](https://github.com/thejpster/bare-metal-arm-rust) project uses these (according to my extern crate statements anyway...)

[volatile_register](https://crates.io/crates/volatile-register) - for memory mapped registers
Tags: kernel_mode

[spin](https://crates.io/crates/spin) - a spinlock Mutex
Tags: no_std

[linked_list_allocator](https://crates.io/crates/linked_list_allocator) - basic memory allocation
Tags: bare_metal

[rlibc](https://crates.io/crates/rlibc) - pure Rust implementation of some required C library functions (like memset)
Tags: no_std

[compiler_builtins](https://github.com/rust-lang-nursery/compiler-builtins) - Instrinsics LLVM needs
Tags: no_std

emk, 2016-10-16:
@japaric:

> nice, an allocator! I've been looking for one. Is this one x86 specific? Do you know if it will
run on a Cortex-M processor?

I've only tried it on x86_64, but it can probably be made to work almost anywhere with a few tweaks. The major advantage of `alloc_buddy_simple` is that it's _tiny_, and you can understand the whole thing front-to-back in an hour or so. And it's good enough to make `collections` basically usable. The biggest drawbacks are that (1) it can't grow its heap, and (2) it hasn't been tested under heavy allocation load, so there might still be some bugs.

[Buddy allocators](https://en.wikipedia.org/wiki/Buddy_memory_allocation) are one of the simplest allocators. It's a fun algorithm.

thejpster, 2016-10-16:
linked_list_allocator is from Phil Opperman's x86-64 blog but but it works on a Cortex-M4.

emk, 2016-10-16:
I also note that nobody has mentioned this popular crate yet:

[x86](https://crates.io/crates/x86). Library to program x86 (amd64) hardware. Contains x86 specific data structure descriptions, data-tables, as well as convenience function to call assembly instructions typically not exposed in higher level languages.

Geal, 2016-10-16:
some parsers can allocate and return a Vec, but the rest won't rely on
allocations. You can provide your own alloc and collections replacement.
Even without, it should build correctly, since the code using Vec is in
macros.

japaric, 2016-10-16:
@aochagavia yes, thank you.

@thejpster @emk Thanks for the pointers. I'll read more about these allocators. :+1:

whitequark, 2016-10-16:
> Some bare-metal targets, like the thumbv6m-none-eabi,
> can't even compile the alloc crate because they don't support atomics and the alloc crate uses
> atomics for its oom handler :-(.

@japaric This is of course not the case, thumbv6m supports atomic reads/writes via regular read/write instructions and atomic read/modify/writes through a cpsid/cpsie sequence. LLVM only needs to be taught that.

sbarral, 2016-10-17:
A crate to load and control bare-metal programs on the PRU, an ARM co-processor found among others on the BeagleBone.

[`prusst`](https://github.com/sbarral/prusst) - Rust interface to the PRU subsystem on AM335x processors (BeagleBone & other SBC)
Tags: `hardware` `linux` `pru` `beaglebone`

japaric, 2016-10-21:
I've started porting `libm` to Rust because I needed to do some floating point math on my microcontroller. The crate has very few functions right now though.

[`m`](https://crates.io/crates/m) - A C free / pure Rust mathematical library ("libm") for `no_std` code
Tags: `no_std` `math` `float`

---

@whitequark 

> thumbv6m supports atomic reads/writes via regular read/write instructions

That's true. I personally don't know how one would implement the `Atomic*` Rust API with just that but I suppose it should be possible.

> atomic read/modify/writes through a cpsid/cpsie sequence.

I'm unsure about an implementation like that. cpsid/spsie only work in privileged mode and are no-op in user mode. An implementation of `Atomic*` that is not atomic in some cases doesn't sound compelling to me.

cwoodall, 2016-10-21:
There are a few other ways to do atomics. 

You can set basepri to max which will mask out all interrupts (except those with 0 priority). This is what brittle and freertos seem to do. I suppose this is more for critical sections than atomics. 

You can also use the `strex` and `ldrex` instructions to form some form of a to. Uc operation (this is what the gcc atomic primatives seem to do) http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dht0008a/ch01s02s01.html

cwoodall, 2016-10-21:
This stackoverflow.com article appears to be somewhat insightful: http://stackoverflow.com/questions/11894059/atomic-operations-in-arm

Might be hard to wrap this properly.

japaric, 2016-10-21:
> You can set basepri to max which will mask out all interrupts

This, like cpsid/cpsie, can only be done in privileged mode.

> You can also use the strex and ldrex instructions

These are not available for the thumbv6m target (which is what I said is doesn't support atomics -- keep reading). For thumbv7 and newer, rustc (or rather LLVM) lowers `Atomic*`'s methods to strex/ldrex instructions.

whitequark, 2016-10-21:
@japaric 

> I personally don't know how one would implement the Atomic* Rust API with just that but I suppose it should be possible.

You cannot. You need RMW cycles to implement the complete API.

> I'm unsure about an implementation like that. cpsid/spsie only work in privileged mode and are no-op in user mode. An implementation of Atomic* that is not atomic in some cases doesn't sound compelling to me.

Indeed. Fortunately... thumbv6m doesn't have the concept of user mode, as far as I'm aware! So this is not a problem.

thejpster, 2016-10-22:
I pulled out the drivers from my Stellaris Launchpad project, so I'd like to add:

[lm4f120](https://crates.io/crates/lm4f120) - A set of drivers for the TI LM4F120.
Tags: `bare-metal` `drivers`

japaric, 2016-10-28:
@whitequark oh, then it sounds like we could provide the required intrinsics in some standard-ish crate. I've opened rust-lang-nursery/compiler-builtins#114 to track this.

japaric, 2016-12-06:
@Razican on #rust-embedded

[`fel-cli`](https://crates.io/crates/fel-cli) - CLI tools for dealing with Allwinner devices in FEL mode, in pure Rust.
Tags: `arm` `sunxi` `allwiner` `fel`

> They are tools for working with Allwinner ARM devices in FEL mode. I would like to use it to learn to develop a small bootloader/OS for the C.H.I.P. device, and I think it could be useful to somebody else

japaric, 2016-12-07:
An easy to plug in allocator crate based on `linked_list_allocator`

[`alloc-cortex-m`](https://crates.io/crates/alloc-cortex-m) - A heap allocator for Cortex-M processors
Tags: `arm` `cortex-m` `allocator`

japaric, 2017-02-20:
(I just realized that I hadn't yet posted these two projects here.)

`cortex-m-semihosting` is a crate to do semihosting on ARM Cortex-M devices. It provides a `syscall!` macro to invoke operations like WRITE. It also provides a convenience `hprintln!` macro to print formatted messages on the host's stdout.

[`cortex-m-semihosting`](https://crates.io/crates/cortex-m-semihosting) - Semihosting for ARM Cortex-M processors
Tags: `arm` `cortex-m` `semihosting`

---

`cortex-m-template` is a Cargo project template (i.e. `cargo new foo --template https://github.com/japaric/cortex-m-template`) for ARM Cortex-M bare metal development. It comes with all the stuff required to build a minimal program: a linker script, startup code, etc. and requires minimal configuration -- you just have to specify the memory layout of your device. Plus it ships with other goodies like a `.gdbinit` file that flashes and starts a debug sessions, a default exception handler with access to the faulty stack frame, an overridable panic handler, opt-in semihosting, several examples, etc.

[`cortex-m-template`](https://github.com/japaric/cortex-m-template) - A Cargo project template for ARM Cortex-M bare metal development
Tags: `arm` `cortex-m` `Cargo` `template`

japaric, 2017-02-28:
I built a custom `test` crate that supports `no_std` targets. It can run unit `#[test]`s on real Cortex-M microcontrollers as well as on emulated Cortex-M processors (QEMU user emulation).

[`utest`](https://github.com/japaric/utest) - Unit `#[test]`ing for microcontrollers and other `no_std` systems 
Tags: `arm` `cortex-m` `test` `runner`

vitiral, 2017-05-27:
[byteorder](https://github.com/BurntSushi/byteorder) should be included here. It supports `no_std` with `default-features = false`

japaric, 2017-05-29:
@vitiral Added.

A few more crates

[`cortex-m-quickstart`](https://crates.io/crates/cortex-m-quickstart) - A template for building applications for ARM Cortex-M microcontrollers
Tags: `arm` `cortex-m` `template`

[Introductory blog post](http://blog.japaric.io/quickstart)

[`cortex-m-rtfm`](https://crates.io/crates/cortex-m-rtfm) - Real Time For the Masses (RTFM), a framework for building concurrent applications, for ARM Cortex-M microcontrollers
Tags: `arm` `cortex-m` `concurrency`

[Introductory blog post](http://blog.japaric.io/fearless-concurrency)

[`heapless`](https://crates.io/crates/heapless) - `static` friendly data structures that don't require dynamic memory allocation
Tags: `no-heap` `static`

thejpster, 2017-05-29:
I'm using [stm32f7-discovery](http://github.com/embed-rs/stm32f7-discovery) to play with, erm, an STM32F7 Discovery.

vitiral, 2017-05-31:
**[build-const](https://crates.io/crates/build_const)** - a helper library for computing constants at compile time in `build.rs` or a script. Also includes `no_std` macros for importing them.

Tags: `["embedded", "no_std", "build", "const", "static"]`

**[defrag](https://crates.io/crates/defrag)** - safe defragmenting memory manager for microcontrollers

Tags: `["no_std", "embedded", "memory", "manager", "microcontroller"]`

**[bridge](https://github.com/vitiral/bridge)** - (in development) a protocol designed to allow a system of different devices, to communicate with each other and issue commands through (optionally) guaranteed unique remote procedural calls. It allows inter-network communication through the concept of "bridges". Both nodes and bridges can be highly resource constrained devices (such as microcontrollers). Intended supported networks include tcp/ip, UART, CAN and ZigBee.

This is still in the design phase and I am actively looking for feedback.

japaric, 2017-06-02:
[`aligned`](https://crates.io/crates/aligned) - Statically allocated arrays with guaranteed memory alignments
Categories: `embedded` `memory-management` `no-std`
Tags: `alignment` `aligned` `array`

[`cty`](https://crates.io/crates/cty) - Type aliases to C types like c_int for
use with bindgen
Categories: `embedded` `ffi` `no-std`
Tags: `bindgen` `ffi` `c` `types`

goandylok, 2017-06-06:
[`scroll`](https://crates.io/crates/scroll) A suite of powerful, extensible, generic, endian-aware Read/Write traits and contextual conversion traits (Byte serializing and deserializing)
Tags: `pread` `bytes` `endian` `pwrite`

vitiral, 2017-06-07:
I just started work on [ubyte](https://github.com/vitiral/ubyte), a rust library for serialization/deserialization on microcontrollers. The API is still in progress, but I got `serde::Serializer` to compile so I'm well on my way to a functional library

vitiral, 2017-06-08:
on the above note:

[`ssmarshal`](https://crates.io/crates/ssmarshal) - stupid simple (no_std) serde serialzation library
Tags: `no_std`, [see issue](https://gitlab.com/robigalia/ssmarshal/issues/3_

vitiral, 2017-06-08:
Also, one I recently wrote:

[`build_const`](https://crates.io/crates/build_const) - replacement for `lazy_static` to build constants at compile time (instead of lazily) by using the build script.
Tags: `no_std`, `embedded`, `const`, `static`, `build`

whitequark, 2017-06-17:
<code>[smoltcp](https://github.com/m-labs/smoltcp)</code> ­— a no_std TCP/IP stack

goandylok, 2017-06-28:
[`byte`](https://crates.io/crates/byte) A low-level, zero-copy, panic-free, binary serializer and deserializer. (parser and encoder)

Tags: `no_std` `byte` `encode` `binary` `parser`

kunerd, 2017-06-30:
[clerk](https://crates.io/crates/clerk) - A hardware agnostic HD44780 LCD controlling library.
Tags: `embedded` `driver`

japaric, 2017-07-07:
[`stlog`](https://crates.io/crates/stlog) A lightweight logging framework for resource constrained devices

Categories: `embedded` `no-std` `development-tools::debugging`
Keywords: `log` `elf` `symtab`

`stlog` can log strings of arbitrary size to a host in a constant number of instructions (O(1) runtime). Furthermore these strings are *not* stored in the device memory (O(0) memory usage). `stlog` supports 5 logging levels: TRACE, DEBUG, INFO, WARN and ERROR; each level can be individually disabled at *compile time* (there's no branching cost at runtime). `stlog` also offers a global logging mode: `info!("Hello")` will log the string using the global logger; the global logger can be configured using the `set_global_logger!` macro.

Due to its focus on minimal overhead `stlog` doesn't support formatting -- only literal strings can be logged.

cc @SimonSapin ^

idubrov, 2017-09-16:
[`lcd`](https://github.com/idubrov/lcd) - Yet another HD44780/LCD display library (third, at least)! Batteries are _not_ included: all binding to the hardware/GPIO initialization is the responsibility of the user (including the sleep function).
Tags: `no_std` `embedded`

fluffysquirrels, 2017-12-25:
[genio](https://crates.io/crates/genio) - io::{Read, Write} traits for no_std. It has wrappers for std::io::{Read, Write} to allow use in crates that support std and no_std. Its Read, Write traits have associated type parameters to support custom error types.

fluffysquirrels, 2017-12-31:
[`framed`](https://crates.io/crates/framed) - Rust crate to send and receive data over lossy streams of bytes.

* Supports sending slices of bytes or serde-serialized types.
* Uses a checksum to detect and skip corrupted frames.
* Includes a command-line tool to decode and display serialized types.

Full disclosure: I wrote this.

Amanieu, 2017-12-31:
[`cstr_core`](https://crates.io/crates/cstr_core) - `#[no_std]` port of `std::ffi::CStr` and `std::ffi::CString`. `CString` support is optional, and uses the `alloc` crate instead of `std`.

[`hashmap_core`](https://crates.io/crates/hashmap_core) - `#[no_std]` port of `std::collections::HashMap` and `std::collections::HashSet`. This crate uses `alloc` instead of `std` and uses FNV as the default hash function since SipHash requires a source of random numbers.

whitequark, 2017-12-31:
@Amanieu any plans to push this through the RFC process? In particular HashMap not being in `alloc` is ridiculous.

Amanieu, 2017-12-31:
@whitequark The problem is that the default `HashMap` constructor uses `SipHash`, which requires a call to the system RNG to get a random seed.

whitequark, 2017-12-31:
@Amanieu I know. Can't the libstd facade provide it, though?

DoumanAsh, 2018-12-06:
Just wanted to mention my logging crate, I've been working in past.

[cortex-m-log](https://github.com/DoumanAsh/cortex-m-log) It provides generics over possible outputs (for now it is only semihosting, itm and dummy), I didn't have much of feedback so far and I haven't been working with embedded stuff for quite some time, but I appreciate increased visibility since I'd like to start working on embedded again.
It also contains simple integration with `log` crate and allows you to implement own writers/printers to be used with `log` integration.

therealprof, 2018-12-06:
@DoumanAsh Interesting. Have to check it out, I've been using the log crate previously to log output to a UART and wrote an adapter to use that for `panic`s, too. It's a bit foot-in-mouth to use though...

DoumanAsh, 2018-12-06:
@therealprof Here is example in my playground with log https://github.com/DoumanAsh/2B/commit/35d6bc5dc4bee4322e411af12759ddbad54894b6

If you want to write custom destination for logs, you need to implement [Printer](https://docs.rs/cortex-m-log/0.3.3/cortex_m_log/printer/trait.Printer.html) then it will just work

eldruin, 2018-12-06:
[`embedded-hal-mock`](https://crates.io/crates/embedded-hal-mock) - A collection of mocked devices that implement the embedded-hal traits.
Tags: `testing` `hal`

jamesmunns, 2019-02-03:
I believe that this issue has been superseded by the github.com/rust-embedded/awesome-embedded-rust repo. I would propose that we close this issue.

Marking this for a cleanup sweep. If we would like this to stay open, please provide an update to what this issue should be focused on.

jamesmunns, 2019-02-14:
I am closing this issue, please feel free to open another issue if you would like this discussed further.

closed 2019-02-14 by jamesmunns