rust-embedded/wg #17: Let's participate in the novemb.rs Code Sprint

japaric, 2016-11-04:
[Website](http://novemb.rs/)

TL;DR It's an offline/online/remote Rust code sprint with a focus on
European/African time zones/locations but everyone's welcome no matter their
time zone/location.

The event will last two days: November 19 and 20.

It seems like a great opportunity to get together and work on embedded stuff.
We can use the #rust-embedded IRC channel to coordinate our work.

We are currently discussing what we could work on in rust-community/novemb.rs#4.
The discussion went a little off topic but I've posted what areas I think we
could focus on in [my latest comment][0] (see the etherpad link).

[0]: https://github.com/rust-community/novemb.rs/issues/4#issuecomment-258422966

Please fill the list in the etherpad if you'd like to participate and let's
discuss about the "focus areas" on *that* thread.

wesen, 2016-11-06:
Here is a link to the etherpad: [https://public.etherpad-mozilla.org/p/embedded-novemb.rs](https://public.etherpad-mozilla.org/p/embedded-novemb.rs)

japaric, 2016-11-24:
We had a great and productive time during novemb.rs!

Here's a list of accomplishments during that weekend:

- The number of LEDs blinked using Rust increased exponentially. :smile:

- The number of "permanent" people in the #rust-embedded channel increased by ~50%. Welcome all!

- An easy to use allocator crate for Cortex-M microcontrollers: [alloc-cortex-m].

[alloc-cortex-m]: https://crates.io/crates/alloc-cortex-m

- A [few] low level [crates] to manipulate registers popped up. All auto-generated using [svd2rust].

[few]: https://github.com/NathanielPerkins/stm32f411xx-memory-map
[crates]: https://github.com/brandonedens/stm32l4x6-memory-map
[svd2rust]: https://crates.io/crates/svd2rust

- One high level [crate] to manipulate the STM32F411E-DISCO board.

[crate]: https://github.com/ShaunKarran/stm32f411e-disco

- Someone managed to debug a microcontroller program from within Visual Studio.

- Someone managed to build a microcontroller program on Eclipse using the EclipseDT plugin. (Though it was rather hard but I hope it'll be much easier in the future)

- We heard a success story about using Rust on embedded Linux. (Sorry, #rust-embedded has no logs)

- A [crate][pn532] to interface the PN532, a NFC/RFID controller.

[pn532]: https://github.com/Funcoil/pn532-rs

- We sketched a few APIs for the Serial and SPI peripherals. These have been posted to #19

- And probably lots of other stuff that didn't get reported to IRC.

Thanks to all the participants. Let's e-meet again sometime in the future!

closed 2016-11-24 by japaric