rust-embedded/wg #20: MSP430

japaric, 2016-11-11:
Parent issue: #2

Next steps:

- [ ] Compiler changes: Enable the MSP430 LLVM backend, implement cabi stuff,
  etc. See rust-lang/rust#37672
- [ ] Experiment with a custom target out of tree
- [ ] Write documentation about writing/compiling/flashing/debugging Rust
  programs for this target
- [ ] Add a built-in MSP430 target to rustc
- [ ] :tada:

japaric, 2016-11-11:
> Experiment with a custom target out of tree

rust-lang/rust#37672 has a custom target specification in the PR description but
it has only been lightly tested. What needs to be done after that PR lands is
compiling a Rust program that actually "works", i.e. that can be flashed and
that boots correctly.

To do that, we need to know low level details about the microcontrollers:

- How does the boot process work?
- How is the memory address space organized? What addresses belong to RAM, which
  ones belong to Flash? etc.

And about the tooling, in particular linker scripts.

msp430-gcc ships with several linker scripts. Using the linker scripts
(`memory.x` and `periph.x`) of the msp430g2202 micro and an empty file I get
this:

```
$ touch empty.c

$ clang --target=msp430 empty.c

$ msp430-objdump -Cd a.out
0000f800 <__watchdog_support>:
    f800:       55 42 20 01     mov.b   &0x0120,r5
    f804:       35 d0 08 5a     bis     #23048, r5      ;#0x5a08
    f808:       82 45 00 02     mov     r5,     &0x0200

0000f80c <__init_stack>:
    f80c:       31 40 00 03     mov     #768,   r1      ;#0x0300

0000f810 <__do_copy_data>:
    f810:       3f 40 00 00     mov     #0,     r15     ;#0x0000
    f814:       0f 93           tst     r15
    f816:       08 24           jz      $+18            ;abs 0xf828
    f818:       92 42 00 02     mov     &0x0200,&0x0120
    f81c:       20 01
    f81e:       2f 83           decd    r15
    f820:       9f 4f 4a f8     mov     -1974(r15),512(r15);0xf84a(r15), 0x0200(r15)
    f824:       00 02
    f826:       f8 23           jnz     $-14            ;abs 0xf818

0000f828 <__do_clear_bss>:
    f828:       3f 40 00 00     mov     #0,     r15     ;#0x0000
    f82c:       0f 93           tst     r15
    f82e:       07 24           jz      $+16            ;abs 0xf83e
    f830:       92 42 00 02     mov     &0x0200,&0x0120
    f834:       20 01
    f836:       1f 83           dec     r15
    f838:       cf 43 00 02     mov.b   #0,     512(r15);r3 As==00, 0x0200(r15)
    f83c:       f9 23           jnz     $-12            ;abs 0xf830

0000f83e <__stop_progExec__>:
    f83e:       32 d0 f0 00     bis     #240,   r2      ;#0x00f0
    f842:       fd 3f           jmp     $-4             ;abs 0xf83e

0000f844 <__ctors_end>:
    f844:       30 40 48 f8     br      #0xf848

0000f848 <_unexpected_>:
    f848:       00 13           reti

Disassembly of section .vectors:

0000ffe0 <__ivtbl_16>:
    ffe0:       44 f8 44 f8 44 f8 44 f8 44 f8 44 f8 44 f8 44 f8     D.D.D.D.D.D.D.D.
    fff0:       44 f8 44 f8 44 f8 44 f8 44 f8 44 f8 44 f8 00 f8     D.D.D.D.D.D.D...
```

I expect this program should "work" if flashed into the micro. As least it
should boot properly.

Changing the C source code to:

``` c
int main() {
    while (1) {}
}
```

Changes the disassembly like this:

``` diff
     f818:      92 42 00 02     mov     &0x0200,&0x0120
     f81c:      20 01
     f81e:      2f 83           decd    r15
-    f820:      9f 4f 4a f8     mov     -1974(r15),512(r15);0xf84a(r15), 0x0200(r15)
+    f820:      9f 4f 5a f8     mov     -1958(r15),512(r15);0xf85a(r15), 0x0200(r15)
     f824:      00 02
     f826:      f8 23           jnz     $-14            ;abs 0xf818

@@ -38,10 +38,19 @@
     f842:      fd 3f           jmp     $-4             ;abs 0xf83e

 0000f844 <__ctors_end>:
-    f844:      30 40 48 f8     br      #0xf848
+    f844:      30 40 58 f8     br      #0xf858

-0000f848 <_unexpected_>:
-    f848:      00 13           reti
+0000f848 <main>:
+    f848:      04 12           push    r4
+    f84a:      04 41           mov     r1,     r4
+    f84c:      21 83           decd    r1
+    f84e:      84 43 fe ff     mov     #0,     -2(r4)  ;r3 As==00, 0xfffe(r4)
+    f852:      00 3c           jmp     $+2             ;abs 0xf854
+    f854:      ff 3f           jmp     $+0             ;abs 0xf854
+       ...
+
+0000f858 <_unexpected_>:
+    f858:      00 13           reti

 Disassembly of section .vectors:
```

But compiling the following Rust program:

``` rust
#![feature(lang_items)]
#![feature(no_core)]
#![no_core]
#![no_main]

#[export_name = "main"]
pub fn main() -> ! {
    loop {}
}

#[lang = "copy"]
trait Copy {}

#[lang = "sized"]
trait Sized {}
```

produces the same disassembly as the empty C source file. This means that the
linker is dropping all the Rust symbols. Have to figure out why that's the case.

pftbest, 2016-11-11:
I can get a MSP430 board tomorrow, to do some tests.
Also I have a question, how does rust support `__attribute__((interrupt))` declaration?
And there is no way (as far as I know) to exit low-power mode, because `__bic_SR_register_on_exit` built-in is not implemented in llvm.

japaric, 2016-11-11:
> how does rust support __attribute__((interrupt)) declaration?

It doesn't right now. We'll have to add a new ABI, e.g. `extern "msp430-interrupt"`, to get the ABI/prelude/calling convention right. The AVR folks have done something like this, I think.

japaric, 2016-11-11:
> because __bic_SR_register_on_exit built-in is not implemented in llvm.

Could you elaborate about what this does? Is a gcc intrinsic that lowers to an assembly instruction? Or something else?

pftbest, 2016-11-11:
@japaric 
>Could you elaborate about what this does?

It modifies the value on the stack that is located one word below the current frame (I don't know the right words how to say this). Basically when CPU enters interrupt it does this:

1. CPU pushes status register on stack
2. CPU jumps to interrupt handler
3. interrupt handler pushes some stuff on the stack (it's frame)
4. here we need to modify the value that was pushed on stack by CPU on step 1 but we can't locate it because only compiler knows how much stuff we have pushed on step 3, that's why we use built-in.
5. interrupt handler pops it's frame from the stack
6. CPU pops the value from stack and writes it to status register.

As far as I know this is the only way to modify the status register and exit low power mode.

pftbest, 2016-11-11:
@japaric 
Maybe it is possible to implement this behaviour using naked functions and some assembly code, but I have not tried.

japaric, 2016-11-11:
The built-in looks like a function call that goes inside an "interrupt handler"? Something like this:

``` c
__irq void foo() {
    // ..

     __bic_SR_register_on_exit();
}
```

> Maybe it is possible to implement this behaviour using naked functions and some assembly code, but I have not tried.

Maybe. But I think that if you use a naked function then you have to manually modify the stack pointer during the prologue/epilogue of the function according to the number of local variables you use. My worry is that this approach may require the (final) user to deal with low level ABI-ish details.

japaric, 2016-11-11:
> The built-in looks like a function call that goes inside an "interrupt handler"? 

Or is an attribute?

pftbest, 2016-11-11:
> The built-in looks like a function call that goes inside an "interrupt handler"?

Yes, it looks and feels like normal function.

## Synopsis
```C
unsigned __bic_SR_register_on_exit(unsigned mask);
```
## Description
`__bic_SR_register_on_exit` clears the bits specified in mask in the saved status register of an interrupt function (i.e. it bitwise ands the complement of mask into the saved status register). This allows you to change the operating mode of the MSP430 on return from the interrupt service routine, such as changing the low power mode.
`__bic_SR_register_on_exit` returns the value of the saved MSP430 status register before the update.
`__bic_SR_register_on_exit` is an intrinsic function and produces inline code.
## Note
`__bic_SR_register_on_exit` can only be used in interrupt functions—an error is reported if it is used outside an interrupt function.

pftbest, 2016-11-11:
I think it is not possible to write llvm IR that will do this kind of thing. So this function should be added as intrinsic in llvm (and as built-in in clang). and only then we can have it in rust.

japaric, 2016-11-16:
rust-lang/rust#37672 has landed and the MSP430 LLVM backend should be enabled in the next nightly. People should be able to experiment with custom targets out of tree with that nightly.

jck, 2016-11-16:
I've got a couple of MSP430 fram boards. I'll try out a blinky tomorrow.

pftbest, 2016-11-17:
So I tried to compile core and I get this errors from assembler:
```
core.s: Assembler messages:
core.s:9468: Error: junk at end of line, first unrecognized character is `@'
core.s:9740: Error: junk at end of line, first unrecognized character is `@'
...
```
the code on line 9468 looks like this:
```asm
_ZN4core3num7dec2flt5rawfp10prev_float14_MSG_FILE_LINE17h5bcd06a3b0d78233E:
	.short	str.3@
	.short	27
	.short	str.1v
```
I tried both `msp430-gcc` (4.6.3) and `msp430-elf-gcc` (5.3.0) and both give the same errors. Then I've renamed symbol `str.3@` to `str.3_at` in assembly and it compiled successfully. I think msp430 assembler just does not like `@` in symbol names. So I see 2 options:
1. Modify MSP430 target in LLVM so it will rename symbols with `@`
2. Modify [`generate_local_symbol_name`](https://github.com/rust-lang/rust/blob/master/src/librustc_trans/context.rs#L958) to not emit `@` on MSP430

crypto-universe, 2016-11-27:
Hi!
I have MSP430G2553 and I can make some tests on it. Just let me know.

pftbest, 2016-11-28:
Anyone working on this `@` symbol issue? It is blocking us from compiling core. What will be the right way to fix it? I don't know yet how hard it will be to hack it in llvm, fixing it in rustc seems easier to me.

durka, 2016-12-01:
@pftbest how did you get around the LLVM assertion compiling core?

durka, 2016-12-01:
@japaric re: the linker deleting everything, the problem seems to be `-Wl,--gc-sections`. We have to get rustc to stop passing that to the linker. I ran `cargo rustc` with `-Z print-link-args`, copied the linker command, and removed `-Wl,--gc-sections` and I get [this disassembly](https://gist.github.com/anonymous/debeed0822c36959021537dc4bfae734):

```
0000c048 <start>:
    c048:	04 12       	push	r4		
    c04a:	04 41       	mov	r1,	r4	
    c04c:	21 83       	decd	r1		
    c04e:	00 3c       	jmp	$+2      	;abs 0xc050
    c050:	00 3c       	jmp	$+2      	;abs 0xc052
    c052:	ff 3f       	jmp	$+0      	;abs 0xc052
```

durka, 2016-12-01:
OK, so `cargo rustc --target=msp430 -- -C link-dead-code` seems to do the trick. But I don't know why it's telling the linker that public symbols are dead in the first place.

pftbest, 2016-12-01:
> how did you get around the LLVM assertion compiling core?

So I fixed the issue in LLVM, but it is not committed in upstream yet. See [#37829](https://github.com/rust-lang/rust/issues/37829). Don't know how long it will take to get it into rust. Alternatively, you can just disable the assertions in `configure` script, the code for these functions will be bad anyway. Also there is this issue with `@` in symbol names, that is not fixed yet.

> We have to get rustc to stop passing that to the linker.

There should be a better way to fix this, because without `gc-sections` we won't be able to fit `libcore` in flash memory.

japaric, 2016-12-01:
@pftbest 

> Anyone working on this @ symbol issue? 

I didn't see this issue when compiling core but I used msp430-gcc 6.2.0.

@durka 

If `--gc-sections` is dropping `start`, that means that `start` was never going to be called by the program. This means that the startup objects expect the user entry point to be named differently. IIRC, I tried `start`, `_start` and `main` but none of those worked.

`nm -u crt*.o` should yield some clues about the right name of the user entry point. Otherwise, we could look at the disassembly of a minimal *C* program; that should also give us some clues.

durka, 2016-12-01:
Alternatively, we can pass `-Wl,-emain` to force it.

pftbest, 2016-12-01:
@japaric
> I didn't see this issue when compiling core but I used msp430-gcc 6.2.0.

I believe you saw this issue before [link](https://github.com/rust-lang/rust/pull/37672#issuecomment-260195110)

It happens with both mspgcc and msp430-elf 5.4 (latest stable version from TI) 

Maybe I should create a separate issue for it.

japaric, 2016-12-01:
After re-checking with mps430-gcc 6.2, I see that core doesn't compile due to the "@" business. Weird, I tought that was already working

pftbest, 2016-12-01:
Submitted [#38116](https://github.com/rust-lang/rust/issues/38116)

pftbest, 2016-12-10:
Finally managed to blink a real LED on MSP430 board. The code [is here](https://github.com/pftbest/rust_on_msp), but unfortunately, you need a patched `rustc` to compile it. I've put some info in README for some of the the issues I've ran into while trying to get it working.

japaric, 2016-12-11:
Nice work, @pftbest!

re: MSP430 and MSP430X. Is `-mmcu` the only way to pick between those two variants *during the .s -> .o step*? We could add a `external-asm-args` field to target specifications that would let us pass arguments to `msp430-elf-gcc` during that step but I'd prefer if we used a more "coarse" flag to select between those variants then we'd ideally have only two built-in targets: `msp430-none-elf` and `msp430x-none-elf` and `-mmcu` would only be passed to the *linker* via RUSTFLAGS. IIRC, setting `-mmcu` also picks up some device-specific linker scripts so I don't think that flag should appear in the built-in target specification.

re: -nodefaultlibs. In theory we could implement the startup stuff in Rust to not depend on the startup objects that get (implicitly?) passed to the linker when -nodefaultlibs is *not* used thus achieving pure Rust programs like in the Cortex-M case. In any case, it makes sense to set `no-default-libraries: false` in the built-in target because one can manually pass `-nodefaultlibs` to the linker via RUSTFLAGS but the opposite (removing the -nodefaultlibs flag in the `no-default-libraries: true` case) is not possible without a custom target.

japaric, 2016-12-11:
@pftbest Could you post (in a gist) the disassembly of a minimal program that gets the startup part right? i.e. your rust-on-msp project but with an empty (or `loop {}`) `main` function.

pftbest, 2016-12-11:
@japaric 
compiled minimal rust program and minimal c program, they have almost identical assembly code.
https://gist.github.com/pftbest/917bfa4cf1a0326be464bd885803181d

pftbest, 2016-12-11:
@japaric 

> Is -mmcu the only way to pick between those two variants during the .s -> .o step?

I've looked into binutils source code, and I see that there is only two options, `-mmcu` flag and `-mcpu` flag. You can pass generic names into those like `msp430` or `430x`, but that's it.

> In theory we could implement the startup stuff in Rust to not depend on the startup objects

I think it is possible to write the initialization in pure rust, but I have not tried this yet. But then you will have to manually specify the right linker script in target json anyway.

japaric, 2016-12-12:
> You can pass generic names into those like msp430 or 430x

Seems like those would do the trick.

> But then you will have to manually specify the right linker script in target json anyway.

Not necessarily. You can pass the linker script via RUSTFLAGS / .cargo/config like [the Cortex-M targets do](https://github.com/japaric/f3/blob/ccba1d326a47cfd7ee3d18ad1646ef3faa78faa4/.cargo/config#L4).

pftbest, 2016-12-12:
@japaric
> You can pass the linker script via RUSTFLAGS / .cargo/config
I didn't know about this feature, nice, we can remove the custom target then. Would you make this changes to the compiler?

Also there is one small issue left, right now rust code is not compatible with C code, because llvm is still using old ABI. I am working on a patch that will move llvm to the new EABI. It will be a breaking change, because `mspgcc` is deprecated and it is too much hassle to keep two ABIs around.

japaric, 2016-12-12:
> I didn't know about this feature, nice, we can remove the custom target then. Would you make this changes to the compiler?

By "making these changes to the compiler" you mean adding "built-in" targets for msp430 to rustc? If so, I'd like to more extensively test the targets out of tree before adding them to rustc because that would make them insta-stable and would make it harder to modify them if we didn't get something right in the first try.

> I am working on a patch that will move llvm to the new EABI.

Awesome!

pftbest, 2016-12-17:
Updated the code on https://github.com/pftbest/rust_on_msp
Now it using xargo and works on the latest nightly.
Also it no longer requires startup code from gcc, so this fix with `default-libraries` is not necessary anymore. So the only one big issue left is interrupt CC.
But I think it is already possible to write something useful (or a least to blink a LED) in rust for MSP430, so anyone who has a board may start playing with it.

japaric, 2016-12-17:
@pftbest Excellent! I'd suggest posting your repo to reddit and/or to the thread (or to a new thread) in the users forum to let others know.

Seems like the two most pressing issues on the rustc side are: (a) adding some way to pass flag to the assembler to override the msp430x default and (b) adding support for `extern "msp430-interrupt" fn`. I'll see if I can tackle those during the weekend.

pftbest, 2016-12-17:
@japaric I was thinking maybe I should do some cleanup first, before letting it go out into the wild. I was writing 'how to port to other MCUs' part of the readme, and it involves too many steps that I'd like. Maybe I should make generic `msp430.json` target and use `.cargo/config` instead of this mess.

Regarding this msp430x issue, I think it can be done using llvm features. So if `+ext` feature is not enabled we can safely pass `-mcpu=msp430` flag to the assembler. Also msp430 is backwards compatible with msp430x so it is possible to run msp430 only programs on msp430x CPU.

pftbest, 2016-12-17:
@japaric Also is there a way to prevent xargo from printing out this huge wall of errors when it fails to compile libc crate? I think it may scare somebody 😄

japaric, 2016-12-18:
> So if +ext feature is not enabled we can safely pass -mcpu=msp430 flag to the assembler.

Hmm, wouldn't be better to have to two targets instead of adding special logic to check if `+ext` is present? In any case, I'm certain the tools team would prefer to add a new field to the target specification rather than adding target-specific logic to the compiler.

> Also is there a way to prevent xargo from printing out this huge wall of errors when it fails to compile libc crate? 

Not right now.

Xargo tries to compile everything "below" `std` so it has to show you the errors in case you *actually* wanted to compile the `libc` crate. I know that it doesn't look nice but I'm not sure how to address the issue.

awygle, 2017-01-14:
I am no longer able to compile core for the MSP430 target, either on my own or by using @pftbest 's repo. I am hitting an llvm_unreachable line in "release mode", or if I give rustc "-Z debug-llvm" I hit an assertion. This appears to be different from rust-lang/rust#37829 - something to do with argument lowering.

Should I file an issue for this? Is there more information I could extract that would be useful? I spent a little bit of time poking around but I'm not deeply experienced with LLVM.

Using the rust_on_msp repo, the Unreachable:
```
$ xargo build --release --target msp430
   Compiling core v0.0.0 (file:///home/awygle/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/src/libcore)
Call result #4 has unhandled type i16
UNREACHABLE executed at /buildslave/rust-buildbot/slave/nightly-dist-rustc-linux/build/src/llvm/lib/CodeGen/CallingConvLower.cpp:167!
error: Could not compile `core`.

To learn more, run the command again with --verbose.
error: `"cargo" "build" "--release" "--manifest-path" "/tmp/xargo.OxcnXcVX7N4q/Cargo.toml" "--target" "msp430" "-p" "core"` failed with exit code: Some(101)
note: run with `RUST_BACKTRACE=1` for a backtrace
```

And the Assertion:
```
$ rustc --target msp430 -Z debug-llvm /home/awygle/rust/src/libcore/lib.rs
<MILES AND MILES OF INFORMATION>
=== _ZN4core3num7flt2dec17digits_to_exp_str17hd092cccb448c45aeE
rustc: /home/awygle/rust/src/llvm/lib/CodeGen/SelectionDAG/SelectionDAGBuilder.cpp:7960: void llvm::SelectionDAGISel::LowerArguments(const llvm::Function&): Assertion `EVT(Ins[i].VT) == InVals[i].getValueType() && "LowerFormalArguments emitted a value with the wrong type!"' failed.
zsh: abort      rustc --target msp430 -Z debug-llvm /home/awygle/rust/src/libcore/lib.rs
```

pftbest, 2017-01-14:
@awygle This is issue https://github.com/rust-lang/rust/issues/38824

To fix this we need to patch LLVM so it will support 128bit numbers on MSP430.
Until this fixed, please use older version of nightly, you can get it like this:
```
rustup install nightly-2016-12-29
rustup override set nightly-2016-12-29
rustup component add rust-src
```

akovaski, 2017-04-23:
This is very cool, I was able get a small screen connected through SPI to output text using this :+1:. 2 things I noticed, though:
1. Using pftbest's rust_on_msp repository, the stack pointer is not set on initialization.
* From what I've seen, the stack pointer is retained even when you restart the device. I have resorted to inserting an instruction to set the stack pointer to 0x400. (Above, on a Nov 11, 2016 post, you can see clang sets it to 0x0300. I do not know why clang puts it there). I am not sure what the proper solution for this is (the start file provided by crt0 seems to initialize the stack pointer, but I'm not sure how to link into the start code.) And perhaps people would like to choose where the SP starts.

2. There is something wrong with comparing and setting u32's. I'm not familiar enough with the compiler chain to know what is going wrong.
* I have 3 functions: `u32_ge_bool`, `u32_ge_u32_or`, and `u32_ge_const_u32_or`:
```rust
#[no_mangle]
#[inline(never)]
pub fn u32_ge_bool(a: u32,test: u32) -> bool {
    if a >= test {
        true
    } else {
        false
    }
}
#[no_mangle]
#[inline(never)]
pub fn u32_ge_u32_or(a: u32,test: u32) -> u32 {
    if a >= test {
        a | test
    } else {
        0
    }
}
#[no_mangle]
#[inline(never)]
pub fn u32_ge_const_u32_or(a: u32,test: u32) -> u32 {
    if a >= 5 {
        a | test
    } else {
        0
    }
}
```

* I would expect:
    * `u32_ge_bool(0,5) == false`
    * `u32_ge_u32_or(0,5) == 0`
    * `u32_ge_const_u32_or(0, 5) == 0`
* `u32_ge_bool` and `u32_ge_const_u32_or` work as expected, but `u32_ge_u32_or` does not. What happens is:
    * `u32_ge_u32_or(0,5) == 5`

I suppose I'll have to lay off the u32 for now, unfortunately (u64 seemed to work fine, but that may just be a gap in my testing)

Here is the objdump of `u32_ge_u32_or`, which may or may not be helpful:
```
0000cef8 <u32_ge_u32_or>:
    cef8:	0b 12       	push	r11		;
    cefa:	0a 12       	push	r10		;
    cefc:	0d 9f       	cmp	r15,	r13	;
    cefe:	0b 42       	mov	r2,	r11	;
    cf00:	1a 43       	mov	#1,	r10	;r3 As==01
    cf02:	0c 9e       	cmp	r14,	r12	;
    cf04:	02 24       	jz	$+6      	;abs 0xcf0a
    cf06:	0a cb       	bic	r11,	r10	;
    cf08:	01 3c       	jmp	$+4      	;abs 0xcf0c

0000cf0a <.LBB9_2>:
    cf0a:	0a c2       	bic	r2,	r10	;

0000cf0c <.LBB9_3>:
    cf0c:	0b 43       	clr	r11		;
    cf0e:	0a 93       	cmp	#0,	r10	;r3 As==00
    cf10:	0a 43       	clr	r10		;
    cf12:	02 20       	jnz	$+6      	;abs 0xcf18
    cf14:	0e dc       	bis	r12,	r14	;
    cf16:	0a 4e       	mov	r14,	r10	;

0000cf18 <.LBB9_5>:
    cf18:	02 20       	jnz	$+6      	;abs 0xcf1e
    cf1a:	0f dd       	bis	r13,	r15	;
    cf1c:	0b 4f       	mov	r15,	r11	;

0000cf1e <.LBB9_7>:
    cf1e:	0c 4a       	mov	r10,	r12	;
    cf20:	0d 4b       	mov	r11,	r13	;
    cf22:	3a 41       	pop	r10		;
    cf24:	3b 41       	pop	r11		;
    cf26:	30 41       	ret			
```

and finally, here is the versions of the compilers:
```
$ (xargo --version && rustc --version && msp430-elf-gcc --version) | head -3
xargo 0.3.6
cargo 0.19.0-nightly (c416fb60b 2017-04-05)
rustc 1.18.0-nightly (1785bca51 2017-04-21)
msp430-elf-gcc (GCC) 6.3.0
```

pftbest, 2017-04-23:
Glad to hear that you got it working.

Yes, you're right, I forgot about the stack pointer. 
I will take a look at this u32 issue.

poisson, 2017-04-23:
Note that the crt0 stuff isn't linked in @pftbest's repo because of passing -nostartfiles in .cargo/config. You may wish to pass -minrt instead, or link explicitly to crt0.o., or implement the crt0 features using [the r0 crate](https://crates.io/crates/r0). If you do link to crt0.o your first Rust function should be called "main" rather than "_start".

awygle, 2017-04-23:
Note that the crt0 stuff isn't linked in @pftbest's repo because of passing -nostartfiles in .cargo/config. You may wish to pass -minrt instead, or link explicitly to crt0.o., or implement the crt0 features using the r0 crate. If you do link to crt0.o your first Rust function should be called "main" rather than "_start".

awygle, 2017-04-24:
I took a look at this again today and noticed the remaining TODO item on this rfc:

`rustc: support the __bic_SR_register_on_exit gcc built-in somehow. Needed to enter/leave low-power mode.`

@pftbest said 

> I think it is not possible to write llvm IR that will do this kind of thing. So this function should be added as intrinsic in llvm (and as built-in in clang). and only then we can have it in rust.

That's not quite true - one can actually implement `__bic_SR_register_on_exit` and `__bis_SR_register_on_exit` using the LLVM intrinsic @llvm.frameaddress. An example of this for the bit-set variant:

```
target datalayout = "e-p:16:16:16-i1:8:8-i8:8:8-i16:16:16-i32:16:32"
target triple = "msp430-generic-generic"

define msp430_intrcc void @funfun() #0 {
  %i = alloca i32, align 2
  %j = alloca i32, align 2
  store i32 0, i32* %i, align 2
  store i32 7, i32* %j, align 2

  %fa = call i8* @llvm.frameaddress(i32 0)
  %sra = bitcast i8* %fa to i16*
  %sr = load i16, i16* %sra
  %nsr = or i16 %sr, 1
  store i16 %nsr, i16* %sra

  ret void
}

declare i8* @llvm.frameaddress(i32)
```

Which happily compiles to:
```
    .text
    .file   "sr.ll"
    .globl  funfun
    .p2align    2
    .type   funfun,@function
funfun:                                 ; @funfun
; BB#0:
    push.w  r4
    mov.w   r1, r4
    sub.w   #8, r1
    mov.w   #0, -2(r4)
    mov.w   #0, -4(r4)
    mov.w   #0, -6(r4)
    mov.w   #7, -8(r4)
    bis.w   #1, 0(r4)
    add.w   #8, r1
    pop.w   r4
    reti
.Lfunc_end0:
    .size   funfun, .Lfunc_end0-funfun

    .section    ".note.GNU-stack","",@progbits
```
And is modeled after this C code:
```
__attribute__((interrupt)) void funfun() {
    long a1 = 0;
    long a2 = 7;
    __bis_SR_register_on_exit(1);
}
```
Which GCC compiles as:
```
    .file   "sr.c"
.text
    .balign 2
    .global funfun
    .type   funfun, @function
funfun:
; start of function
; attributes: interrupt
; framesize_regs:     0
; framesize_locals:   8
; framesize_outgoing: 0
; framesize:          8
; elim ap -> fp       2
; elim fp -> sp       8
; saved regs:(none)
    ; start of prologue
    SUB.W   #8, R1
    ; end of prologue
    MOV.W   #0, 4(R1)
    MOV.W   #0, 6(R1)
    MOV.W   #7, @R1
    MOV.W   #0, 2(R1)
    BIS.W   #1, 8(SP)
    NOP
    ; start of epilogue
    ADD.W   #8, R1
    RETI
    .size   funfun, .-funfun
    .ident  "GCC: (crosstool-NG crosstool-ng-1.23.0-rc1) 6.3.0"
```
So anything that causes rustc to emit this block:
```
  %fa = call i8* @llvm.frameaddress(i32 0)
  %sra = bitcast i8* %fa to i16*
  %sr = load i16, i16* %sra
  %nsr = or i16 %sr, $THE_BITS_TO_SET
  store i16 %nsr, i16* %sra
```
 will suffice to implement the above-mentioned intrinsic (and the clear version is of course obvious). The downside is that currently the backend forces the use of a frame pointer whenever frameaddress is used, which wastes a register and some attendant manipulation. I am pretty sure I know how to improve this, just playing around with it in upstream LLVM right now, but ultimately it's irrelevant to Rust anyway.

pftbest, 2017-04-24:
@akovaski
I've traced the codegen for `u32_ge_u32_or` and found that CSE is incorrectly removing CMP instruction. This can be either a bug in MachineCSE or something is wrong with CMP instruction.
I have submitted a bug in llvm bugzilla https://bugs.llvm.org/show_bug.cgi?id=32769

@awygle Maybe you know more about machine instructions?

eddyb, 2017-04-27:
FWIW you can import the `llvm.frameaddress` symbol into Rust, if you want to do that in a library, and you *could also* just fake a platform intrinsic that `librustc_trans` codegens.

pftbest, 2017-04-27:
@eddyb, Is there any other target that has intrinsics in a language? If not, I don't want to be the one who opened that pandora box.

> you can import the llvm.frameaddress symbol
How can I import a name with a dot in Rust?

eddyb, 2017-04-27:
@pftbest `#[link_name = "llvm.frameaddress"]` on the `fn` declaration. This works for both imports and exports, i.e. you don't *need* to use `#[no_mangle]` and you can do more.

pftbest, 2017-04-27:
@eddyb, ok, I'll try to import this symbol, thank you.

eddyb, 2017-04-27:
>  Is there any other target that has intrinsics in a language?

I think this is not relevant. We implement our intrinsics in the compiler *and* the common platform intrinsics (the ones that start with `simd` instead of `x86` or `arm` etc.).

I wouldn't hold a custom intrinsic implementation against its addition *as long as there was no plan for stabilizing it*. I think these intrinsics are misguided in how they modify stack memory and this functionality should've been designed *into the interrupt ABI*.

awygle, 2017-05-05:
I'm posting a slightly modified version of my MSP430 TODO list from [this reddit thread](https://www.reddit.com/r/rust/comments/687fhh/status_of_msp430_support_in_rust/) here in case somebody who comes by wants to work on one of them or has a strong opinion about priority.

- [x] ~~Convert the LLVM backend to generate libcalls which comply with the EABI from TI for things like multiplication and division so that we can link in libgcc and do useful stuff~~ patch submitted, awaiting review
- [ ] Add EABI-compatible definitions to compiler-rt (eventually also to compiler-builtins?)
- [ ] Add assembler support to the MSP430 backend
- [ ] Add support for MSP430 to lld
- [ ] Add support in rustc and possibly LLVM for some of the useful/tricky intrinsics, such as __bic_SR_on_exit as mentioned above (or equivalent functionality)
- [ ] Whip up a quick crt0 implementation based on the r0 crate and the startup code given by TI
- [ ] Add support in LLVM for the MSP430X instructions
- [ ] Add support in LLVM and eventually rustc for the 20-bit pointers of the "large memory model"

My rough plan was to work on assembler support next, but if somebody would like something else I can certainly change that ordering.

japaric, 2017-05-17:
@awygle @pftbest how do you feel about merging the existing msp430(x) target(s) in rustc proper so that you don't need to carry around a custom target specification file. I think the target definiton you have been using hasn't changed in months.

pftbest, 2017-05-17:
Well, I think the only option that can change in the near future is `no-integrated-as`  (and when it does, `asm-args` will no longer be needed).
Also there is an open question what do we do with brand new `mhwmult` option.

pftbest, 2017-05-17:
@japaric, Can rust use llvm feature flags? Or is there any other preferred way to handle codegen options? So the problem is that some micros have hardware multiplier and some don't, and also there is 2 types of multipliers.

awygle, 2017-05-17:
Yes, I am working on adding integrated assembler support, though I have no immediate ETA. We should still support external assemblers though - this can be overridden, right?

Let me make sure I understand the proposed targets. There are four sets of flags - rustc, llvm, assembler, and linker. We can obviously control rustc flags, and linker flags are said (above) to be able to be modified with RUSTFLAGS. Assembler flags are in the target.json, and I don't know about llvm (llc) flags.

The flags for MSP430s are `cpu` (`msp430`, `msp430x`, technically msp430xv2 but we can probably treat all x as xv2), `hwmult` (`none`, `16bit`, `32bit`, and `f5series`), and `mcu` (every micro ever). These flags are needed at the following times:

- cpu - codegen (llvm) and assembly (just for verification that X instructions aren't used in vanilla mode)
- hwmult - codegen (llvm) and link
- mcu - link only

If we can pass llvm options outside the target.json but not assembly, we only need 2 targets. Otherwise, we need 8, which I think is too many.

Also - what is the expectation for targets in the compiler in terms of functionality/stability? For example, I wouldn't trust the extended instructions as far as I could throw them right now (assuming they can be generated at all, which I doubt). Does that mean we should only add the vanilla target?

japaric, 2017-05-17:
> Well, I think the only option that can change in the near future is no-integrated-as

The scenario in that case would be: LLVM learns to emit object files and we no longer need to use an external assembler. rustc would directly generate object files and still depend on an external linker but the asm-args flag could be dropped. The target specification would be updated to change no-integrated-as to true, drop the asm-args and to emit msp430 (not msp430x) object files.

> Can rust use llvm feature flags?

There's `-C llvm-args`. The preferred way is to only use `-C target-cpu` and `-C target-feature` for codegen.

> we only need 2 targets. Otherwise, we need 8

In my mind we only need as many targets as the number of ABIs that exist. Here by ABI I mean something like ARM's soft float vs hard float ABIs where you can't link two objects unless they have the same ABI. My main concern here is being able to link to TI's startup objects as that's required for all programs (AFAIK). IIRC, there are two ABIs in that sense: msp430 and msp430x and that's why we had to add the no-integrated-as option: the external assembler produced msp430x code by default and that couldn't be linked to the startup object which used the msp430 ABI; the no-integrated-as let us change the ABI of the Rust side to msp430.

awygle, 2017-05-17:
> In my mind we only need as many targets as the number of ABIs that exist. Here by ABI I mean something like ARM's soft float vs hard float ABIs where you can't link two objects unless they have the [...]

There are several issues here. First, the CPU core. TI produces two different CPU cores under the larger MSP430 "umbrella" brand. msp430 cores have 16-bit registers and can access a 64k address space. msp430x cores have 20-bit registers (mostly) and can access a 1M address space. msp430x cores also support an extended ISA which includes instructions for manipulating 20-bit registers and data values as well as some useful utility instructions like multi-push and multi-pop. libgcc builds two different versions, one for msp430x and one for msp430, the difference presumably being that the msp430x one uses the more-efficient extended instructions.

Second, the hardware multiplier. Depending on the hwmult option passed to LLVM, it will emit different library calls for integer multiplication as specified in [the EABI](http://www.ti.com/general/docs/lit/getliterature.tsp?baseLiteratureNumber=slaa534). These functions are defined in libgcc (eventually also compiler-rt), but in separate .a files (libmul_16.a, libmul_32.a, etc). So everything uses the same ABI, calling convention, ISA, etc., but different library calls are generated and linked.

Then there's the memory models, large and small. This has to do with whether address spaces >64k are supported, and whether pointers are 16-bit or 20-bit. libgcc also builds a large-model version of itself for linking against large memory model programs. Since only msp430x CPUs support 20-bit pointers, this makes 3 versions of libgcc (and libmul_*) - small/430, small/430x, large/430x.

Also, there *used* to be two different MSP430 ABIs - one was in the old mspgcc, a heavily patched fork of GCC 4, while the newer one is in TI's CCS, IAR's Embedded Workbench, and upstream GCC. LLVM used to use the mspgcc ABI, but I've since patched it to use the newer ABI, since mspgcc is effectively dead. So this can be ignored now.

Finally, the startup object (crt0.o) is built as part of newlib usually, and has a large and small version but no differences based on CPU type. So just large and small.

You can link msp430 code to msp430x code as long as it's run on an msp430x CPU. You can't link large and small objects together (you might be able to but it's a nightmare). You can link to either the hwmult library you have or the software version (libmul_none.a), but you can't link to a hwmult you don't have.

Total, this gives (small/large/430)*(none/16/32/f5) is 12 configurations, eventually. The default configuration is small/none. The chip used by @pftbest in his repo is an msp430 CPU, so it needed -mcpu=msp430 in order to link to the proper libraries (msp430/none).

Personally I don't see why we can't have just one target and specify all the other stuff on the command line. Currently there's only one target triple in both LLVM and GCC.

pftbest, 2017-05-17:
> You can link msp430 code to msp430x code as long as it's run on an msp430x CPU. 

In theory this is possible, but the linker will not allow it:
```
/Users/vadzim/.msp430/bin/../lib/gcc/msp430-elf/6.2.1/../../../../msp430-elf/bin/ld: error: foo.o uses MSP430 instructions but /Users/vadzim/.msp430/bin/../lib/gcc/msp430-elf/6.2.1/../../../../msp430-elf/lib/crt0.o uses MSP430X
/Users/vadzim/.msp430/bin/../lib/gcc/msp430-elf/6.2.1/../../../../msp430-elf/bin/ld: failed to merge target specific data of file foo.o
collect2: error: ld returned 1 exit status
```
So they must be a separate targets.

Multiplier option on the other hand is not so strict, for example, gcc will happily link two object files, where one was compiled with `-mhwmult=32bit` and the other with `-mhwmult=none`. So for this option we probably don't need to create separate targets.

What can be done about small/large memory models I'm not sure yet.

awygle, 2017-05-17:
> In theory this is possible but the linker will not allow it

You're right, I don't know why I didn't notice the 430/ folder for crt0. Curious that they chose to do this but I suppose it would almost always be an error.

pftbest, 2017-05-17:
I think we would need only 3 targets in the end:

- msp430
- msp430x-small
- msp430x-large

But right now we only have the first one working, so if we are going to merge something, I think we should only do `msp430`.

[13 more replies]

closed 2018-08-06 by japaric