rust-embedded/wg #28: Performance tracking tool for ARM Cortex-M microcontrollers

japaric, 2017-06-03:
In my [last blog post] I talked about the Instrumentation Trace Macrocell (ITM)
peripheral as a mean to log data and/or debug messages from a microcontroller to
a PC or some other host. However the ITM can be used to trace many other things
like:

[last blog post]: http://blog.japaric.io

- It can sample the Program Counter which can be mapped back to source code to
  produce flame graphs (where in the program is the processor spending most of
  its time?).

- It can trace entries to and exits from exceptions and interrupts which you
  could use to count context switches and measure the time spent in interrupt
  context.

- It can trace memory accesses to arbitrary (?) memory regions.

All this with timestamps accurate to the clock cycle.

I'm opening this issue to see if there are people interested in building a cross
platform Rust tool that can collect this performance information and present in
some way (text or graphs; it doesn't matter much how it's presented in the first
versions) as I'm not sure how much time I can put into this myself.

Here's more or less what needs to be done:

There are two aspects to this tool: the register manipulation necessary on
the device side, and the frame decoding and processing on the host side. So

- [ ] First thing is to collect information about the ITM registers and the ITM
  frames related to the tracing functionality.

- [ ] Extend the ITM API on the [`cortex-m`] crate to deal with setting up the
  tracing stuff.

[`cortex-m`]: https://crates.io/crates/cortex-m

- [ ] Extend the [`itm`] crate to handle parsing of ITM frames related to
  tracing.

[`itm`]: https://crates.io/crates/itm

- [ ] Build a basic tool that grabs the ITM frames from a device (e.g.
  /dev/ttyUSB0) or a file (itm.dump) and just prints the decoded frames to
  stdout. This both meant as a sanity check and to get an idea of the
  performance requirements of the tool (can it handle the data throughput?)

- [ ] Start adding functionality like context switch counting, interrupt time
  measurement, and flame graphs (or just % time spent vs PC address to begin
  with)

- [ ] Figure how to feed debug info into the tool to get flame graphs with
  proper line information (instead of just PC addresses). This sounds tricky to
  me as I know nothing about using debuginfo without a debugger.

- [ ] :tada:

whitequark, 2017-06-03:
> Figure how to feed debug info into the tool to get flame graphs with
> proper line information (instead of just PC addresses). This sounds tricky to
> me as I know nothing about using debuginfo without a debugger.

Thankfully this is utterly trivial. Run `arm-none-gnueabi-addr2line -Cfi -e <binary>`, feed it addresses on stdin, get decoded info on stdout.

fluffysquirrels, 2017-12-18:
I started a debug tooling [thread](https://users.rust-lang.org/t/arm-embedded-tooling-feature-discussion-on-itm-crate/14486/3) on users.rust-lang before finding this RFC.

There I wrote:

> On the microcontroller we probably want Rust support for:
> 
> * TPIU configuration (multiplexes and encodes debug data)
> * ITM (transfers application data)
> * ETM (instruction-level tracing)
> * DWT (profiling, breakpoints, watchpoints)
> 
> I see TPIU, ITM, DWT code in the `cortex-m` crate. I'll try to add anything appropriate I come up with there. I think ETM code should live there too. Perhaps it should be behind a feature flag, as the ARM v7-M architecture reference manual indicates ETM is optional.
> 
> For profiling we would probably configure the peripherals with Rust, but then other tools probably do most/all of the host-side work. `orbuculum`'s `orbstat` converts PC samples for [KCachegrind](https://kcachegrind.github.io/html/Home.html) ([screenshots](https://kcachegrind.github.io/html/Screenshots.html)), which has a source code view annotated with time per line and call tree graph. Converting lines of PC samples for other tools should be very easy, including [FlameGraph](https://github.com/brendangregg/FlameGraph).

paoloteti, 2018-02-27:
Lautherbach Trace32 is the de-facto tool **everybody** in the professional world use. Yes it is really expensive, I agree. Here an example of T32 on a Safety Critial ARM Cortex-4RF CPU in Lock-Step mode I use daily in lab. and it work seamless in Rust. (Perf Trace, Power Cons., Event, PMUs counters)
Collecting ITM data or similar, without hw support (JTAG, ICD with up to 5GB of trace mem. and logic analyzer channels) it is not useful. It is just wrong data.
[image]
 
**Disclaimer: I do not work for Lautherbach.**
There are few other professional tool (one from ARM) that are similar to T32.

Chip like XDS100, that you can find in low-cost boards, are low performance chip. Just to sell an evaluation boards. 
At the end what I want to say it that "tracing without assistance of a dedicated hw (JTAG/ICD) it is really hard or quite impossible".

korken89, 2019-02-05:
Awesome!

jamesmunns, 2024-06-11:
Closing this as a part of the 2024 triage.

As of now, the charter of the WG is to focus on foundational crates and functionality, rather than develop solutions for all use cases. However, as of today there are the following existing solutions, and folks finding this issue are suggested to contribute to one of the following projects:

* itm-tools (linked above, though I'm not sure if this is still active)
* `probe-rs`'s "profile" command: https://github.com/probe-rs/probe-rs/blob/master/probe-rs-tools/src/bin/probe-rs/cmd/profile.rs

If you think this was closed incorrectly, please leave a comment and we can revisit this decision in the next weekly WG meeting on Matrix.

closed 2024-06-11 by jamesmunns