rust-embedded/wg #34: Embedded.fm Wednesdays in Rust

enricostano, 2017-08-14:
I would like to learn embedded programming and Rust and I thought that maybe I could follow this blog post serie: http://embedded.fm/blog/embedded-wednesdays

Hopefully taking notes and sharing my findings in the process.

**Which microprocessor / development board should I try**? In the serie they use STM32F401C discovery board but I don't think Rust support it. Please correct me if I'm wrong.

Please give a look at the end of this blog post http://embedded.fm/blog/2016/3/22/embedded-tools

What do you think? Thanks!

alevy, 2017-08-14:
That board is definitely programmable in Rust! The microcontroller is Cortex-M4.

My guess is you ought to be able to follow the embedded.fm blog to the letter, but just using Rust instead of C --- the compile toolchain will be different, as will the syntax for doing things like reading/writing hardware registers, but those are all things folks are happy to help with on the #rust-embedded IRC. If your goal is to learn embedded programming, my recommendation is to do that. It'll take you longer to get an LED blinking, but it's worth the steeper learning curve to get more of the fundamentals. It's also very very rewarding!

@japaric has a good guide on how to get started with a similar discovery board: https://japaric.github.io/discovery/

The pinouts and some of the hardware might be different, but the basic ideas should be right.

On Twitter, Steve Klabnik pointed to Tock (I contribute to Tock), which is more of an operating system than a bare-metal programming framework. It may or may not be what your looking for (the embedded.fm blog post series is less easily translatable to Tock than to something like @japaric's guides). So, YMMV, though i of course personally recommend it :stuck_out_tongue:.

niklasad1, 2017-08-14:
I would consider to use [nRF52-DK](https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF52-DK) if I were you. The kit comes with radio, LEDs, buttons, temperature sensor etc.

1. If you want to build stuff from bare-metal you can start out from [https://github.com/jamesmunns/nrf52dk-sys](https://github.com/jamesmunns/nrf52dk-sys) more stuff is in that repo but you can perhaps start out with the blinky example and then start to implement your own stuff.
2. nRF52-DK is also supported by [TockOS](https://github.com/helena-project/tock) if you want to use it underlying operating system.

However, STM32F401C will probably work out fine as well 😃

closed 2017-12-19 by enricostano