rust-embedded/wg #37: Question: Best practices - semantics of init functions

dvc94ch, 2017-09-12:
Do people expect a Device.init() function to be from (reset state -> device initialized) or from (unknown state -> device initialized)?

thejpster, 2017-09-12:
Given you might have a bootloader fiddle with state before you run, I'd assume the state was unknown.

dvc94ch, 2017-09-12:
A related question would be should a device driver keep it's state - making sure for example that it was initialized before use? With simple devices it's only a matter of initialized vs uninitialized, but I imagine that for more complex devices there are more states. This is (since rust doesn't support dependent types (types can't be changed at runtime)) not a zero cost abstraction. I see that the cortex-m device drivers (blue-pill) get by without internal state, but to implement the embedded hal traits for the e310, state needs to be embedded anyway since the peripherals are extremely minimal. For example the Timer trait has a pause and a resume, this is only possible by remembering the counter value on pause, disabling the interrupt and resetting the counter value on resume.

thejpster, 2017-09-12:
Yes, it send reasonable to me for a driver to track peripheral state. The trick, I guess, is to ensure drivers are singletons (or rather, there's only one driver per peripheral).

closed 2017-11-14 by dvc94ch