rust-embedded/wg #38: Question: Hardware designed to run rust (bikeshed)

dvc94ch, 2017-09-12:
GPIO peripheral registers have the structure register{pin0: bool, pin1: bool ...}. If one where designing a GPIO IP with rust in mind, would one prefer reg_pin0 {input_en: bool, ...}?

The reason why I'd think it's a good idea is because pins could be owned by device drivers. Downside may be requires more gates because of unused bits. What are your opinions?

thejpster, 2017-09-12:
So far I've only implemented the C model where pins are enums and there's no ownership, which is sub-optimal.

whitequark, 2017-09-12:
> Downside may be requires more gates because of unused bits.

Unused bits are just missing from the register, they don't take up any FFs.

In any case, hardware designed to run Rust should likely feature a capability-based access control system, which can represent ownership directly.

closed 2017-11-14 by dvc94ch