rust-embedded/wg #41: Panic handler binary size bloat

japaric, 2018-02-21:
There are two issues here. The first one I consider a (compiler) bug and the second one a missing
feature.

### Panic strings kept in the binary

Even when `panic_fmt` (the panic handler) is defined to simply `abort` the compiler still produces
machine code that passes the (unused) arguments (panic message, file name, line number and column
number) to that function at runtime. This causes the arguments to be stored in the program which
takes up a sizable amount of non volatile memory (.rodata).

@nagisa was looking into this and they may have found the root of the problem. They'll open an issue
in the rust-lang/rust repo with more details.

This is also a problem on WASM land; see rust-lang-nursery/rust-wasm#19. If this can't be fixed in
the compiler then this RFC (rust-lang/rfcs#2305) for optionally removing the arguments of
`panic_fmt` could be a solution.

### Reduce footprint of panic messages

When `panic_fmt` is defined to print / log panic messages complete strings are stored in the binary.
This takes up precious space in non volatile memory (.rodata).

It would be great to be able to store the strings, along with file position data, in a table in
debuginfo, which is not flashed into the device, and store only the address to each table entry
in the binary. A external tool would be required to turn that address back into a panic location.

A related RFC has been proposed: rust-lang/rfcs#2154

cc @cr1901 @pftbest

nagisa, 2018-02-21:
I [wrote down my ideas](https://github.com/rust-lang/rust/issues/44489#issuecomment-367301641) on how we should fix the underlying issue. I still am going to write a separate issue for the underlying cause.

cr1901, 2018-02-21:
Relevant commit for my use case: https://github.com/cr1901/AT2XT/commit/b6e299208dd90cf8ec1bb054ebfbaa87797eb93a

Notably, embedding unused strings does _not_ happen with `Result`, only `Option`.

fitzgen, 2018-02-21:
See also https://github.com/rust-lang-nursery/rust-wasm/issues/19

japaric, 2018-04-03:
Update: we discussed this during Rust All Hands. Here's a summary by @nagisa (originally posted in https://github.com/rust-lang/rust/issues/44489#issuecomment-378058031)

> We did discuss various approaches to handling the issue of panic size, and figured out that we’d love to have something that works in all scenarios such as non-lto builds and even debug builds. We didn’t arrive at any particular design, but we managed to reject a number of designs that wouldn’t work and also realised that the only real way to properly implement any design would rely on MIR-only rlibs.
>
> With that in mind, this design seems to be as good as any other and, as long as we stabilise `#[panic_implementation]` carefully (i.e. the signature and behaviour, but not the exact expansion), we should be in a good place to implement a solution for panic sizes as well.

Where "this design" = RFC 2070

therealprof, 2018-04-03:
I'm hopeful this can be sorted out. Panic handling is a bit painful at the moment.

johnthagen, 2019-02-05:
Related: [`panic_immediate_abort` libstd feature](https://github.com/rust-lang/rust/pull/55011).

eldruin, 2020-11-25:
Is there a solution for this now that RFC-2070 has been closed?

Dirbaio, 2024-06-11:
`panic-immediate-abort` is the solution in nightly. See https://github.com/rust-embedded/wg/issues/551#issuecomment-821477413.

No solution in stable yet.