rust-embedded/wg #45: Multitasking models

japaric, 2018-02-21:
Unlike Linux / Windows / *BSD / etc. targets where the OS provides threading support,
there exist no standard way to do multitasking on embedded targets. This means that we
either have to implement a multitasking mechanism ourselves or bind to a C implementation.
There are few different ways to do multitasking:

- [Real Time for The Masses].

[Real Time for The Masses]: https://docs.rs/cortex-m-rtfm/0.3.1/cortex_m_rtfm/

Preemptive multitasking with support for task prioritization. Hardware scheduler with zero software
(bookkeeping) overhead. Dead lock free execution plus all the memory safety that Rust grants.

@japaric is working on / maintaining the Cortex-M port.

- Tokio-like cooperative event loop.

Basically, cooperative multitasking that avoids the "all wakeups must poll all events" problem.

There's an implementation for the STM32F4 [here] but I haven't look at it in detail. I think
that whatever implementation we end up adopting should meet these requirements:

[here]: https://github.com/andreytkachenko/async-hal-stm32

- Should integrate with the device sleep functionality.
- It should be easy to use it with `embedded-hal` drivers.
- It should not require dynamic memory allocation, or features that require memory allocation should
  be opt-in.
- It should be possible to integrate this event loop with preemptive multitasking models like RTFM
  and the thread model.

@japaric is looking into a Cortex-M implementation of this and will write about his findings in a
blog post.

- (RTOS) Threads.

A lot of embedded programmers are used to thread programming so this multitasking model should be
well supported. Here we have the choice to implement threading support in Rust or to bind an
existing RTOS / C implementation.

If we go the Rust route I think [Chromium OS Embedded Controller][ec] implementation (MIT license)
could be used as a reference.

[ec]: https://chromium.googlesource.com/chromiumos/platform/ec

If we go the C route perhaps we could bind Amazon's [FreeRTOS]. I think that at some point I saw
some FreeRTOS bindings but that was before Amazon bought it.

[FreeRTOS]: https://aws.amazon.com/freertos/

Maybe @thejpster can recommend some RTOS to bind?

@japaric won't work on this due to conflict of interests (he's already working on RTFM)

---

Ideally, all these models should have ports for the all the embedded targets. Also, the
`embedded-hal` drivers should play nicely will all these multitasking models (without
duplicating a bunch of code).

thejpster, 2018-02-21:
I think there are two pieces to the RTOS story - the API and the implementation(s). The first implementation might be a FreeRTOS wrapper but I'd love to write a Rust port of it in due course.

The API would involve:

* Static and dynamic queues
* Static and dynamic threads
* Static and dynamic event groups (bitfields you can pend on)
* Static and dynamic mutexes (although in my world we really try and avoid these)
* Static and dynamic timers (these post messages or flag events on expiry)
* High level ISRs (pseudo-ISRs that are scheduled like threads and so have access to the full API)
* Hardware ISR safe subset API (mainly triggering HISRs)

jcsoo, 2018-02-21:
In this context, can you go further into what you mean by static and dynamic? Do you mean static creation before starting up tasks vs. creating / destroying these objects within tasks?

Looking at your list, I wonder how far it is possible to get using a SPSC queue supporting zero-sized objects as the main communications primitive.

thejpster, 2018-02-21:
In brief, yes. Sometimes you know the resource requirements at compile time - six threads, six queues, one event group, etc. Allocating static space means you don't take up heap/pools. Sometimes you also want to spin up background threads, or bring up a new queue for a test that you later want to delete. It's good to support both models.

I imagine the former would use an RTFM style macro.

adamgreig, 2018-02-21:
The CMSIS-RTOS ([v1](http://www.keil.com/pack/doc/cmsis/RTOS/html/index.html), [v2](http://www.keil.com/pack/doc/cmsis/RTOS2/html/index.html)) APIs might be a useful reference too. In theory a wide range of RTOSs implement them, so Rust bindings to that API would allow you to plug in multiple RTOSs under the hood. This includes FreeRTOS: https://github.com/ARM-software/CMSIS-FreeRTOS

In general the CMSIS RTOS APIs provide access to all the common and standard RTOS stuff, but many RTOSs have a few extra things on top that might require additional/special-cased bindings.

Another good and open-source (though GPL only) RTOS for Cortex-M devices is [ChibiOS](http://www.chibios.org/dokuwiki/doku.php) which I'd love to see/write Rust bindings for!

ryankurte, 2018-02-21:
Really like the idea of a `rust-embedded/rtos-api` similar to cmsis-rtos that rtos implementations or bindings can implement. I guess a bit like `embedded-hal`, if we have a set of primitive traits that RTOS implement it should be pretty easy to build "thread" safe primitives on top of that API.

It seems to be falling out of favour but just to chime in, I think predictability is critical if rust is going to be taken seriously for embedded, and hopefully that could be reflected in these APIs. Static allocation, Mutexes and primitives that work from ISRs, NO runtime exceptions etc. (which are all _features_ of most of the modern/iot rtos's). One of the reasons we struggle to use c++ is that while it's possible to write safe embedded code in c++, it's not obvious / a chore to get it right / remember what's ok and what's not.

japaric, 2018-02-27:
There seems to be agreement on the direction of a standardized RTOS API / traits. Could someone make a work item issue about that? I expect the issue would also include the task of making a reference implementation by binding some C RTOS.

jamesmunns, 2018-02-27:
If we select a target RTOS to bind against (seems like FreeRTOS or cmsis-rtos are the frontrunners), I would be happy to apply the same techniques I used with https://github.com/jamesmunns/nrf52dk-sys.

A word of warning: Most C based project rely HEAVILY on compile time macro settings to change configuration settings, and Cargo/Rust does not really have the same kind of mechanisms to deal with this, as the `#[cfg()]` flags typically operate in a union style format. I will mention this in #48, since I think it is not specific to RTOS's.

japaric, 2018-02-28:
I saw this FreeRTOS [bindings](http://www.hashmismatch.net/libraries/freertos_rs/) on reddit today. I don't know how it maps to what you folks have in mind.

Also, I have turned this issue into an RTOS-only discussion since that's the multitasking model that has been discussed so far.

jamesmunns, 2018-04-03:
@japaric This topic is going to be covered in https://github.com/rust-lang-nursery/embedded-wg/issues/56 - We may want to close this issue as a duplicate

memoryruins, 2018-05-25:
Discovered drone today:
https://github.com/drone-os
https://docs.rs/drone/0.6.0/drone/

seanybaggins, 2020-08-04:
Hi. Has there been any progress on this? It has been a while since a post in this issue.

frank-amy, 2023-05-31:
is there some graceful ways to comm with threads by rusty, in RTOS such freertos, threadx
I only have some ideas by FFI, based on RTOS C apis, maybe messagequeue

newAM, 2023-05-31:
It depends on the RTOS of course, but most of the native rust RTOS' have similar mechanisms to ThreadX/FreeRTOS.  Maybe take a look at `embasssy_sync` for some inspiration?

adamgreig, 2024-06-11:
Closing as part of today's issue triage. A number of RTOSs have Rust bindings (see https://arewertosyet.com/), but in the years since this issue was created it's turned out that alternative frameworks such as RTIC and Embassy are also viable for embedded Rust and have become very popular.

In particular, I don't think we need standard RTOS traits now - some RTOSs can implement std (Espressif do this, for instance), which is probably the best set of common interfaces imaginable, and for Embedded-specific things, embedded-hal has become popular and widely used.

closed 2024-06-11 by adamgreig