rust-embedded/wg #49: Material for the revamped website

japaric, 2018-02-22:
A revamped Rust website will be released with the Epoch 2018 in mid September. The four domain WG
will have a space in the website to showcase their accomplishments and make the case for Rust in
their domain.

Some ideas that were discussed in the first meeting.

### Who's our audience?

- Early adopters: students, hobbyists, researchers, startups
  - Not a lot of existing organizational momentum
  - Compatibility / portability isn't too important, as long as it works for them
  - No / little expectation of technical support
  - Free (as in beer) matters
  - Community matters, good chance they will join and contribute

- Regular firmware developers
  - Expect a mostly polished environment
    - e.g. vendor SDKs / IDEs
  - Willing to pay $$$$ for technical support / SDK / IDE / tooling
    - Or free (as in beer) is not that important
  - Organizational momentum
    - Established development workflows / programming patterns
  - Important to be able to easily port code between processor lines or even families. Examples
    - STMF1xx to STMF3xx (need more CPU/RAM/Flash than initially thought)
    - STMF3xx to STMF1xx (need less than initially needed, cost optimization, dev board to product transition)
    - STMF1xx to KL2x (different peripherals/uC features needed, cost savings, etc)
  - Important to be able to reuse code across projects
  - Community may or may not be important, unlikely to contribute back though

- Safety critical market: automotive, aviation, etc.
  - Requires certified tooling (compiler / language) which costs $$$$$$ to certify
  - Regulatory overhead
    - Must convince safety orgs that Rust is "fit for purpose", not just managers

There was consensus on focusing on the first group in the short term, on the second group in the
medium term and on the last group in the long term.

### What do we want to highlight?

- Tooling: Cargo, package manager; RLS, IDE support; debugging Just Works, etc.

- Zero cost abstractions: we are competing against C, not against MicroPython, eLua, Espruino, etc.

- Type system preventing memory safety / logic bugs

- Reusable code: crates.io, `embedded-hal` drivers

- One step / easy setup

- Consistent development experience across different architectures cf. #44

### What should we show on the website?

- Hardware demos

- Getting started guide that walks you through the development of an embedded application using an IDE

- Cargo project templates

### TODO

- @japaric will ask the Rust team what kind of content can go on the website. Blog posts? videos? searchable crate catalogs? guides like [tokio's]?

[tokio's]: https://tokio.rs/docs/getting-started/hello-world/

posborne, 2018-02-22:
Under "what do we want to highlight?" I would definitely add bidirectional C interop.  Few places are going to want to go "whole hog" on Rust for an embedded product but might be willing to implement a module in Rust (with C as the "master") or use Rust as the "master" but continue to make use of existing things like drivers they have written in C.

japaric, 2018-07-03:
Belated update: During the Rust All Hands we talked with the website team and drafted the embedded WG webpage. This is the content that we agreed upon (remember that this single page is basically a marketing sheet that answers the question "why Rust for embedded development?"):

- Videos of embedded Rust projects (carousel)
- "Attributes" that answer the question "why Rust for embedded development?". e.g. Portability,
  interop, etc. (grid or three column thing with icons)
- Links to documentation: embedded Rust book, embedonomicon, etc.
- Commercial testimonials

Summary of what we briefly discussed in the last meeting:

We have issue #108 for collecting commercial testimonials. Videos will come from commercial users and from popular hobby projects like the [monotron]. Work on books is tracked elsewhere (#56 and #59).

[monotron]: https://github.com/thejpster/monotron

japaric, 2018-07-27:
Initial layout and content was submitted. The page has been integrated into the new website and styling was applied by the website team. The revamped website is not public yet so I can't share a picture with you at this time but it already looks nice, IMO.

This issue is done! Next stage is to iterate on the submitted content. #131 will track that work.

closed 2018-07-27 by japaric