rust-embedded/wg #50: Ship llvm binutils with the Rust toolchain

japaric, 2018-02-23:
As per the discussion in #43, it would be convenient to have LLVM tools like `llvm-objdump` and
`llvm-size` shipped with the Rust toolchain. These would remove the need to have to install GCC
binutils which require you to do one binutils install *per target architecture*. OTOH, LLVM binutils
have multi-arch support and support all (most?) the architectures that rustc supports (since rustc
is LLVM based).

### Tasks

We are looking for someone to take on this task:

- [ ] Submit an RFC to ship llvm binutils with the Rust toolchain. @japaric can help with the
  edition of the RFC.

The RFC should make the case for having these tools shipped with the Rust toolchain. `llvm-objdump`,
`llvm-nm` and `llvm-size` are pretty useful to diagnose problems and to measure / estimate metrics
like binary size and performance. `llvm-objcopy` can be used to translate ELF files (what rustc
usually outputs) into binary files (what program loaders expect). `llvm-cov` nicely complements
rustc profiling support (`-Zprofile`).

@whitequark is `llvm-cxxfilt` actually needed? It shouldn't contain architecture specific code,
right? Also, there's a [Rust implementation] of `c++filt`.

[Rust implementation]: https://crates.io/crates/rustfilt

The RFC should note that note as a drawback that does this makes us more reliant on LLVM but we are
already shipping lld (a much bigger LLVM tool) with the Rust toolchain as of rust-lang/rust#48125
(we should submit the RFC after that PR lands).

whitequark, 2018-02-23:
> Also, there's a Rust implementation of c++filt.

Rustfilt is not an implementation of c++filt, it takes Rust symbols and not C++ symbols; the former are a tiny subset of the latter. But I suppose if you're compiling C++ code then you already have clang, and that probably already comes with llvm-cxxfilt, so we don't  need it.

japaric, 2018-02-28:
Adding to the epoch 2018 milestone as discussed in yesterday's meeting

japaric, 2018-04-02:
I have opened rust-lang/rust#49584 to discuss this with the dev-tools and other Rust teams.