rust-embedded/wg #51: `cargo-objdump`, `cargo-size`, etc.

japaric, 2018-02-23:
Once #50 is accepted / implemented it would be nice to have wrapper over e.g. `llvm-objdump` for
convenience. The wrapper would take care of passing `-triple=$target` to `llvm-objdump`, demangling
the output, passing the path to the binary to `llvm-objdump`, etc.

We don't have to wait until #50 is implemented though; we can start prototyping on top of
`arm-none-eabi-objdump`.

### Tasks

- [ ] Define what kind of user interface we want.

- [ ] Prototype `cargo-objdump`, `cargo-size`, etc. on top of `arm-none-eabi` binutils
  - Bonus: support AVR, MSP430, RISCV by automatically detecting the compilation target (cf.
    .cargo/config) and automatically choosing the right binutils (`msp430-elf`, etc.)

- [ ] Make `cargo-objdump` et al. use LLVM binutils once #50 is implemented.

Originally discussed in #43 but I think this more in line with #44 

cc @thejpster

japaric, 2018-02-28:
Adding to the epoch 2018 milestone as discussed in yesterday's meeting

dvc94ch, 2018-03-30:
Can we also add `cargo-dwarfdump` [0]

[0] https://llvm.org/docs/CommandGuide/llvm-dwarfdump.html

whitequark, 2018-03-30:
When was the last time you needed `dwarfdump` and were *not* developing an LLVM backend?..

grossws, 2018-04-04:
@m4b wrote [`cargo-sym`](https://github.com/m4b/cargo-sym) which can be used as limited `objdump` replacement.

japaric, 2018-04-04:
@dvc94ch I didn't include that tool in the discussion issue I opened (rust-lang/rust#49584). You can bring it up in the comments.

During yesterday's meeting @jcsoo brought up the idea of having a step 0 for these subcommands: a minimal implementation could simply proxy the LLVM tool; for example, `cargo nm -- -C path/to/binary` would be sugar for `$(rustc --print sysroot)/path/to/llvm-nm -C path/to/binary`.

japaric, 2018-04-07:
> During yesterday's meeting @jcsoo brought up the idea of having a step 0 for these subcommands: a minimal implementation could simply proxy the LLVM tool

I have implemented step 0 in https://github.com/japaric/cargo-binutils

japaric, 2018-06-29:
The `llvm-tools` rustup component now exists and [`cargo-binutils`](https://github.com/japaric/cargo-binutils) is making use of it. Check this [announcement](https://internals.rust-lang.org/t/llvm-tools-a-new-rustup-component-for-binary-inspection-objdump-nm-size-and-profiling-profdata/7830) for details.

Let's discuss the final UI we want for the Cargo subcommands in the `cargo-binutils` [issue tracker](https://github.com/japaric/cargo-binutils/issues).

closed 2018-06-29 by japaric