rust-embedded/wg #53: Fix linking issues when using multiple codegen units

japaric, 2018-02-26:
To improve build times rustc is now defaulting to building crates using multiple codegen units. The
problem is that using multiple codegen units can sometimes break linking of no-std applications
("undefined reference to `rust_begin_unwind`").

We are looking for someone who can help us fix the problem in the compiler

### Tasks

- [ ] Tweak rustc linking process to properly support multiple codegen units. The problem and
  solution is described in https://github.com/rust-lang/rust/issues/47074#issuecomment-354588718

The current workaround for this issue is to force the compiler to use a single codegen unit in
Cargo.toml

``` toml
# workaround rust-lang/rust#47074
[profile.dev]
codegen-units = 1
incremental = false
```

thejpster, 2018-02-26:
I've also seen cargo/xargo emit entirely empty binaries unless this property is set.

japaric, 2018-02-26:
@thejpster with the cortex-m-rt linker script or with some other custom linker script? I have patched the cortex-m-rt linker script a few times to make it work with multiple codegen units but I wouldn't be surprised if there are still issues out there ...

thejpster, 2018-02-26:
This was with stellaris-launchpad, which had its own linker script dating back to my original C implementation. Would it be useful if we wrote up requirements for linker scripts somewhere, rather than relying on some magic in one example?

nunolucas, 2018-02-26:
The empty binaries, on Windows, may be related to this bug under heavy load:
https://randomascii.wordpress.com/2018/02/25/compiler-bug-linker-bug-windows-kernel-bug/

thejpster, 2018-02-26:
That is *beautifully* done, but I still wonder if it's worth pulling out into something that's more generic and not Cortex-M specific, to help people with other platforms and improve discoverability.

japaric, 2018-03-27:
> Tasks
> Tweak rustc linking process to properly support multiple codegen units. The problem and
> solution is described in rust-lang/rust#47074 (comment)

PR rust-lang/rust#49316 should fix this.

japaric, 2018-04-03:
Update: rust-lang/rust#49316 didn't quite fix this. See https://github.com/rust-lang/rust/pull/49316#issuecomment-378251276.

japaric, 2018-04-06:
Update: rust-lang/rust#49672 should fix this for real

japaric, 2018-04-08:
Fixed in rust-lang/rust#49672. :tada:

closed 2018-04-08 by japaric