rust-embedded/wg #54: Guide for doing embedded development in an IDE

japaric, 2018-02-26:
We have [instructions] for doing embedded development from the command line but we are missing
instructions for doing embedded development in an IDE.

[instructions]: https://docs.rs/cortex-m-quickstart

I heard that VS Code plus the cortex-debug extension works nicely for embedded development but I
have no first hand experience.

We are looking for someone who can help us write down some instructions.

### Tasks

- [ ] Write instructions for setting up a Cargo project in VS code including instructions for
  setting up the cortex-m extension, Xargo and the RLS. It could be any other IDE but VS code is the
  IDE with official RLS support.

- [ ] Bonus points if you write instructions for getting semihosting, ITM / SWO, register (SVD) view
  to work.

- [ ] Identify any changes that we could make to `cortex-m-quickstart` to make integration with VS
  Code, or any other IDE, easier. Like checking in some configuration file (?).

niklasad1, 2018-02-26:
I got debugging in VSCode to work in [TockOS](https://github.com/helena-project/tock) you can grab the script for remote debugging [here](https://github.com/helena-project/tock/blob/master/.vscode/launch.json) and a very brief tutorial [here](https://github.com/helena-project/tock/blob/master/doc/debugging/VSCode_Debugging.md). Feel free to use it but I haven't tried the cortex-m extension cheers.

Lakier15, 2018-03-01:
I've been using [IntelliJ IDEA Community](https://www.jetbrains.com/idea/) with the [Rust plugin](https://intellij-rust.github.io/) for almost 9 months. It works really well. Only problem might be the different approach from using the RLS.
I believe that they use their own parsing engine.

What really is lacking, out there, is some form of Rust IDE debugging. [Rust-DT](https://github.com/RustDT/RustDT) was a good effort to use the Eclipse capabilities to also debug embedded systems. However, 
the project has been abandoned, and well, its Eclipse.... 👎

nallar, 2018-03-02:
Debugging in CLION with intellij-rust works.

Lakier15, 2018-03-02:
I thought there were some problems with debugging in CLION.

Anyway, I think CLION is not free. It doesn't have a community edition, so it would not be accessible to everybody equally.

davidtibbetts, 2018-03-05:
I've had success with VSCode using OpenOCD and STM32 MCUs on Windows.
[launch.json](https://gist.github.com/davidtibbetts/ae8a9ba29f99c1d812d3aeffe20fedee)
[tasks.json](https://gist.github.com/davidtibbetts/9ae8131d9f7e87c62ac064302405c453)

AJAnderson, 2018-06-23:
Hi @japaric, I set up the cortex-debug extension in vs-code and got it working, would love to write instructions but where should I write them up?

mathk, 2018-06-24:
My experience with ITM was that I would never really know how to setup the TPIU clock from gdb. Once I freeze the MCU to maximum speed (216Mhz) the ITM port didn't work properly.

japaric, 2018-08-10:
This issue was moved to rust-embedded/book#12

closed 2018-08-10 by japaric