rust-embedded/wg #58: Embedded WG newsletter #1

japaric, 2018-03-10:
We'll be releasing periodic (biweekly) newsletters (on u.r-l.o, unless someone wants to suggest some other medium) about the progress the embedded community is making.

These newletters are a great chance to:

- Thank old and new contributors for their work :heart: 
- Inform the broader Rust community about what we are doing :loudspeaker: 
- Bring in new contributors :busts_in_silhouette: 
- Highlight `help-wanted` issues and issues that need more input :exclamation: 

The first newsletter will be released on 2018-03-15

---

Here's some stuff off the top of my head:

- @emilgardis, @jamesmunns and @ryankurte have become collaborators of the [svd] and [svd2rust]
  projects.

[svd]: https://github.com/japaric/svd
[svd2rust]: https://github.com/japaric/svd2rust

- @hannobraun, @ilya-epifanov, @thejpster, @therealprof have become collaborators of the
  [embedded-hal] project.

[embedded-hal]: https://github.com/japaric/embedded-hal

- @pcein has released the third weekly driver: [adc-mcp3008]

[adc-mcp3008]: http://pramode.in/2018/02/24/an-introduction-to-writing-embedded-hal-based-drivers-in-rust/

- [WIP] @varkor has fixed a [long standing LLVM bug][gh41315], that artificially increased the
  binary size of Rust programs, and backported the patch to rustc.

[gh41315]: https://github.com/rust-lang/rust/issues/41315

- LLD is now being shipped with the Rust toolchain. @FenrirWolf [patched Xargo][xargo-lld] so that
  Xargo users can make use of it too.

[xargo-lld]: https://github.com/japaric/xargo/pull/200

- We are [pretty close][cortex-m-rt-lld] to being able to use LLD to link Rust programs compiled for
  ARM Cortex-M but there are some [LLD bugs][lld-issues] that still need to be fixed. Help reporting
  these bugs upstream would be greatly appreciated!

[cortex-m-rt-lld]: https://github.com/japaric/cortex-m-rt/issues/53
[lld-issues]: https://github.com/japaric/cortex-m-rt/issues/53#issuecomment-371972935

- @dvc94ch has created the [riscv-rust] organization for all your embedded RISCV needs!

[riscv-rust]: https://github.com/riscv-rust

- @dylanmckay has begun [rebasing][avr-rust-rebase] the avr-rust/rust fork onto a recent
  rust-lang/rust version, bringing in LLVM 6.0 support.

[avr-rust-rebase]: https://github.com/avr-rust/rust/pull/91

Feel free to suggest more content / more details in the comments

cc @dylanmckay @dvc94ch @hannobraun @jamesmunns @jcsoo @pftbest @thejpster

japaric, 2018-03-14:
More stuff:

- @japaric has [released][1] the fourth weekly driver for the [`enc28j60`], an Ethernet controller
  with SPI interface.

[1]: http://blog.japaric.io/wd-4-enc28j60/
[`enc28j60`]: https://crates.io/crates/enc28j60

- @dbrgn has [released][2] the fifth weekly driver for the [`mcp3425`], a 16-bit ADC with I2C
  interface.

[2]: https://blog.dbrgn.ch/2018/3/13/rust-mcp3425-driver/
[`mcp3425`]: https://crates.io/crates/mcp3425

- [HelpWanted] Incremental compilation is in the roadmap for this year but it doesn't work with
  `no_std` binaries due to an [old rustc bug][gh18807] related to linking. Help us fix it! There's a proposed solution [here].

[gh18807]: https://github.com/rust-lang/rust/issues/18807
[here]: https://github.com/rust-lang/rust/issues/47074#issuecomment-354588718

- [HelpWanted] We want to explore stabilizing some assembly operations in core as an alternative to
  the unstable `asm!` macro. We would love some help identifying the assembly operations that need
  to be provided this way as some can be implemented in external assembly files without losing
  semantics. Details in #63.

jamesmunns, 2018-03-15:
Closed by #66.

closed 2018-03-15 by jamesmunns