rust-embedded/wg #60: Building a series hardware product/module which should be friendly for Rust programmers.

kfihihc, 2018-03-12:
I am newbie with Rust, but i am work on hardware/IoT over 5 years.

Some of my friends, who work on internet company, they don't have hardware background. But still want to try to build some IoT projects.

So I think it should be very interesting if we can provide a series friendly development boards/modules.

I and my friend create an open source Nordic nRF52832 BLE development board: https://github.com/makerdiary/nrf52832-mdk, but it's not friendly enough. As I know, there are a lot of sensors have been support by Rust with our community contributor https://github.com/rust-lang-nursery/embedded-wg/issues/39. So I want to build a lot of sensors/actuators which connect to main board with a easy-to-use connector.

How do you think? Do you have any advice?

kfihihc, 2018-03-12:
BTW, we were ported [TockOS](https://github.com/helena-project/tock) to nRF52832-MDK which will pull request to TockOS's repo soon.

LinusU, 2018-03-12:
Have you seen [the Tessle](https://tessel.io)?

It has support for both JS and Rust ❤️

therealprof, 2018-03-12:
@LinusU That's a rather weird device and not quite like what @kfihihc is proposing... If a cheap embedded device with Rust support is needed you can always get a BBC micro:bit.

@kfihihc That's really nifty. I like it. I'm not quite sure what exactly you're looking for though, care to elaborate?

kfihihc, 2018-03-12:
@LinusU actually, Tessle makes me confuse, and it based on MT7620n(It's MIPS ISA) which different with ARM Cortex-M series chips, MT7620n has MMU that means it can runing an OS, like OpenWRT and not design for IoT.

@therealprof Right now, I just have a simple idea of the hardware stack: main board + extension board.

1. Main board has a unified form factor. So that we can build different main board with different MCUs, but all of them are compatible with extension board in the same form factor. 
2. Extension board should be as a single function module, like OLED module, IMU module, etc. And most important is that it's stackable which means we can stack multiple modules.

I like BBC Micro:bit's connector, very easy-to-use, just like the gameboy ROM cartridge. But it can not suport stack multiple modules. And Seeed has a series module called [Grove](http://wiki.seeed.cc/Grove_System/#interface-of-grove-modules), but just suport 4-pins, so you can not connector with SPI module(there are 5 pins). Adafruit has a lot of [sensors](https://www.adafruit.com/category/35) but you have to soldering by yourself if you want to connect it with breadboard or other main board, I think it's not friendly.

So we have two choices, one is like Raspberry Pi or 96Board:
1. Raspberry Pi with male pin header:
[image]

2. 96Board with female pin header:
[image]

Other is normally pin header(it's compatible with breadboard):
[image]
[image]

Something maybe like this(reference [here](http://therebelarmory.com/thread/9344/teensy-v3-extension-board?page=1)):
[image]

therealprof, 2018-03-12:
@kfihihc There's some more "standards" you could use, like "Arduino Nano", used by Arduino Nano and e.g. Nucleo-32 board (e.g. https://os.mbed.com/platforms/ST-Nucleo-F042K6/). The last one you've provided a picture for is, I believe, the Teensy. Then (if you don't mind larger) there's ST Morpho, and of course Arduino.

Last but not least you could also do PMOD.

kfihihc, 2018-03-13:
@therealprof PMOD? You means this: https://store.digilentinc.com/pmod-modules/ ? It seem like Seeed's Grove modules but more flexible.

Also there are some other connectors, like https://www.littlebits.cc, but it's too expensive for development boards. Arduino-like form factor maybe the best choice right now. 

BTW, Particle's Photon has a nice example.
[image]

therealprof, 2018-03-13:
@kfihihc Yes, that PMOD. Very popular with the FPGA crowds.

Almost forgot: MikroBUS is also very popular: https://www.mikroe.com/rs/mikrobus

kfihihc, 2018-03-13:
@therealprof Well, MikroBUS seems very good. Thanks for suggestions.

Let me try to build some prototyping, and i will update here later.

kfihihc, 2019-02-14:
@jamesmunns Until now, we still use the breadboard frinedly 2x pins header, like below.

But we have a plan to release a new 7x1 connecter for our border router level product(WiFi+BLE+Zigbee+Thread). I will update it ASAP.

[image]

stappersg, 2020-02-28:
Because there  is [STM32L0X2 H.A.L.](https://crates.io/crates/stm32l0xx-hal) seems it to me
that [PX-HERO Board](https://www.crowdsupply.com/piconomix/px-her0-board/) is a safe bet to buy.

Where to report the actual succes (some where in August)?

I'm asking because that is what this issue is about. (Isn't it?)

adamgreig, 2024-06-11:
Closing this old issue as part of today's triage. Many boards are now popular with, and well supported by, various embedded Rust efforts, such as the microbit in our own discovery book and Espressif's [dedicated Rust board](https://www.espressif.com/en/dev-board/esp32-c3-devkit-rust-1-en).

There's no need for the working group to create any new hardware today; lots of hardware exists and is well supported by Rust.

closed 2024-06-11 by adamgreig