rust-embedded/wg #65: Embedded WG newsletter #2

jamesmunns, 2018-03-15:
Placeholder for the next Embedded-WG Newsletter, planned for 2018-03-29

DoumanAsh, 2018-03-25:
I started working on HAL for my l4 board so JFYI https://github.com/DoumanAsh/stm32l4x6_hal

jamesmunns, 2018-03-25:
Note to self: embedded hal mocking: https://github.com/rust-lang-nursery/embedded-wg/issues/70

jamwaffles, 2018-03-25:
There's an [SSD1306 driver](https://github.com/jamwaffles/ssd1306) me and @scowcron have been working on. Nearly ready for release, just needs some good examples and documentation (as with any new project...)

I'm also working on an [embedded-graphics](https://github.com/jamwaffles/embedded-graphics) crate with iterator-based constructs for drawing primitives, images and text. Would appreciate some input from the community to see if I'm going in the right direction.

japaric, 2018-03-29:
I'll write some notes about what happened during the All hands for the newsletter.

jamesmunns, 2018-03-29:
Closed by #73

closed 2018-03-29 by jamesmunns