rust-embedded/wg #68: Awesome embedded Rust: A curated list of embedded crates

japaric, 2018-03-18:
We have list of embedded / no-std crates in #12 and #39 but, for example, the first list is not currently being maintained. Both lists should be reviewed to ensure that the crates in the lists (a) still work and (b) are being maintained. The lists could also make use of some categorization: data structures, device crates, drivers, etc.

So I propose we create an awesome-embedded repo like [this one](https://github.com/rust-unofficial/awesome-rust) to curate a single list of embedded crates.

We can probably put the repo under the rust-embedded org. (@posborne @nastevens thoughts?)

Also, we need people to make and maintain the list. If you would like to help let us leave a comment below!

We could also make a website with search function and stuff but I think a README is a good start. If you have front end chops and would like to volunteer for making a website let us know in the comments.

cc @jamesmunns @therealprof

therealprof, 2018-03-18:
@japaric Yes, I'd be interested in maintaining such a list but my website chops are rather limited.

nastevens, 2018-03-18:
@japaric Definitely onboard to have the list on rust-embedded org. Maybe just call the repo `awesome-embedded-rust`?

@therealprof I think these lists are usually just managed as Markdown files rendered directly in-repo, so no website chops needed

therealprof, 2018-03-18:
@nastevens The list is certainly a good start but the approach lacks in dimensionality to capture the complexity of the topics. I think we should be able to provide at the very least (actually depending on the type of the subject, i.e. peripheral access, HAL implementation, BSP, driver):
- Name
- Type
- Short description
- Crates.io name
- Repository
- State of the implementation
- Location of the data sheet
- Dependencies
- Associated blog articles
- Examples/Tested environments

I'd also like to see a section where authors can ask for help, feedback and testing. An section for ideas for new implementations as well as the possibility to note WIP to prevent duplicated effort.

posborne, 2018-03-19:
> We can probably put the repo under the rust-embedded org. (@posborne @nastevens thoughts?)

berkus, 2018-03-21:
JFYI, community maintains https://github.com/berkus/awesome-embedded-rust/

japaric, 2018-03-26:
@berkus would you be willing to take this [list of HAL impl and driver crates](https://github.com/japaric/embedded-hal#implementations), as well as the [list of WIP driver crates](https://github.com/rust-lang-nursery/embedded-wg/issues/39#issue-289457410). As well as make @therealprof and other interested people as a collaborators and take the stream of WIP stuff posted to https://github.com/rust-lang-nursery/embedded-wg/issues/39?

I'd like to have all this in a single place; if there exists a place for this I'd rather not create a new list. Then we can point people to that single list from the embedded-hal repo and the weekly driver issue.

japaric, 2018-04-01:
I have created a new awesome-embedded-rust list in [rust-embedded/awesome-embedded-rust](https://github.com/rust-embedded/awesome-embedded-rust). It contains all the drivers from #39 and the HAL impl and device crates from [japaric/embedded-rust](https://github.com/japaric/embedded-hal)

@therealprof I've sent you a collaborator invite to the new list. If anyone else wants to help maintain the list let me know!

@berkus I'd be happy to merge the two lists at any time.

RandomInsano, 2018-04-02:
I’m game. I’m finding it fun to follow the new device crates as they come in anyway. I guess we’ll just make sure the WIP items exist in some form and that the completed drivers are well documented and have some news item to introduce it.

Do we have volunteer mentors for quality and standards? My AXP209 crate could use some review: It’s of a beast of a chip and code org has been a challenge. If some people are willing to step up I can coordinate when MRs call out a need for review.

berkus, 2018-04-02:
@japaric you want to merge them into rust-embedded one? I think go ahead and do it, or I can create a PR adding the entries from the community one. I'd be happy to be on the collaborators list in the new one. Wdyt?

japaric, 2018-07-16:
This exists as https://github.com/rust-embedded/awesome-embedded-rust

closed 2018-07-16 by japaric