rust-embedded/wg #69: The binary size - performance tradeoff

japaric, 2018-03-22:
As of the latest LLVM upgrade (4.0 -> 6.0 on 2018-02-11) LLVM seems to now perform loop unrolling more agressively; this increased the binary size of a [minimal] program that only zeroes .bss and initializes .data from 130 bytes .text (nightly-2018-02-10) to **1114** bytes (nightly-2018-03-20) when using opt-level=3 + LTO -- FWIW, I highly doubt the loop unrolling actually improves performance at all. Original report: rust-lang/rust#49260

[minimal]: https://github.com/japaric/stable-embedded-rust/blob/7cc87cff95b2d90b5cab258d912cf4692130312f/examples/minimal.rs

This put us in a bad spot because by default we'll end with large optimized (`--release`) binaries -- I can already foresee future comparisons between C and Rust pointing out that the smallest embedded C program is only a hundred bytes in size whereas the smallest embedded Rust program is 1 KB.

So we should make sure we clearly document why Rust programs are so large by default and how to make Rust programs small. Using opt-level=s + LTO on the minimal program mentioned above brings the size back to 130 bytes .text.

cc @jamesmunns ^ that should be included in the book

There are other possibilites to explore here: like having something like C's / clang's `#pragma nounroll` to prevent LLVM from optimizing loops marked with that attribute, but I doubt we'll get any of that into the 2018 edition release -- it's too late, I think.

therealprof, 2018-03-23:
FWIW: I think funky stuff like unrolling (which is really worthless on embedded architectures) is to be expected to be done at higher optimisation levels and I've seen all kind of funky size regressions. I'm always using (and recommending) `-s`.  Potentially `-z` could also be tried, but `-O3` is a big nono...

therealprof, 2018-03-23:
> the smallest embedded C program is only a hundred bytes in size whereas the smallest embedded Rust program is 1 KB.

NB: I highly doubt that. As soon as one uses some the initialisation code from some of the typical SDK the code will be well in the kBs already. To even stay in Rusts range you'll have to manually bang the memory mapped registers and write your own linker scripts.

Case in point, this is the smallest possible binary for a `main { while(1) {} }` loop for the STM32F051 I could achieve with based on `STM32Cube` initialisation:
```
# arm-none-eabi-size .pioenvs/disco_f051r8/firmware.elf
   text	   data	    bss	    dec	    hex	filename
    892	   1080	   1600	   3572	    df4	.pioenvs/disco_f051r8/firmware.elf
```

Emilgardis, 2018-03-23:
Could we get a RFC for something like `#[no_unroll]`/`#[unroll(disable)]`?

whitequark, 2018-03-23:
@Emilgardis No need for an RFC, marking the function with the loop as `#[cold]` should suffice.

whitequark, 2018-03-23:
> This might be the cause for some amount of bloat due to unnecessary branches, but shouldn't #[inline] be a strong enough hint to LLVM to still inline the function?

The inlining thresholds in LLVM are tailored for C, which produces functions with relatively compact IR, and likely aren't well suited for Rust. In our in-house language we had to raise them significantly to get decent reductions in code size.

Emilgardis, 2018-03-23:
@whitequark I've never heard of that attribute, seems like it should work however.

therealprof, 2018-03-23:
@jonas-schievink I can not confirm that it produces larger files with opt-level=s, at least not in general. This all has quite a bit of premature optimisation smell to it, same as with the `#[inline(always)]` we had sprinkled all over the map...

RandomInsano, 2018-03-24:
> There has been a small amount of discussion about unrolling atttributes:
> https://github.com/rust-lang/rfcs/issues/2219

The choice to unroll or not should be at the LLVM layer. With the Linux HAL I could very well want loop unrolling if my x86_64 machine was using a driver over the SMBus.

Is it possible for the LLVM backends to automatically opt-out when appropriate?

japaric, 2018-08-10:
This issue was moved to rust-embedded/book#11

closed 2018-08-10 by japaric