rust-embedded/wg #71: Device Crates without svd2rust?

andre-richter, 2018-03-26:
I hope I haven't overlooked some obvious resources or documentation, but are there any guidelines for creating device crates *without* `svd2rust`?

I am asking because the embedded-hal documentation states:
> The HAL traits should be implemented for device crates generated via svd2rust to maximize code reuse.

So I wonder what would be best practice if SVD files are not available?
Raspberrypi, for example, would be a valuable device crate, but I couldn't find SVD files.

therealprof, 2018-03-26:
I have to agree that this is a nomenclature issue. If SVD is available then it makes a whole lot of sense to use it but there's really no use in trying to enforce that. In fact the author of this paragraph has obviously violated that rule himself: https://github.com/japaric/linux-embedded-hal

japaric, 2018-03-26:
> The HAL traits should be implemented for device crates generated via svd2rust to maximize code reuse.

> should

This is a recommendation and not a hard requirement. I agree the text should be tweaked to recommend svd2rust for Cortex-M and point to the linux-embedded-hal crate as an example that doesn't use a device crate.

japaric, 2018-03-26:
Also shouldn't this issue be filled on the embedded-hal repo as that's where the text is?

andre-richter, 2018-03-26:
Sorry for posting on the wrong repo. I had multiple tabs open and maybe got lost.

linux-embedded-hal is a good example for providing embedded-hal on top of a Rich OS that already abstracts away the unpleasant details like defining MMIO locations, etc.

However, there is also bare-metal programming on Raspberry Pi, for example, which in my opinion is a middle-ground where you neither have `svd2rust` crates to get you kickstarted nor Rich OS interfaces to get you going easily.

So I was wondering if guidelines for creating device crates for such use-cases/platforms would be helpful. Or is looking at existing `svd2rust` output and trying to emulate that enough? I found it a bit overwhelming at first try to go that way, that's why I wanted to reassure.

closed 2018-08-08 by andre-richter