rust-embedded/wg #72: Embedded WG Newsletter #3

jamesmunns, 2018-03-29:
Placeholder issue for the next WG Newsletter, scheduled for 2018-04-12

whitequark, 2018-03-29:
Do our LLVM/Rust forks with OR1K support (which is quite good at this point, we've been using it in production for several years now) count as something that should be included in the newsletter? OR1K is an architecture that is used primarily on FPGA softcores. (There's been a few ASIC implementations but nothing general-purpose I believe.)

jamesmunns, 2018-03-29:
@whitequark definitely! Do you have any repos or blog posts that I can link to for the next post?

whitequark, 2018-03-29:
I don't have latest stable running quite yet (because it was released today, and I'm updating the OR1K backend for LLVM 6.0 as we speak), but I'll comment in this issue once that's done.

andre-richter, 2018-03-31:
I've written a bare metal Hello World over UART0 on the Raspberry Pi 3. Some init code is taken from the Embedonomicon. Kindly have a look at https://github.com/andre-richter/rust-raspi3-tutorial

japaric, 2018-04-06:
We should include the [Awesome embedded Rust](https://github.com/rust-embedded/awesome-embedded-rust) list in the next edition.

The next instance of the weekly driver section should include the total number of released and WIP drivers in one of the first paragraphs.

jamesmunns, 2018-04-08:
Note to self:

* https://github.com/japaric/cargo-binutils
* https://users.rust-lang.org/t/psa-breaking-change-extern-crate-compiler-builtins-is-now-included-in-no-std-crates/16704
* https://users.rust-lang.org/t/psa-you-no-longer-need-xargo-to-do-arm-cortex-m-development/16703

jamesmunns, 2018-04-08:
Note to self:

* http://ithinuel.me/embedded-rust-why/

japaric, 2018-04-09:
Another one for the news / updates section:

- https://users.rust-lang.org/t/cortex-m-rt-v0-4-0-now-you-can-link-arm-cortex-m-programs-using-lld/16751

The help wanted section should include:

- Implementation of RFC 2070. [nagisa is willing to mentor](https://github.com/rust-lang/rust/issues/44489#issuecomment-378058031).

- Figure out which Cortex-M assembly ops are widely used in practice and need to be inlined to have the right semantics. Basically, the first work item of https://github.com/rust-lang-nursery/embedded-wg/issues/63#issue-305114817

myeisha, 2018-04-12:
https://crates.io/crates/thumb2-stack-size might be useful to some.

closed 2018-04-16 by japaric