rust-embedded/wg #76: How to subscribe to newsletters?

ooxi, 2018-04-02:
I see you prepare weekly newsletters for the embedded wg. How do I subscribe to receive one via mail?

jamesmunns, 2018-04-02:
Hey @ooxi, we put out a newsletter every other week, though we don't really have a mechanism for subscribing by email at the moment. We do typically post them in a couple places, like https://internals.rust-lang.org, and @japaric and I always tweet them, but that doesn't specifically answer your question.

I guess it would be pretty easy to throw together an RSS feed, though I don't know how I would get an email newsletter going.

ooxi, 2018-04-02:
I'm new to the rust community in general and was expecting some kind of mailing list I've missed. But following this repository isn't that bad either :-)

closed 2018-04-02 by ooxi

nickray, 2018-10-13:
Perhaps set up a Google Group to dispatch the newsletters? Seeing that Google Calendar is already used for the meetings.