rust-embedded/wg #79: Expose the -warn-stack-size= LLVM option somehow

whitequark, 2018-04-06:
I've just learned that LLVM can, in fact, do a *limited* stack size analysis: https://lists.llvm.org/pipermail/llvm-commits/Week-of-Mon-20130603/177010.html.

It's not interprocedural but I think it's still quite useful e.g. if LLVM ends up allocating a large array on stack, which seems to be a source of stack overflows that might be on the second place after recursion IME.

japaric, 2018-04-07:
In theory `-C llvm-args=-warn-stack-size=$N` should pass the flag to LLVM, but since these are warnings rustc won't report them to the console, I think.

I'm looking at the LLVM source code and it seems that there's [some option](https://github.com/rust-lang/llvm/blob/6ceaaa4b0176a200e4bbd347d6a991ab6c776ede/lib/CodeGen/PrologEpilogInserter.cpp#L953-L957) to emit (as debug logs?) the number of stack bytes that each function uses. I have no idea how to make rustc print these statistics though.

Perhaps @nagisa or @eddyb can give us some clues here?

whitequark, 2018-04-07:
@japaric I think you need to call `LLVMContextSetDiagnosticHandler`.

japaric, 2019-02-04:
@jamesmunns the work I did was related to #106. The llvm feature described here is a different feature.

jamesmunns, 2019-02-14:
@japaric do you think this should stay open? If so, please remove the cleanup tag.

japaric, 2019-03-04:
@jamesmunns not sure, but I can tell you that I don't intend to work on this one. We don't rely have a policy for keeping "interesting project" issues around. Perhaps this could be moved into the "not yet awesome embedded Rust" list? Then we can close this issue.

adamgreig, 2024-06-11:
Closing this issue as part of today's triage. I'm sure there's still work to be done on improving stack size handling and overflow detection, but no one in the WG is currently planning to do it. There have been some interesting developments in handling stack overflows, such as [flip-link](https://github.com/knurling-rs/flip-link), and newer Arm cores even have stack limit registers that are checked in hardware.

closed 2024-06-11 by adamgreig