rust-embedded/wg #84: Embedded WG #4

jamesmunns, 2018-04-14:


djmcgill, 2018-04-16:
My utility [https://github.com/djmcgill/form](Form) is useful when working with tools like svd2rust which generate massive lib files, might be worth a mention - alway happy to accept pull requests.

japaric, 2018-04-24:
[`heapless`] v0.3.1 has been released; this version provides fixed capacity [re-implementations] of std::HashMap and std::HashSet that work on `#![no_std]`, do not require a memory allocator, can be allocated on the stack or stored in a static variable, and use 32-bit hashes (32-bit FNV and MurmurHash3 hashers are available in the [`hash32`] crate).

[`heapless`]: https://crates.io/crates/heapless
[re-implementations]: https://japaric.github.io/heapless/heapless/struct.IndexMap.html
[`hash32`]: https://crates.io/crates/hash32

astro, 2018-04-24:
> `heapless` v0.3.1 has been released; […]

When discussing the `heapless` crate, I would like to hint at the [`managed`] crate. It additionally works on `Vec` et al if built with `std` and it features a `BTreeMap` behind the `map` feature-flag.

[`managed`]: https://crates.io/crates/managed

danielgallagher0, 2018-04-25:
I've started a [BLE](https://github.com/danielgallagher0/ble-rs) crate. I'm currently "only" aiming to support BLE hosts, so it would be used to interface with an external controller, not to implement a controller. It's still very early, but I have a framework in place so it should be straightforward to add the remaining commands and events. I'd love to get some feedback, advice, etc.

I also have a first implementation, [BlueNRG](https://github.com/danielgallagher0/bluenrg) which should support STMicro's family of BlueNRG chips. I have an [SPBTLE-RF](http://www.st.com/content/st_com/en/products/wireless-connectivity/bluetooth-bluetooth-low-energy/spbtle-rf.html) that I am working with, so those two crates will at least support that chip.

A particular request would be for someone to add to BLE and/or use it on a different chip, preferably from a different vendor.

Oh, and if anyone has an idea for a better name, that would be great. [ble](https://crates.io/crates/ble) is already claimed.

therealprof, 2018-04-25:
@danielgallagher0 Nice. So you're implementing an adapter for the HCI interface I would presume?

I'm not a big fan of 3 letter acronyms anyway as they make for horrible search terms. How about writing out Bluetooth and if you're writing an HCI adapter -- throw that in as well?

jamesmunns, 2018-04-25:
@djmcgill wow, `form` is everything I didn't know I needed :smile:

I just used it on my `nrf52` crate and it worked perfectly: https://github.com/jamesmunns/nrf52/tree/master/src

Thanks for that!

japaric, 2018-04-25:
The newsletter should mention that we our IRC meetings are open for anyone to attend and that guests can bring up issues / topics to discuss by commenting on the [agenda issues](https://github.com/rust-lang-nursery/embedded-wg/issues/91). We should also mention that members of the embedded WG will be present during the RustFest Paris impl days and that we are [looking for contributors](https://github.com/rust-lang-nursery/embedded-wg/issues/90) interested in working towards the 2018 edition milestone.

danielgallagher0, 2018-04-25:
@therealprof Yes, it's an implementation of the HCI specification. I'm looking over the BT spec, and I think just calling it the Bluetooth HCI would be accurate, and bluetooth-hci makes a decent crate name, and can be extended in the future (bluetooth-core, bluetooth-controller, etc.).

japaric, 2018-04-26:
We have preview of embedded Rust on stable and would love some feedback. See https://github.com/rust-lang-nursery/embedded-wg/issues/42#issuecomment-384524779 for details.

hannobraun, 2018-04-26:
Version 0.3 of the [lpc82x crate](https://crates.io/crates/lpc82x) has been released. It now uses the newest versions of svd2rust and cortex-m.

hannobraun, 2018-04-27:
I realize it's past the deadline(?), but since the newsletter is not out yet, I'm adding this. Happy to wait for the next one though, if that works better for you.

Version 0.1 of LPC82x HAL, the hardware abstraction layer for NXP's LPC82x family of microcontrollers, has been released. Links: [Announcement](https://users.rust-lang.org/t/lpc82x-hal-0-1-hardware-abstraction-layer-for-nxp-lpc82x-mcus/17116), [crates.io](https://crates.io/crates/lpc82x-hal), [GitHub](https://github.com/braun-robotics/rust-lpc82x-hal)

jamesmunns, 2018-04-27:
Hey @hannobraun, I plan on writing the post today, busy week :). Your post will make it in this week.

PR #94 (2018-04-28): Fourth Newsletter
Fourth Newsletter

Closes #84

[Rendered Here](https://github.com/rust-lang-nursery/embedded-wg/blob/newsletter-4/newsletters/2018-04-28.md)
closed 2018-04-28 by japaric