rust-embedded/wg #87: Semantics for RW CPU registers

andre-richter, 2018-04-15:
Hi all,

I recently started with a `cortex-a` crate, and tried to closely follow what `cortex-m` does so that we are aligned between architecture crates. For registers, we currently have two kinds of interfaces:

1. For registers that store a single value, simple read and write methods are provided that take and return integers, e.g.:
    1. `register::read() -> u32`
    2. `register::write(bits: u32)`

[Here's a link](https://github.com/japaric/cortex-m/blob/master/src/register/psp.rs) to an example.

2. For registers whose values are subdivided into flags that encode control or status information, we provide an own type that currently has getter methods for both the whole register and the respective flags. [Example here](https://github.com/japaric/cortex-m/blob/master/src/register/control.rs).

For these kinds of registers, up until now, only the reading part is defined. I'd like to discuss how the write part should look like, because I stumbled upon some questions when think about it.

There are use-cases where you want to update one or more flags of a register before you push the new value to the actual CPU (aka calling multiple setters consecutively). I think here we could go the same way as with `volatile_register` and provide a `modify` function that provides the register type in a closure. The register type should then have setter methods implemented complementary to the getter methods. What do you think?

Should the `modify()` function return the register like the `register::read()` does, or not?

Also, sometimes we don't care what the previous value of a register is. Take the example of an enable-bit. Should there be dedicated functions for these cases? This would basically result in two setters per register flag. First as a function of the register module, and second as a method for the register type.

I'm sure there's a lot more I did not think of yet. I'm happy to get some input.

BR,
Andre

andre-richter, 2018-04-16:
I got a bit of inspiration from the [x86_64 crate](https://github.com/rust-osdev/x86_64), and came up with the following implementation now:
https://github.com/andre-richter/cortex-a/blob/redesign_timers/src/register/cntp_ctl_el0.rs

One usage example can be seen here:
https://github.com/andre-richter/rust-raspi3-tutorial/blob/timers/09_delays/src/delays.rs

It differs from the `cortex-m` implementation in that it doesn't use custom enums for the bitflags, but leverages the `bitflags` crate. This way we get a `bits` member variable by default, as well as functions for checking (`::contains()`), setting (`::set()`), toggling, and many more. For writing to registers, there's both `modify()` and `write()`, like in `volatile_register`.

@japaric, any comments maybe? Do you think it would be useful to align the crates to some extent? Are you open for discussions or do you think it's not needed? 

Maybe a trait for register R/W operations could be useful to force some kind of compatibility?

ryankurte, 2018-04-16:
I like the modify in a closure idea a lot! And I think it'd be neat to have this well documented and for it to be easier to generate and write these accessors.

Just for experimentations sake I have been [playing with](https://github.com/ryankurte/rust-embedded-builder) the same problem and came up with a [builder style](https://docs.rs/embedded-builder/0.1.1/embedded_builder/register/struct.Register.html) approach that lets you chain write operations, along with macros that apply [register specific traits](https://github.com/ryankurte/rust-gba/blob/master/src/memory.rs#L28) [over top](https://github.com/ryankurte/rust-gba/blob/master/src/graphics/bitmap.rs#L53) of the register implementation.

It's far from perfect, but, I am enjoying the builder style approach.

andre-richter, 2018-04-19:
Hey Ryan,

thanks for sharing and feedback.

> I think it'd be neat to have this well documented and for it to be easier to generate and write these accessors.

Absolutely, I started to [move these accessors to macros](https://github.com/andre-richter/cortex-a/blob/master/src/register/macros.rs), which reduced boilerplate code immensely and [improves the ease of adding them](https://github.com/andre-richter/cortex-a/blob/master/src/register/cntp_ctl_el0.rs#L44-L50) when writing new registers. On the documentation front, still work to do :)

I also like the chaining approach. I wish the functions generated by `bitflags` would return `&self` by default, so this would be possible by default. Not sure which way to go here.

I did not wrap my mind around adding trait(-bounds) yet. In my case, with `bitflags` being restriced to integer types anyways, not so much I guess.

ryankurte, 2018-04-19:
Neat! I do like the idea of doing as much as possible in code rather than macros from a readability perspective, but it seems like we're going to need a combination of both.

I ran into some bumps trying to support both chaining and non-chained approaches in the same api, I suspect it's possible but wasn't clear to me how to do it. 
I had a play with bitflags but wasn't convinced it does what we want, the bitwise accessors are important to help but it seemingly doesn't support multi-bit fields (?) and I think the APIs we're generating should eventually be typed with bools and generated enums etc.

idk whether it's sensible, but I like the idea of splitting the read/write traits out to a `register-ops` or similar package which would be implemented by the base embedded crates, and we could then implement the chaining and trait building etc across generic register types (which is what I was trying [here](https://github.com/ryankurte/rust-embedded-builder/blob/master/src/register.rs#L20).

andre-richter, 2018-04-20:
Btw I just realized that there's no need to restrict this to _CPU_ registers, most or all of it should apply to _MMIO_ as well. I edited the title accordingly.

Let me try to summarize what we have so far, kindly adapt/edit:

A register can be `r`, `w`, or `rw`. Here, traits defining a simple interface would make sense I guess.

```Rust
mod register_ops {
    pub trait Read<T> {
        fn read_raw() -> T;
    }
    
    pub trait Write<T> {
        unsafe fn write_raw(val: T);
    }
    
    // If a register implements r+w, this can be pulled in with a default impl.
    pub trait Modify<T> : Read<T> + Write<T> {
        unsafe fn modify_raw<F>(f: F)
        where
            F: FnOnce(&mut T),
        {
            let mut val = Self::read_raw();
            f(&mut val);
            Self::write_raw(val);
        }
    }
}
``` 

R/W functions can take the most different implementations. E.g. taking ARM architectures, they could be:
- asm `mov`s instructions
- asm `mrs/msr` instructions
- `volatile` r/w operations on MMIO
- Maybe more?

On top of that, we would need some "higher level" functionality for modifying register contents by setting and checking `bitfields`, which are a collection of one or more (consecutive?) bits that encode a register's flags. Flags, in turn, maybe subject to being further typed with `bools` or `enums`. These types should be associated types of the register type to portray the connection between both, something like:

```Rust
struct CONTROL_REGISTER {
    const ENABLE = 1;
    const MASK = 1 << 1;

    enum ...
}

impl register_ops::Write<u64> for CONTROL_REGISTER { ... }
```

Nemo157, 2018-04-20:
I had a comment relevant to this against the `svd2rust` register definitions: https://github.com/japaric/svd2rust/issues/198

It seems you've got a similar plan since you're not requiring the closure to return the writer (`where F: FnOnce(&mut T)`), you could potentially expand this to allow returning a different value from the `modify` method.

andre-richter, 2018-05-09:
Related: https://www.tockos.org/blog/2018/mmio-registers/

andre-richter, 2018-07-16:
I have a combined CPU and MMIO register interface, with type-safe and convenient bitfield manipulation now published at https://crates.io/crates/register

It's based on tock's register interface.

closed 2018-07-16 by andre-richter