rust-embedded/wg #88: Add targets upstream for new ARM Cortex chips

jamesmunns, 2018-04-17:
We should consider adding target JSONs and build rust-std compenents for the following targets before the 2018 era lands:

* ARM Cortex-R Target(s) based on thumbv7em (Cortex-R4, R5, R7, R8)
* ARM Cortex-M Target(s) based on thumbv8em (Cortex-M23 and Cortex-M33)
* ARM Cortex-R Target(s) based on thumbv8em (Cortex-R52)

This may also require updates to low level crates like `cortex-m`, maybe `compiler_builtins`, though we should probably open separate issues for that.

We likely will need physical chips to verify functionality before adding them upstream, which may be difficult for chips based on thumbv8em, as they are not currently retailing at the moment (as far as I know, the Nordic nRF91 is the only one shipping, and it is only in a limited sample release).

paoloteti, 2018-05-14:
Here [https://github.com/paoloteti/ti-hercules-bsp](https://github.com/paoloteti/ti-hercules-bsp) you can find a board support package + JSON for TI TMS570LS3xx (tested on TMS570LS3137) that is a BigEndian Cortex-R4F (in LockStep mode).

Linaro toolchain (`armeb-none-eabi`) it is perfect and you don't need to use TI ARM CGT.

I already sent patches on `compiler_builtins` to add some missing features (mainly on VFP) , so it works as it is now!

Only the F021 flash memory need the TI F021 SDK because the algorithm is proprietary and the overall logic it is totally different form other flash memories.  Mainly to ensure a set safety constraints at silicon level.  [Here](https://github.com/paoloteti/f021-flash-api) you can find a Rust binding.

I don't have time at the moment to add/integrate my work on upstream, but I can test on real hardware on my spare time.

BTW do not compare Cortex-R with Cortex-M: it is a different beast.

paoloteti, 2018-05-20:
Cortex R4F/R5F (BE,  ARM mode) is now upstream (PR 50813)

ithinuel, 2018-07-09:
I don't know if these devices are already available but I also heard about : 

Nuvoton M2351
---

https://www.cnx-software.com/2017/03/21/numicro-m2351-trustzone-enabled-arm-cortex-m23-mcu-is-designed-for-fingerprint-applications/
http://armdevices.net/2017/03/20/worlds-first-arm-cortex-m23-in-nuvoton-m2351/
https://github.com/ARMmbed/mbed-os/pull/7302

Arm Musca A1
---

https://developer.arm.com/products/system-design/development-boards/iot-test-chips-and-boards/musca-a-test-chip-board
https://github.com/zephyrproject-rtos/zephyr/pull/7645

japaric, 2018-08-31:
I was looking at the LLVM features for the ARMv8-M architecture and there are two ATM: `v8m` for baseline ARMv8-M (i.e. Cortex-M23) and `v8m.main` for mainline ARMv8-M (i.e. Cortex-M33). What I'm not sure about is how to call these new targets: `thumbv8m-none-eabi` for the M23, but then what would the M33 be called? `thumbv8m.main-none-eabi`? Furthermore the M33 has an optional FPU so we'll want two targets for that one: `eabi` and `eabihf`.

paoloteti, 2018-08-31:
@japaric 

> What I'm not sure about is how to call these new targets: thumbv8m-none-eabi for the M23, but then what would the M33 be called? thumbv8m.main-none-eabi? Furthermore the M33 has an optional FPU so we'll want two targets for that one: eabi and eabihf.

Exactly, infact `llc` accept both `thumbv8m.main-none-eabi` and `thumbv8m.base-none-eabi` and with `eabihf` sets the VFP attribute (as minimum).  (something like  `.eabi_attribute	28, 1	@ Tag_ABI_VFP_args`).

hannobraun, 2018-10-11:
NXP has a Cortex-M33 MCU in pre-production now: https://www.nxp.com/products/processors-and-microcontrollers/arm-based-processors-and-mcus/lpc-cortex-m-mcus/lpc5500-series/high-efficiency-arm-cortex-m33-based-microcontroller-family:LPC55S6x

jamesmunns, 2018-10-13:
Nice! Thanks @evq!

evq, 2018-10-29:
@thejpster sure, I upload later today. I have an svd2rust crate with LED example code for the SAM L11 as well as needed changes to the `cortex-m`, `cortex-m-rt`, `cortex-m-semihosting`, and `panic-semihosting` crates.

thejpster, 2018-10-29:
Awesome, thanks @evq !

evq, 2018-11-04:
@thejpster You can find the svd2rust crate with example code here - https://github.com/evq/atsaml11xxx

hug-dev, 2018-11-16:
Armv8-M is trendy! I have been working with Rust and Armv8-M at Arm for the past few months and just made a pull-request for Mainline support rust-lang/rust#56000 !

codido, 2019-03-17:
FWIW, the LPC55S69-EVK development board from NXP which has two Cortex-M33 cores is now shipping as well.

Things appear to work well including basic peripheral functionality, and only required a few mild modifications to the svd file.
The only minor gotcha was that cortex-m-rt includes the thumbv8m.base-none-eabi prebuilt library, but not thumbv8m.main-none-eabi.

By the way, it seems that out of the three thumbv8m targets, only thumbv8m.main-none-eabi is included in the manifests (and can be installed with rustup). Wasn't sure if that's intentional.

hug-dev, 2019-03-17:
@codido I think `thumbv8m.main` is currently being added to `cortex-m-rt` at rust-embedded/cortex-m-rt#167 and rust-embedded/cortex-m-rt#182

`thumbv8m.mainhf` and `thumbv8m.base` are currently being added to Rust CI (for `rustup` to pull) here: rust-lang/rust#59182 (as soon as I have more time)

codido, 2019-03-17:
Thanks @hug-dev, missed those cortex-m-rt PRs, guess I don't need to push my version :)

hug-dev, 2019-04-04:
All the Armv8-M targets are now available with `rustup`:
* `thumbv8m.base-none-eabi`: Cortex-M23
* `thumbv8m.main-none-eabi`: Cortex-M33
* `thumbv8m.main-none-eabihf`: Cortex-M33 with FPU

tarcieri, 2019-05-25:
Will `thumbv8m.base-none-eabi` land on stable soon? I was able to use it on nightly.

thejpster, 2019-05-27:
I assume it's riding the release train, and will turn up in two stable releases time?

joelimgu, 2023-07-26:
Any news for the armv8-r for the R52?

thejpster, 2023-08-01:
It won't happen until someone both needs it and has the ability to do it and has the time or the funding to do it and has the appropriate hardware or simulator to run it on, sadly. Cortex-R is a bit specialist compared to Cortex-M.

adamgreig, 2024-06-11:
I believe these targets are all available now, including `armv8r-none-eabihf`.

closed 2024-06-11 by adamgreig