rust-embedded/wg #90: Impl days at RustFest Paris

japaric, 2018-04-24:
This ticket is to see who's attending the [impl days] after RustFest Paris and to try to organize some
work items (related to the 2018 edition milestone) for them to tackle during the impl days.

[impl days]: https://paris.rustfest.eu/about_impl_days/

For starters, comment here indicating which days you'll attend and what you'd be interested in
working on -- it can be anything: embedded-hal stuff, adding support to a development board, binding
a C library / SDK, writing documentation, polishing a no_std crate, etc.

japaric, 2018-04-24:
I'll be there on Monday and Tuesday and I'd like to focus on the embedded Rust book and the embedonomicon. Also, happy to answer embedded Rust related questions and provide feedback and review embedded crates / projects during that time.

hannobraun, 2018-04-25:
My current plan (which I made before becoming aware of the impl days) is to leave Monday around noon, but I'm considering to change that. I'll have to make that decision closer to the conference though, as I'm not yet sure if circumstances will allow me to stay longer.

Assuming I manage to stay for the impl days, I'll probably work on [LPC82x HAL](https://github.com/braun-robotics/rust-lpc82x-hal) and/or [embedded-hal](https://github.com/japaric/embedded-hal), if left to my own devices. Happy to help out with other work, though.

japaric, 2018-05-15:
Here are some tasks I can mentor, but probably not all of them at the same time.

- Port the code examples in the [Discovery] book to work with cortex-m-rt v0.5.0, and identify the parts of the text that need to be updated to reflect the changes. I can borrow you a [DISCOVERY] board to work on this.

[DISCOVERY]: http://www.st.com/en/evaluation-tools/stm32f3discovery.html
[Discovery]: https://japaric.github.io/discovery/

- Make the [`heapless`] crate compile on stable. cf. japaric/heapless#40. This is not an embedded crate per se but it's a `no_std` crate used by embedded crates like [`cortex-m-rtfm`].

[`heapless`]: https://github.com/japaric/heapless
[`cortex-m-rtfm`]:  https://github.com/japaric/cortex-m-rtfm

- Port the contents of the [quickstart] blog post to cortex-m-rt v0.5 and land them in the [embedded Rust book][] (chapter 2 and the start of chapter 3).

[quickstart]: http://blog.japaric.io/quickstart/
[embedded Rust book]: https://book.rust-embedded.org/blinky/blinky.html

- New to embedded development or embedded Rust? No problem. I'll give you a [DISCOVERY] board and you can work your way through the [Discovery] book and learn some embedded Rust. You help us by giving us feedback about the book and you keep the board at the end of the day. I only have two boards to give away though.

rnestler, 2018-05-16:
I'll be there on both Impl days. I'll probably work on https://github.com/coredump-ch/lpc11uxx-rs.

musitdev, 2018-05-16:
I'll be there on monday only. I've not decided on what I'll work, probably on Discovery. I'll bring one.

rnestler, 2018-05-18:
If anyone is interested in writing embedded drivers for any Sensirion sensor (humidity, temperature, air quality) I can provide you with hardware.
For example a RaspberryPi shield with SHTC1 and SGPC3: [image]

Or AdaFruit breakout boards
 * [SHT31-D](https://www.adafruit.com/product/2857)
 * [SGP30](https://www.adafruit.com/product/3709)

Or Xplained Pro boards:
 * [SHTC1](https://www.digikey.com/products/en?keywords=1649-1022-ND)
 * [SHTW2 dual](https://www.digikey.com/products/en?keywords=1649-1068-ND)

I'll bring some boards anyway, but if you want one for sure just contact me!

japaric, 2018-06-29:
This happened! Thanks to everyone who joined us :heart:. Looking forward to see you all again in the next RustFest!

closed 2018-06-29 by japaric