rust-embedded/wg #92: Does Rust need better support for conditional compilation and/or package configuration?

achan1989, 2018-04-26:
I was looking at FreeRTOS, trying to work out how feasible it would be to implement something similar in Rust.  One of the obvious stumbling blocks is Rust's limited support for conditional compilation.

For a concrete example: in C you can `#define MAX_QUEUES 5` in a source file, or as a command line option to the compiler.  Elsewhere in the code you can use `#if MAX_QUEUES > 2` to emit different implementations based on the value of `MAX_QUEUES`, or omit a feature completely.  And you can use `MAX_QUEUES` to specify the size of an array.

As far as I can see, this would be difficult to do in Rust. The options are:

- `#[cfg(some_flag)]` or `if cfg!(some_flag)`.  Only works with boolean values, and the usable flags are limited to some built-in flags or named package features.
- Perhaps some hacky use of `if env!("some_flag") == "something arbitrary"`?  This allows the use of arbitrary flags and values, but it's pretty disgusting, and a simple `>` comparison is not possible. And build system support is poor -- you would have to write a build script to set up the necessary environment variables, and come up with your own method for specifying what the flags/values should be.
- Use `include!("a_file")` somehow?  Maybe to define macros?
- Use of `const fn` once it is supported, in combination with `env!` or `include!` ?

Furthermore, none of these options (with the exception of `#[cfg(...)]`) are of much use to configure crates that your project depends on.  As far as I can see you can't write a build script that always executes before your dependencies -- and you wouldn't want to have to do that anyway.  So this means you couldn't make a project that depended on a `free_rtos_for_arm` crate that depended on a `free_rtos_core` crate, if you wanted to be able to configure either of those crates.  When you use FreeRTOS you're supposed to copy the kernel source and platform-specific source code into your project, then configure them, but that's hardly a Rust-like solution.

Perhaps an ideal would be for cargo to support a more general-purpose dependency configuration, as a counterpart to `features = ["foo", "bar"]` ?

Am I unaware of some useful Rust feature, or trying to solve this kind of problem in the wrong way?  Or is this genuinely a significant barrier to porting this style of C code?  If so, what changes could be made to resolve this?

nallar, 2018-04-26:
Const generics could help with this once available. Specialisation in combination with const generics should allow for specialised implementations for a specific value.

https://github.com/rust-lang/rfcs/blob/master/text/2000-const-generics.md

achan1989, 2018-05-03:
That looks like it would be very helpful in a lot of embedded scenarios, thanks for the heads up.

Still, it looks like that would be more helpful in code that you write yourself, and less helpful in third-party libraries.  As in, I think this workflow is still not possible:

1. You find a useful library, and add it to your project config.toml
2. You put some "specific value" into config.toml to configure that library
3. Cargo fetches the useful library and builds it a specialised implementation, based on the specific value that you provided

thejpster, 2018-05-05:
I think, in the general case, it would be possible to obtain compile time optimisations through code rather than the horror that is pre-processing. For example, perhaps const generics will offer us something like:

```rust
let queues: [Queue; 5] = magic_rtos::queues::build(5); // maybe it would need an argument
let semaphores: [Semaphore; 4] = magic_rtos::semaphores::build::<5>(); // or a turbofish?
let tasks: [Task; 4] = magic_rtos::tasks::build(); // or maybe it can just be inferred?
```

jamesmunns, 2019-02-04:
By the way, I have a proof of concept of how we can achieve something like this: https://github.com/jamesmunns/mcf-config-demo

achan1989, 2019-04-04:
@jamesmunns that's fantastic :)

adamgreig, 2024-06-11:
Closing this issue as part of today's triage. We _do_ still need better conditional compilation support in Rust, but the embedded WG isn't the place to develop it. Some workarounds exist and have been mentioned in the thread, but anything more than that will need to be a focused push in rust-lang instead.

closed 2024-06-11 by adamgreig