rust-embedded/wg #93: Embedded WG Newsletter #5

jamesmunns, 2018-04-28:
Reporting issue for the next Newsletter, roughly scheduled for `2018-05-12`.

@japaric I would like to move the publishing of the newsletters to Saturdays, as I often do not have time to prepare them during the week.

If we would like to keep the publishing of the newsletters during the week, we may want to officially publish the following Monday. Let me know if you would like to discuss.

kjetilkjeka, 2018-04-28:
[uX](https://crates.io/crates/ux) (pronounced: unsigned X) is a library which aims to handle the problem of using a larger container for non-byte-aligned types once and for all. The only stabilization (1.0) blocker is a sensible way of construction, and due to the policy of not adding features not available for rust built in types (u8, u16, etc) breakage (after 1.0) will only happen if rust stdlib breaks.

adamgreig, 2018-04-28:
* [stm32-rs](https://github.com/adamgreig/stm32-rs) collects SVD files for all STM32 devices, patches them via a YAML DSL to add lots of enumeratedValues and fix bugs, then uses svd2rust to generate device crates which are published per-family ([stm32f1](https://crates.io/crates/stm32f1), [stm32f4](https://crates.io/crates/stm32f4), etc) with feature gates for individual devices. The first released versions are now on crates.io but there's lots of work still to do on the svd patching, publishing the generated documentation (several GB), and possibly reorganising how the crates are collated ([issue 8](https://github.com/adamgreig/stm32-rs/issues/8)).
* [blethrs](https://github.com/AirborneEngineering/blethrs) is a Rust ethernet bootloader for STM32F4 using the Rust [smoltcp](https://github.com/m-labs/smoltcp) TCP/IP stack

jamwaffles, 2018-04-30:
[ssd1306](https://crates.io/crates/ssd1306) is a driver for SSD1306-based OLED display modules. There's an introductory blog post [here](https://wapl.es/electronics/rust/2018/04/30/ssd1306-driver.html).

jscarrott, 2018-04-30:
I've added a new device crate made using svd2rust -> form -> fmt for the [nRF52840](https://crates.io/crates/nrf52840) .I eventually got it working - with a little help from @jamesmunns - on my dev board.

teozkr, 2018-04-30:
I also uploaded a (svd2rust) device crate and corresponding HAL for [stm32f0x0](https://crates.io/crates/stm32f0x0) ([HAL](https://crates.io/crates/stm32f0x0-hal)).

jamesmunns, 2018-05-03:
Note to self: https://github.com/jkarns275/stdgba

rahul-thakoor, 2018-05-03:
[mma7660fc](https://crates.io/crates/mma7660fc) is an i2c driver for mma7660fc-based 3-axis accelerometer. Intro blog post can be found [here](https://rahul-thakoor.github.io/an-i2c-rust-driver-for-mma7660fc-based-3-axis-digital-accelerometer/)

jamesmunns, 2018-05-03:
Note to self: https://www.219design.com/rust-on-the-stmicro-nucleo/

jamesmunns, 2018-05-04:
Note to self: https://nercury.github.io/rust/embedded/experiments/2018/04/29/rust-embedded-01-discovery-vl-flipping-bits.html

whitequark, 2018-05-05:
Long ago, I wrote a library that lets you put UTF-8 text in a ring buffer, and extract it from there as a `&str`: github.com/whitequark/rust-log_buffer

krk, 2018-05-05:
I have created a svd2rust device for nRF51822 @ https://github.com/krk/nrf51822

therealprof, 2018-05-05:
@krk NB: There're at least two more of those now, one here by me: https://crates.io/crates/nrf51 and an outdated one by japaric. I'd be happy to let you collaborate on that one and/or the hal implementation I've created: https://crates.io/crates/nrf51-hal .

jamesmunns, 2018-05-09:
https://www.tockos.org/blog/2018/mmio-registers/

memoryruins, 2018-05-13:
Proptest 0.7.0 now has no_std and alloc support.
https://github.com/AltSysrq/proptest

ZackPierce, 2018-05-13:
@memoryruins That was fun, and my work on doing the same for regex and its ecosystem is [well under way](https://github.com/rust-lang/regex/issues/476) , though currently on pause as we consider whether the use of the `alloc` facade is the best way to build for the future.

memoryruins, 2018-05-13:
@ZackPierce wow! it's well appreciated :)

closed 2018-05-23 by jamesmunns