rust-embedded/wg #100: Providing help

bkchr, 2018-05-20:
Hi, I would like to help the embedded wg with implementing some issues. I looked over the current issues in this repository, but most of them are already being worked on. Is there something where I could help?
It would be nice to do something in rustc, but I'm open for almost everything :)

hannobraun, 2018-05-24:
I feel like I don't have a good overview over everything that's going on in the ecosystem right now, so the following are just some suggestion from my own perspective.

I think the [embedded-hal](https://github.com/japaric/embedded-hal) ecosystem can always use help. Here are some concrete examples of what you could do:
- [Drivers](https://github.com/rust-embedded/awesome-embedded-rust#driver-crates): Write drivers for hardware that you're interested in, or test/improve existing drivers. If that gives you any insights about the embedded-hal API, give feedback on the [issue tracker](https://github.com/japaric/embedded-hal/issues).
- [HAL implementations](https://github.com/rust-embedded/awesome-embedded-rust#hal-implementation-crates): The work here is similar to the driver work. Create an embedded-hal implementation for platforms that interest you, or test/improve existing implementations. Provide feedback on the embedded-hal issue tracker.
- Participate in embedded-hal discussions. The [issue tracker](https://github.com/japaric/embedded-hal/issues) (and some [pull requests](https://github.com/japaric/embedded-hal/pulls)) have unresolved discussions. Quite often, there are open questions about what kind of hardware capabilities exist in the wild, so other perspectives are always helpful.
- Help with embedded-hal management. The embedded-hal project currently suffers from a lack of maintainer bandwidth. Many discussions and pull requests have stalled because of that. I expect that I'll be able to increase my participation in the very near future, but regardless of that, any help in that area would be great. Summarizing stalled discussions, pinging people as required, that kind of stuff.

Aside from embedded-hal, here's some other stuff that comes to mind:
- [Generate a better cortex-m API from an SVD file](https://github.com/japaric/cortex-m/issues/7): @docbrown [seemed interested](https://github.com/rust-lang-nursery/embedded-wg/issues/46#issuecomment-367856566), but I'm not sure if anything came of that. I'd love to see this happen!
- (Sorry, this one is selfish): Look at my [svd2rust proposal](https://github.com/japaric/svd2rust/issues/213), give feedback, maybe finish my partial implementation? This is something I'd love to see, but have limited bandwidth to push ahead on myself.

I'm sure there are many other things that you could work on. If you're interested in something that's already being worked on, maybe talk to the person doing the work, to check if they need help or someone to take over.

docbrown, 2018-05-24:
@hannobraun I put together [a generic SVD](https://gist.github.com/docbrown/7bad626215a73a3a533b36d11886edac) from the reference manual. There were some parts that I had trouble mapping into the SVD model and some features were missing from svd2rust at the time.

I ended up continuing my project in C, so I probably won't be updating the SVD in the foreseeable future. Feel free to do whatever you want with it.

hannobraun, 2018-05-25:
@docbrown Thank you!

bkchr, 2018-05-26:
@hannobraun thanks for your detailed listing. There should definitely be something for me.

closed 2018-12-10 by bkchr