rust-embedded/wg #106: Emit a stack size section using LLVM and parse this data

whitequark, 2018-06-15:
From the manual:
```
Emitting function stack size information
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

A section containing metadata on function stack sizes will be emitted when
``TargetLoweringObjectFile::StackSizesSection`` is not null, and
``TargetOptions::EmitStackSizeSection`` is set (-stack-size-section). The
section will contain an array of pairs of function symbol values (pointer size)
and stack sizes (unsigned LEB128). The stack size values only include the space
allocated in the function prologue. Functions with dynamic stack allocations are
not included.
```

japaric, 2018-06-29:
@whitequark how does this relate to #79 (Expose the -warn-stack-size= LLVM option)? At first glance it sounds like using the stack size section (as described in this issue) would give us more fine grained information and that we should do this instead of #79.

japaric, 2018-06-30:
PR rust-lang/rust#51946 exposes this LLVM feature

japaric, 2018-09-29:
Update:

The `-Z emit-stack-sizes` flag is available since nightly-2018-09-27 as an experimental feature. The  (LLVM) feature only supports the ELF object format (as of LLVM 8, AFAICT) and the rustc flag is documented in [the unstable book](https://doc.rust-lang.org/nightly/unstable-book/compiler-flags/emit-stack-sizes.html).

The [`stack-sizes`](https://github.com/japaric/stack-sizes) crate can be used as a library to parse this information and also contains tools to print the information in human readable format.

japaric, 2019-03-04:
Update:

[`cargo-call-stack`](https://crates.io/crates/cargo-call-stack) is a whole program stack analysis tool that uses the `-Z emit-stack-sizes` feature / flag. 

I think we can close this issue since the feature is exposed in rustc and there are libraries to parse the emitted data and end user tools.

It's unclear if this flag will ever be stabilized or when; there's no precedent for exposing stable interfaces to llvm internals from rustc (except for 1.0 mistakes like llvm-args) and I'm not aware of any plans to do anything like that in the near future. If I heard of any movement on that front I'll report back here.

closed 2019-03-04 by japaric