rust-embedded/wg #107: Logo for Embedded WG

jamesmunns, 2018-06-19:
We would like to have a logo for use with newsletters, release announcements, stickers, etc.

Ideally, we would like to have something by 2018-06-23.

thejpster, 2018-06-19:
We should ensure that commercial companies (e.g. chip vendors) are free to use the logo in any Rust SDKs or driver packs they may distribute in the future.

Aaronepower, 2018-06-19:
Here's a logo I made in some spare time.
[image]

thejpster, 2018-06-22:
> Agreed, "general usage requirements" would be good to have in writing with the logo. We can borrow pretty heavily from https://www.rust-lang.org/en-US/legal.html, and maybe customize for our expected use cases.

Yeah, that's the thing. I would have thought making a kick-ass SDK to sell more chips would count as commercial, and so they'd need to get permission to use the Rust logo. At least that was my interpretation.

thejpster, 2018-07-04:
So I would vote for the 'R' logo on a chip. Who should we reach out to to get clarification on companies using this logo for their products (e.g. BSPs written in Rust). And, if it turns out that's a no-go, should we just use this logo for our purposes anyway, because it's really cool?

therealprof, 2018-07-04:
I totally agree with @thejpster here.

jamesmunns, 2018-07-08:
Sounds like we have two votes for @Aaronepower. If anyone else wants to make any comments, let me know. We will discuss at the next embedded WG meeting (about a day from now) - https://github.com/rust-lang-nursery/embedded-wg/issues/111, and hopefully get this decided then.

jamesmunns, 2018-07-15:
Based on discussion here, we have accepted the logo from @Aaronepower. Progress on this will now be tracked in this PR: https://github.com/rust-lang-nursery/embedded-wg/pull/120.

Thank you so much to @TNkemdilim and @Aaronepower for submitting logos!

closed 2018-07-15 by jamesmunns