rust-embedded/wg #108: Collect commercial testimonials and well polished examples for the revamped website / embedded WG page

japaric, 2018-06-19:


thejpster, 2018-06-19:
I've made https://www.cambridgeconsultants.com/insights/presentation-mozillas-rust-and-why-we-love-it and https://vimeo.com/224912526

hannobraun, 2018-06-20:
Have we decided that we're going to include Rust on Embedded Linux in this? Cause I know of a few companies doing Embedded Rust stuff on top of Linux.

hannobraun, 2018-06-25:
> Have we decided that we're going to include Rust on Embedded Linux in this?

We talked about this question in the WG meeting just now, but I don't think we managed to make a decision before the meeting ended. I think there was concensus that showcasing bare-metal stuff (which is something that not many languages can do) is very important and should be the focus.

aep, 2018-07-01:
All my companies are built on rust. (captif.io and devguard.io) All open source, all embedded. Sure would love to provide some praise to the rust gods.

lexxvir, 2018-07-03:
Hi there,

At [Terminal Technologies](http://www.termt.eu/), we use Rust for the our new product, now it is at development and certification stage.

We have Cortex-M3 CPU with 1Mb ROM and 384Kb RAM, so despite it is truly embedded device, we have enough resources to write 'regular' application code.

Our application is mix of C/C++ core libraries and domain-specific logic on Rust.

Some facts about used crates:
 - [FreeRTOS](https://crates.io/crates/freertos_rs) for task scheduling and synchronization
 - [Protobufs](https://crates.io/crates/quick-protobuf) for external control
 - [core_io](https://crates.io/crates/core_io) crate to 'bind' IO-related things together

It worth noting, that it's our first experience using Rust for real product and now we are happy that we've decided using Rust. For now we have not any issue with UB or memory corruption and it gives us confidence. Also really cool that we can use modern nice (even more nice than Java, Python) language in embedded space where usually there is no alternatives to C/C++.

Glad to give you more details if you wish.

mbr, 2018-07-03:
Well, regardless of whether you decide to include embedded Linux or not, here's something from us:

<hr>

Since its founding, [49nord](https://49nord.de) has been using Rust exclusively for its variety of embedded products. It enables us to ship software faster and more correct than we thought possible. Thanks to Rust, we can take memory safety for granted, while other benefits of a zero-overhead language with a sophisticated type system help us develop maintainable software.

We deliver multiple products based on embedded Linux. Being able to natively interface with the kernel/libC without overhead has been a huge boon, letting us treat native Linux data structures as if they were actual Rust types. On the other hand, adding small safety wrappers around these interfaces helps us avoid a lot of pitfalls that plague C programmers regardless of experience.

Most other languages can only aspire to the out-of-box tooling that Rust offers, cross-compilation, testing, managing multiple compiler versions and adding dependencies is a breeze. A lot of functionality that would require third party libraries in other close-to-the-metal languages ships in the high quality standard lib.

Rust makes our customers happy, as well as our engineers.

hannobraun, 2018-07-03:
@mbr Thank you for the testimonial!

In the last meeting, we decided to also include Embedded Linux (which is why I directed you here; sorry, should have been more clear). We'd like to focus the website on bare metal as much as possible, as that's more of a unique selling point compared to other languages. However, at this point it's unclear how much commercial bare-metal development is going on out there, so we're very interested in Embedded Linux testimonials, too. The more the merrier, I'd say.

adamgreig, 2018-07-25:
At [Airborne Engineering Ltd](http://ael.co.uk) we recently used Rust to write an Ethernet bootloader, [blethrs](https://github.com/airborneengineering/blethrs), for our in-house data acquisition system. Manually updating the firmware on every card in the system was getting old fast! We were able to put the first version together in just a couple of days with none of the usual bug hunting you'd expect from traditional embedded languages, and are now using it in production. The excellent out-of-the-box tooling, including build and dependency management, and growing range of embedded libraries such as the [smoltcp](https://github.com/m-labs/smoltcp) TCP/IP stack makes for a very productive alternative to C. Rust is a promising language and we're excited to use it for our future projects, embedded and otherwise.

rnestler, 2018-08-15:
At [Sensirion](https://www.sensirion.com/) we recently used Rust to create an embedded demonstrator for Sensirion's [Particulate Matter Sensor](https://sensirion-automotive.com/products#PM2_5). Due to the easy cross-compilation and the availability of many high quality crates on crates.io we quickly ended up with a fast and robust demonstrator.

During the development we published a [LIN bus](https://github.com/Sensirion/lin-bus-rs) crate with a master implementation and a driver trait. For rapid prototyping we also implemented the [driver Trait over serial](https://github.com/Sensirion/lin-bus-driver-serial-rs) to use with a Raspberry Pi.