rust-embedded/wg #109: Embedded WG meeting, June 25

japaric, 2018-06-20:
[6-7 PM CEST][0] on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-6-25,240,b8jj

Agenda.

- Triaging issues on the [2018 edition milestone].

[2018 edition milestone]: https://github.com/rust-lang-nursery/embedded-wg/milestone/1

- Commercial testimonials. cf. #108 

- [your topic here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

rudihorn, 2018-06-20:
I would like to suggest a topic on somehow improving the way small but annoying issues are handled. Currently it seems like PR's and issues are piling up with e.g. the STM32F103xx-hal some other crates and I feel like there should be more pushing to resolve such issues, by for instance delegating more people who can merge PR's etc. 

A few examples of issues I am talking about:
- STM32F103 not on crates.io (preventing any further crates with examples from being pushed)
- Embedded-HAL having any forms of input pins marked as unproven / general difficulty of being able to use input pins (also in Linux-embedded-hal)
- Broken docs PR's not being pulled (https://github.com/japaric/stm32f103xx-hal/pull/85)
- I2C missing / despite some PR's seemingly being available

It would be nice if we could get the very basics working, even if some of it does have to be broken at some stage. I get the feeling some of this may be related to @japaric being very busy, but in this case it would be nice if we could find others to merge pull requests etc. for some of these "core" projects.

jonas-schievink, 2018-06-20:
@rudihorn https://github.com/rust-lang-nursery/embedded-wg/issues/46 is related, but doesn't mention the crate.

japaric, 2018-06-28:
This meeting happened! The logs have been pushed into this repository.

@rudihorn We didn't have time to discuss the issue you brought up but we have scheduled it for the next meeting (which is a video meeting) cf. #110.

closed 2018-06-28 by japaric