rust-embedded/wg #110: Embedded WG meeting, July 2

japaric, 2018-06-28:
[6-7 PM CEST][0] on Google hangouts

[0]: https://everytimezone.com/#2018-7-2,240,b8jj

Agenda.

- Rust 2018: the home stretch. cf. https://internals.rust-lang.org/t/rust-2018-the-home-stretch/7810

- Maintaining the cortex-m ecosystem. cf. #46, https://github.com/rust-lang-nursery/embedded-wg/issues/109#issuecomment-398783889

- Triaging issues on the [2018 edition milestone].

- Commercial testimonials. cf. #108 

[2018 edition milestone]: https://github.com/rust-lang-nursery/embedded-wg/milestone/1

korken89, 2018-07-01:
Hi, will the link to the meeting be shared here when it's approaching?

japaric, 2018-07-03:
This meeting happened! Minutes are up [here]. The agenda for the next meeting is in #111.

[here]: https://github.com/rust-lang-nursery/embedded-wg/blob/master/minutes/2018-07-02.md

---

@korken89 you made a question on IRC. Answering here since you are not online over there.

> What does it take to join the WG team, and do you happen to have a reference for the how the
> Embedded WG works internally? I only have experience of C++'s SG14 (embedded / low-latency trading
> / gaming) WG where all meeting were open, so closed meetings strikes me as a bit odd, but I am not
> very familiar with how Rust's WGs works yet.

The WGs operate a bit differently from Rust teams. How a WG operates is entirely up to its lead. For
the embedded WG, which I lead, we have a video meetings the first week of each month which only
the ["team"] attends -- minutes for these meetings are published in this repository. All other
meetings are done in IRC and are free for everyone to attend. Discussions in the issue tracker of
this repository are, of course, also open to anyone.

["team"]: https://github.com/rust-lang-nursery/embedded-wg#roster

Rust teams tend to operate more uniformly: most of them only have video meetings, some have a 50/50
mix of video and IRC meetings (mostly for triaging issues) and one or two only have IRC meetings.
Rust teams also have the job of reviewing and merging [RFCs] relevant to their area (compiler
changes, language changes, tooling, etc.).

[RFCs]: https://github.com/rust-lang/rfcs

That being said you don't have to be in the WG "team" to participate in the GH / IRC discussions,
propose stuff (just open an issue here), or work on existing issues. The "team" has a few more
responsibilities than other participants: writing the biweekly newsletter, triaging issues (i.e.
posting updates) in this repo and writing meeting minutes. And the lead has the extra task of
communicating more directly (e.g. reporting needed language features and critical bugs) with the
other Rust teams.

closed 2018-07-03 by japaric

korken89, 2018-07-03:
Thanks you for the clarification! Helps a lot to understand how it all operates.