rust-embedded/wg #113: mutex gives a &T not a &mut

thejpster, 2018-07-06:
I was writing an interrupt handler and wanted to gate access to a static. The `bare_metal::Mutex` looks ideal, but it gives an immutable reference instead of a mutable one. That seems to defeat the point of having a Mutex.

Am I missing something? What's the canonical mechanism for accessing data in an ISR and ensuring the main thread can only access the data when the ISR is disabled? Is RTFM the only answer?

korken89, 2018-07-06:
RTFM is a good answer, but there are other methods. You can use an atomic, or a lock free queue, to name a few depending on what your requirements are.

Usually looking to mutexes in an embedded system though is often a sign of not using event based design methodologies, with the vast majority being event based.
With that said, RTFM is more or less as good as you can make it as you have the MCU do the context switch for you through the interrupt controller :)

pftbest, 2018-07-06:
Rust's Mutex does two jobs at the same time, first it guarantees that your access to memory is not racy, and secondly is guarantees that you access is _unique_. Rust's Mutex is not recursive, you can't lock it twice in the same thread, that is why it is able to give you `&mut` reference without breaking aliasing rules.

Bare metal Mutex only gives you the first guarantee, but it doesn't give you the second one. You can lock it how many times you want in the same thread, so it can only give you a shared reference.

To solve this problem, you can put Cell or RefCell inside your mutex.

therealprof, 2018-07-06:
@thejpster This is how I do it: https://github.com/therealprof/stm32f042-hal/blob/master/examples/flash_systick.rs

japaric, 2018-08-10:
I believe the original question has been answered so I'm going to close this issue.

closed 2018-08-10 by japaric