rust-embedded/wg #115: Book: Review Outline

jamesmunns, 2018-07-10:
This issue is for decisions on the layout or outline of the book contents.

The outline can be seen as [markdown], or [rendered].

[markdown]: https://github.com/rust-lang-nursery/embedded-wg/blob/master/books/embedded-rust-book/src/SUMMARY.md
[rendered]: https://book.rust-embedded.org/

We will open new issues for managing chapters and sections of the book.

jamesmunns, 2018-07-10:
@thejpster proposed moving Sections 7.1-7.4 into a new section (how to think in embedded rust), to be placed after section 2.

jamesmunns, 2018-07-10:
@japaric proposed splitting up Section 3 (embedded-hal) into two Sections, one covering Portablility, and one covering Static Guarantees

japaric, 2018-07-16:
> Tock-OS Ecosystem 

Are Tock userland apps working on stable? The books should only contain stuff that works on stable.

I propose we add a section on concurrency briefly talking about Send and Sync (which are well covered in the Rust book) and then go over the concurrency systems / pattetrns that exists: bare metal (static + interrupt handlers), Real Time For the Masses (tasks, message passing) and, if stable, Tock (multiprocess and cooperative within a process).

thejpster, 2018-07-17:
Will this be updated with the revised layout as discussed at the WG meeting, or is that tracked elsewhere?

ryankurte, 2018-07-17:
@jamesmunns there was some discussion in #56 and from there #112 about building _Running and Debugging_ into a good tool-independent resource (which I definitely support), and in which case I would suggest it could be a separate section?

japaric, 2018-07-17:
@thejpster the issue description has bee updated with the revised outline.

@ryankurte 

the embedded book is aimed at people familiar with embedded development. IME running and debugging Rust programs doesn't look much different from running and debugging C programs -- the same tooling is used, the format is the same (ELF or binary) -- so I'm not sure if such section should be in the embedded book (i.e. it could live elsewhere). It also seems more efficient (less work) to point to existing resources (e.g. Discovery for OpenOCD + GDB, C resources for other tools) and just cover Rust specific stuff: the compiler produces ELF files and places them in target/$TARGET/$binary; use objcopy to get a binary file out of that; Rust symbols look like `foo::bar::$function::$hash`; and so on. 

Of course if someone commits to writing documentation for every combination of flashing / debugging tool out there then we would be happy to publish such documentation somewhere.

thejpster, 2018-07-20:
LGTM

kjetilkjeka, 2018-07-20:
I've already mentioned bobbin-cli in other issues. This tool is unique to rust, integrated well with existing tools and makes debugging and running rust programs a much more pleasant experience than c. I think this tool should be mentioned in the book. It do not need its own section and csn be included into "Introduction" or "Blinking Your First LED"

jamesmunns, 2018-07-29:
Hey all, I've started implementing this new outline in https://github.com/rust-lang-nursery/embedded-wg/pull/133. I would appreciate any review.

If you think any of my old notes should be preserved in the new outline, please feel free to let me know.

jamesmunns, 2018-08-04:
I've written up [my plan](https://jamesmunns.com/blog/working-on-the-book/) for getting the book shipped this year. TL;DR, I am going to commit to shipping one chapter every two weeks until the end of the year.

If anyone has time to contribute, review, or expand on anything towards the book. It would be very welcome. Hopefully I should be much more active on this issue over the next 18 weeks :)

andre-richter, 2018-08-04:
Whoa that's an ambitious plan. Let's see if we can keep you motivated :). I'll try to help with some review work whenever I can in my abundance of free time... 😕

jamesmunns, 2018-08-04:
Yeah, We'll see how far I get! I signed up for this a while back, and life happened. Hopefully after RustConf (in about two weeks), I'll be able to commit more of my "free" cycles. I don't think two weeks is enough to make each chapter a master work, but hopefully enough to ship our first version of the book, and functional enough for people new to working in embedded rust.

I definitely think I would benefit from any additional help, but I wanted to put my commitment out there, hoping other people would benefit from seeing my schedule.

https://github.com/rust-lang-nursery/embedded-wg/pull/133 is up for review if you have any cycles now :)

japaric, 2018-08-07:
@jamesmunns awesome! I'll be assigning all the book issues to you then. I will try to submit some braindumps for memory management, singletons and concurrency this week.

japaric, 2018-08-07:
Oh and we should move out the book ASAP to have a dedicated issue tracker.

japaric, 2018-08-07:
And that means migrating the book issues in this issue tracker to the other one. (I don't know how to do that :-).

japaric, 2018-08-10:
This issue was moved to rust-embedded/book#2

closed 2018-08-10 by japaric