rust-embedded/wg #121: Embedded WG Newsletter 8

jamesmunns, 2018-07-15:
Reporting issue for the next Newsletter, roughly scheduled for `2018-07-29`

jamesmunns, 2018-07-15:
Reach out to people behind https://uavcan.org/Implementations/Uavcan.rs/ ?

jamesmunns, 2018-07-18:
https://www.reddit.com/r/rust/comments/8zk03w/cc1101_crate_can_now_receive_radio_packets/

kjetilkjeka, 2018-07-18:
> Reach out to people behind https://uavcan.org/Implementations/Uavcan.rs/

@jamesmunns I'm the one that has been working on the uavcan.rs project. I guess it's me and/or Pavel Kirienko (the uavcan protocol maintainer) you're looking for. Anything I can help with?

japaric, 2018-07-21:
The Rust port of MUSL libm is done. The [libm](https://github.com/japaric/libm#libm) crate can be used in standalone mode to get math support in #![no_std] code.

paoloteti, 2018-07-21:
I just start to add ARM DSP Intrinsics on rust/stdsimd on **official repo.** (2 PR and ~15 intrinsics)
List is a bit long, so any help is appreciated.
I use to test DSP Intrinsics on Cortex-R boards and of course on QEMU/CI, but I don't own any Cortex-M board. Hope someone else can test on real hw.

droogmic, 2018-07-22:
https://droogmic.github.io/microrust

I have been working on a rewrite of the Discovery book, targeting the micro:bit. 

It is more than just a simple port though, so it may be of some interest to the WG when they write their embedded book. The benefit vs Discovery is an attempt at a simpler and more beginner friendly, as well as only requiring one piece of hardware, the micro:bit itself (which should be easier to source and cheaper).

It is now at what I consider to be a first "beta" version, so there is still a lot of additions to make.

japaric, 2018-07-23:
Related to the ARM DSP intrinsics: The CMSIS intrinsics have also landed in nightly (cf. rust-lang-nursery/stdsimd#518). We should ask the community to test those too. Both the DSP and CMSIS intrinsics should appear in the [nightly API docs](https://doc.rust-lang.org/nightly/core/arch/arm/index.html) but they don't show up there yet for some reason. If the nightly docs get updated before the weekend let's include a link to that in the newsletter; if they don't then we should include a list of the intrinsics in the newsletter.

japaric, 2018-07-27:
We are replacing some of the f32 functions in libm with ports of newlib's implementations to get better performance on devices that don't support double precision floats. If you are interested in helping us with the port leave a comment in [one of these issues](https://github.com/japaric/libm/milestone/3).

japaric, 2018-07-27:
Regarding to https://github.com/rust-lang-nursery/embedded-wg/issues/121#issuecomment-406811822 and https://github.com/rust-lang-nursery/embedded-wg/issues/121#issuecomment-406936490, https://github.com/rust-lang-nursery/embedded-wg/issues/63#issuecomment-408329142 has a more detailed update on the topic of stable Cortex-M/R specific instructions.

jamesmunns, 2018-07-27:
Initial CI harness has landed, and `thumb*` tests are under development by @sekineh and @nerdyvaishali https://github.com/rust-lang/rust/pull/52465

jamesmunns, 2018-07-27:
Hey @kjetilkjeka, I just ran across `uavcan.rs` while looking into some of the quadrotor projects from the last newsletter. If you have a post anywhere that explains what the goal of the project is, or how the process has gone adding support for Rust, it would be great to share it!

Thanks for reaching out

japaric, 2018-07-27:
#### Help wanted

We are looking for someone / people to help us write an RFC on stabilization of the ARM Cortex intrinsics in core::arch::arm. Details in https://github.com/rust-lang-nursery/embedded-wg/issues/63#issuecomment-408509178.

jamesmunns, 2018-07-29:
https://t.co/SMRcFeWJxc

jamesmunns, 2018-07-29:
https://twitter.com/roosmaa/status/1022457886885244929

Rust on a Ledger Nano

jamesmunns, 2018-07-29:
https://twitter.com/ag_dubs/status/1022191996293865472

cargo-generate - useful for Embedded Rust Projects?

jamesmunns, 2018-07-29:
https://twitter.com/therealjpster/status/1021873858826194947

Monotron looking for feature suggestions

closed 2018-07-30 by jamesmunns