rust-embedded/wg #123: Embedded WG meeting, July 16

japaric, 2018-07-15:
[6-7 PM CEST][0] on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-7-9,240,b8jj

Agenda.

- Planning work on the embedded Rust book. cf. #56
  - Final outline. cf. #115 #116 #117 #118 #119
  - Prioritization of topics / sections. Which topics *must* be done and polished for the edition release? Which topics would be nice to have but are not critical?
  - How to have the community contribute to the book? How to commit to writing a section / about a topic? Where to send feedback about the book?

- Triaging issues on the [2018 edition milestone](https://github.com/rust-lang-nursery/embedded-wg/milestone/1).

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

hannobraun, 2018-07-16:
There's an agenda item from last wee that we didn't get to:
* Maintaining the embedded / Cortex-M ecosystem. cf. #46. @japaric will come up with some plan / structure.

japaric, 2018-07-16:
@hannobraun yes, we'll cover it in a later meeting because we have to cover the website today.

gbip, 2018-07-17:
I think that this is a very important agenda item : as the community get more involved into embedded development we really need to have some kind of structure for maintaining crucial crates.

japaric, 2018-07-23:
This meeting happened. Logs are [here](https://github.com/rust-lang-nursery/embedded-wg/blob/master/minutes/2018-07-16.irc.log). Issue for next meeting [here](https://github.com/rust-lang-nursery/embedded-wg/issues/129).

closed 2018-07-23 by japaric