rust-embedded/wg #126: Book section: Tips for embedded C / C++ developers

japaric, 2018-07-17:
We agreed on having a section containing tips for embedded developers that have a C / C++ background. For example, they may be used to using for loop + indexing on arrays but it's more efficient to use iterators in Rust so they should use that instead.

Anything else that should be covered? cc @thejpster

diwic, 2018-07-21:
I wrote [an article](https://github.com/diwic/reffers-rs/blob/master/docs/Pointers.md) about some Rust patterns that replaces C pointers. Might be helpful in this context too?

korken89, 2018-07-21:
One thing that is quite popular now is to use `template meta-programming` in C++ to configure hardware at compile time with guaranteed optimization. This is one part that kept me from coming to Rust for a long time, and I still have not found a way to do this well in Rust.

In C++ there is a continuous move towards `constexpr` functions, which (under certain requirements) are guaranteed to be evaluated at compile time. In Rust, it still seems to be dependent on the optimizer to evaluate an entire function, plus this causes issues that I have not been able to find good solutions for when generating data structures at compile time.
Granted that a lot can be done in `build.rs` and using `include!`, but finding good examples and how/when to use it was not trivial in my experience.

If we can answer these a bit I believe it would be good for convincing the embedded C++ (library) developer.

therealprof, 2018-07-21:
@korken89 No idea where you got that from. It's pretty much exactly the other way around: In Rust variables are const by default and everything is fully propagated through the compilation passes meaning that most functions will just turn into hot air leaving only few instructions behind. All automatically and per default without fiddling with functions trying to convince the compiler to that something is const. It's truly zero cost abstractions something that C++ never really achieved.

Mind you, the incredibly optimised binaries is something you will only get with release builds; debug builds are a bloat nightmare.

The problems are more on the opposite side of the spectrum: The safety guarantees of Rust make it rather cumbersome to share data, work with self-referential data structures or (especially in embedded areas) work with dynamic amounts of data.

therealprof, 2018-07-21:
@korken89 

> I am aware of what you speak, but it is up to the user to make sure that a function will be evaluated at compile time. 

In Rust? No. It's not even possible to ensure that. (Well, with the exception of const functions but they're only usable for initialisation of static data and often not even that).

>> It's truly zero cost abstractions something that C++ never really achieved.
> Here you are misinformed, 

No, not quite. I'm fully aware that this is a goal of C++ and **can** be achieved by expert programmers but is except for a few exceptions not done due to being extremely time consuming. Also with C++ that works only inside a single compilation unit or when excessively using code in header files (which a lot of other people, me included, frown upon), which is neither possible in Rust nor necessary.

> Back to the question at hand now, as coming from a C++ perspective these questions are still active and I still believe that it should be covered to some extent.

I also have some C++ background (and consider myself an expert C programmer) and have experienced the exact opposite of what you're trying to say needs fixing/documentation. Proper zero cost abstractions in C++ require a tremendous amount of effort to implement and ecosystem support (which often simply does not exist) and just happen to work out of the box in Rust, which is one of the main reasons why I completely gave up on embedded C/C++ development.

I'd really like to learn more about the problems you're facing because I have a strong suspicion there's an easy fix to your problem and yes, that's something that should go in the book, too.

korken89, 2018-07-22:
@therealprof  

> Proper zero cost abstractions in C++ require a tremendous amount of effort to implement and ecosystem support

Indeed, and I have been creating these kind of abstrations through header-only libraries as you said.
This is also what has taken me towards Rust, as it is a great pain in C++ - too many ways to subtly fail and generate non-optimal assembler (which also is why [Compiler Explorer](www.godbolt.org) is my main tool to evaluate the generated assembler in). 

> I'd really like to learn more about the problems you're facing because I have a strong suspicion there's an easy fix to your problem and yes, that's something that should go in the book, too.

@Ixrec really hit it on the nail here, thanks for the clarifications! The building block that I use in libraries for embedded system is `constexpr` to guarantee compile time evaluation - and guaranteeing compile time evaluation is really important when providing strong guarantees in libraries. 
The second part is generating compile time tables and data structures, but I think using `build.rs` is sufficient here. Examples could include: generating CRCxx tables as there are a large amount of different CRC polynomials, generating twiddle factor tables for FFT implementations based on the input sizes and if real/imaginary inputs will be available.

I was not aware about `const fn` and `const generics`, the RFCs were a really good read! Knowing this, I would probably put a note for the C++ users to say how to get constant evaluation now (to the extent it allows with `const`) and what we can expect in the future to still the questions for now.

therealprof, 2018-07-22:
@korken89 You might want to have a look at macros, too, to generate tables and/or runtime expressions at compile time.

korken89, 2018-07-22:
@therealprof Thanks, I'm already there :) Trying to learn procedural macros now.

Ekleog, 2018-08-02:
IMO, how to use the build system for cross-compiling to embedded systems is the most important thing to put in this section, if it's not already somewhere else. At least that's what I had most issues with when switching to Rust for embedded development :) I guess that's going to be better with upstreaming of xargo into cargo, though.

jamesmunns, 2018-08-02:
@Ekleog check out https://github.com/rust-lang-nursery/embedded-wg/issues/48

japaric, 2018-08-10:
This issue was moved to rust-embedded/book#9

closed 2018-08-10 by japaric