rust-embedded/wg #128: Non-code support

FlorianUekermann, 2018-07-18:
I couldn't find pointers on how to support the embedded efforts in Rust via hardware donations or monetary support for projects and people. Not sure if this is the right place for this or if there is even interest.

hannobraun, 2018-07-19:
@japaric is on Patreon: https://www.patreon.com/japaric

I don't know if there are any other embedded Rust devs on Patreon, but any money going to @japaric is certainly well-spent.

If you're looking to sponsor specific work, like writing drivers or HAL crates, I'm sure there are developers willing to do take on projects like that (me among them).

FlorianUekermann, 2018-07-19:
> @japaric is on Patreon: https://www.patreon.com/japaric
Done. Thanks, I didn't see that. It would be nice to reference these things somewhere more obvious (or did I just miss it).

I don't think I can sponsor specific work on my own and unfortunately don't have the time to make very significant code contributions at the moment.
I don't know how this is typically done, but if there was some space where people can post what they would like to fund and team up with others, I would be interested (not sure how many other people are interested in that though).

On the specific hardware side I am mostly interested in better support for EFM32HG right now. Especially a USB stack would be really great. If someone is seriously interested in doing something on that front: I have a spare [dev board](https://www.silabs.com/products/development-tools/mcu/32-bit/efm32-happy-gecko-starter-kit) and could maybe part with a [tomu](https://tomu.im/) or two.

hannobraun, 2018-07-19:
> Done. Thanks, I didn't see that. It would be nice to reference these things somewhere more obvious (or did I just miss it).

For Rust in general, there's [a list](http://aturon.github.io/sponsor/). I'd totally support adding an equivalent list (possibly including commercial offerings) to this repository (or another place where it would be visible). I'm a bit loath to push for such a list though, since it would be quite self-serving.

> I don't think I can sponsor specific work on my own and unfortunately don't have the time to make very significant code contributions at the moment.

Yes, I think that options makes sense for companies that intend to use whatever they sponsor commercially. I wanted to mention it for the sake of completeness.

> I don't know how this is typically done, but if there was some space where people can post what they would like to fund and team up with others, I would be interested (not sure how many other people are interested in that though).

I've heard of websites that do that. I think [Bountysource](https://www.bountysource.com/) is one, and there are certainly others. I have zero experience with anything like that though, so I can't make a recommendation.

> On the specific hardware side I am mostly interested in better support for EFM32HG right now. Especially a USB stack would be really great. If someone is seriously interested in doing something on that front: I have a spare dev board and could maybe part with a tomu or two.

That's a bit outside my immediate area of interest, so I don't plan to work on it. But good on you for offering to provide hardware!

therealprof, 2018-07-19:
@FlorianUekermann @hannobraun Actually that's a splendid idea. I also have both a huge backlog of things I want to do/get done and a ton (I haven't weighted it but it should be literally close to a metric ton) of hardware I could donate to interested parties.