rust-embedded/wg #129: Embedded WG meeting, July 23

japaric, 2018-07-23:
[6-7 PM CEST][0] on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-7-23,240,b8jj

Agenda.

- Finalizing the style of the web page. See https://github.com/rust-lang-nursery/embedded-wg/issues/49#issuecomment-406986825

- Triaging issues on the [Preview 2 milestone](https://github.com/rust-lang-nursery/embedded-wg/milestone/2).

- [your topic goes here]

Backlog.

-  Maintaining the embedded / Cortex-M ecosystem. 

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

jamesmunns, 2018-07-23:
I will probably not be able to attend today's meeting. A status update for CI/IRR: the first PR to set up the test harness has landed: https://github.com/rust-lang/rust/pull/52465, so the next step is to start adding tests. We hope to start opening PRs for these initial tests this week.

We also need a couple of code examples and/or crates we should be checking in the CI. I am open to suggestions, probably something like `cortex-m` and `cortex-m-rt` are a good place to start.

I also ordered stickers, they should hopefully show up this week :). Shoot me an email with your address if you want a couple (open to anybody working on embedded rust, not just working group, mailing from Germany)

japaric, 2018-07-26:
This meeting happened. Logs are [here](https://github.com/rust-lang-nursery/embedded-wg/blob/master/minutes/2018-07-23.irc.log). The issue for next meeting is [here](https://github.com/rust-lang-nursery/embedded-wg/issues/130).

closed 2018-07-26 by japaric