rust-embedded/wg #134: Embedded WG Newsletter 9

jamesmunns, 2018-07-29:
Reporting issue for the next newsletter, roughly scheduled for `2018-08-12`.

japaric, 2018-07-31:
Keep an eye on PR rust-lang/rust#52787, which adds RISCV support to rustc.

japaric, 2018-08-03:
> Keep an eye on PR rust-lang/rust#52787, which adds RISCV support to rustc.

The latest nightly has RISCV support. There are a few more details in https://github.com/rust-lang-nursery/embedded-wg/issues/36#issuecomment-410317295.

japaric, 2018-08-09:
The embedded WG has grown! We are now 18 people grouped in 6 teams that are maintaining 23 projects (crates and docs). The new organization of the WG can be found [here](https://github.com/rust-embedded/wg#organization).