rust-embedded/wg #146: Provide cargo-generate templates

japaric, 2018-08-02:
[`cargo-generate`] is going to be the official solution for Cargo templates. The domain WGs are
expected to provide `cargo-generate` templates for applications in their domain.

[`cargo-generate`]: https://github.com/ashleygwilliams/cargo-generate

Let's move the existing templates (`{cortex-m,msp430,riscv}-quickstart`) to the `cargo-generate`
format. If you need any feature to be added to `cargo-generate` to port your template [open an
issue] or leave a comment about your use case in one of the existing ones.

[open an issue]: https://github.com/ashleygwilliams/cargo-generate/issues

Submit feature requests ASAP so there's enough time for them to be implemented. Also let's collect
all the embedded WG feature requests in this issue so we can have the `cargo-generate` team
prioritize them.

# Feature requests

- `cortex-m-quickstart`, custom variables and validation. https://github.com/ashleygwilliams/cargo-generate/issues/17#issuecomment-409987029

cc ~@rust-embedded/cortex-m~ @adamgreig @korken89 @therealprof
cc ~@rust-embedded/msp430~ @cr1901 @pftbest
cc ~@rust-embedded/riscv~ @bradjc @danc86 @dvc94ch

I'm going to tentatively assign this to the RC1 milestone.

korken89, 2018-08-04:
I can fix this for cortex-m

jamesmunns, 2018-08-04:
Has [kickstart] been considered? It seems to have the features we require around custom variables and validation of those variables.

If nothing else, it may be relevant for how to implement these features, if use of `cargo-generate` is really desired.

[kickstart]: https://github.com/Keats/kickstart

Keats, 2018-08-04:
Just saw this thread while looking at the cargo-generate issues and thank you @jamesmunns for mentionning kickstart. I believe kickstart would fit the bill, except the validation of overlapping of memory as validation in kickstart is currently only regex based.

As an aside, seeing `cargo-generate is going to be the official solution for Cargo templates` makes me sad as there was a pre-RFC thread that has been completely ignored and a very basic `cargo-generate` somehow appears from nowhere and becomes the official solution without any public discussion as far as I'm aware.

korken89, 2018-08-05:
I was talking to a colleague of mine, and one thing that came up as missing is the extra help with MCU selection (initial setup), and weeding out all the chip crates.
For example, one would like to do something like this and get the extra help (strawman syntax):
```
$ cargo [my_generator] [mcu] [optional chip crate] [optional extra crates]
```
Where it could be used as, and give the following help:
```
$ cargo generate --mcu cortex-m --chip stm32
Error: Cortex-M is not specific enough.
Maybe you meant: cortex-m0, cortex-m0p, cortex-m3, cortex-m4, cortex-m7
```
Okey, lets improve (I know there is an `stn32` crate, but bear with me):
```
$ cargo generate --mcu cortex-m4 --chip stm32
Error: Too many matches for chip crates:
Maybe you meant: [list of crates]
```

I think a guiding feature like this could really help new users a lot and solve a real pain within embedded, and specifically setting up new project therein. Most of it would come from good suggestions.
Moreover, it would help new users in that they would not need to place the target in the `.cargo/config` file (small thing but easy to miss).
However I am not sure how much of an extension would be needed to the two linked generators here to make something like this work.

Any comments/further ideas/thoughts?

therealprof, 2018-08-07:
@korken89 Sounds great but the syntax is a bit wonky. Why would you specify the core type as `--mcu`? And why would you specify it at all if you have the specific chip?

@ryankurte You mean like https://platformio.org/ for Rust? That sounds like a terrific idea.

korken89, 2018-08-07:
@ryankurte   
Thanks! I was thinking something similar!
The whole issue is probably where all the metadata will come from, so that the system more or less becomes automated in additions and does not become a maintaining nightmare.

@therealprof
Yeah, I agree that the syntax is wonky, it was mostly the overall idea I wanted to put forward.
Any syntax can be bikesheded when/if it comes closer :)

> And why would you specify it at all if you have the specific chip?

Exactly, if this metadata is available the system could say that the core is not needed. The syntax could be a bit more hierarchical:

* If you have board, MCU and Core is not needed.
* If you have MCU, Core is not needed
* Else specify core only

Or something similar.

The core idea is simply to have the system be more automated and easy for beginners and advanced alike. I am however not sure where to propose and push it, as the selection of generator seems to happen now, but it can also be extended later.

japaric, 2018-08-07:
@jamesmunns @Keats I don't have much insight into how `cargo-generate` will get the features we need; that would be up to the Cargo team. At this stage our job is to submit feature requests.

@therealprof @ryankurte @korken89 that sounds interesting but could you leave a comment or open an issue in the cargo-generate issue tracker to make sure the use case doesn't get lost?

@dvc94ch @pftbest will you need more features than what's described in https://github.com/ashleygwilliams/cargo-generate/issues/17#issuecomment-409987029 to port the msp430 / riscv templates to cargo-generate?

japaric, 2018-08-21:
> @dvc94ch @pftbest will you need more features than what's described in ashleygwilliams/cargo-generate#17 (comment) to port the msp430 / riscv templates to cargo-generate?

@rust-embedded/msp430 @rust-embedded/riscv did you get a chance to try this?

---

Triage: implementation is blocked on cargo-generate gaining the features we need.

therealprof, 2018-08-25:
I created an issue in `cargo-generate` for the possibility to use presets: https://github.com/ashleygwilliams/cargo-generate/issues/92

japaric, 2018-09-18:
[cortex-m-quickstart](https://github.com/rust-embedded/cortex-m-quickstart) has moved to the cargo-generate format. As that's what we'll be using the embedded Rust book I'm going to remove this issue from the 2018 milestone to signal that the work required for the 2018 edition is done.

I'm going to leave this open to track the other arches (msp430 and rsicv) moving to / testing cargo-generate.

jamesmunns, 2019-02-04:
@rust-embedded/msp430 
@rust-embedded/riscv 

Are there any plans to update your quickstarts to cargo-generate? Can this issue be closed?

Disasm, 2019-02-14:
@jamesmunns I have this in my roadmap for RISC-V.

Disasm, 2019-07-16:
Updated [riscv-rust-quickstart](https://github.com/riscv-rust/riscv-rust-quickstart).

cr1901, 2020-01-02:
@jamesmunns 
> Are there any plans to update your quickstarts to cargo-generate? Can this issue be closed?

Once `v0.2.0` crates are published, I will update `msp430-quickstart` to use `cargo-generate`; publishing is imminent.

cr1901, 2020-01-12:
@jamesmunns Long overdue, but `msp430-quickstart` has been [updated](https://github.com/japaric/msp430-quickstart/pull/5) to use `cargo-generate`. This issue can be closed I believe.

taurr, 2021-09-19:
I just landed https://github.com/cargo-generate/cargo-generate/pull/445, which should solve the missing issues for this issue.
Please take a look as it'll be part of next `cargo-generate` release

therealprof, 2024-06-11:
There have been a variety of templates available now including https://github.com/rust-embedded/cortex-m-quickstart and https://github.com/rust-embedded/msp430-quickstart under the WG umbrella.

closed 2024-06-11 by therealprof