rust-embedded/wg #148: [RFC] Should we have a blog?

japaric, 2018-08-02:
So far we have been posting our newsletter and other announcements to {users,internals}.rust-lang.org (because it was the easiest thing to start out with) and punted on the question of having a blog or not. As we approach the 2018 edition I think it's time to revisit that question.

Advantages of having a blog:

- Easier to navigate between newsletters and other announcements
- RSS support. People have asked for this before.
- It looks nicer. (subjective)

Disadvantages:

- It takes work to set it up.
- No out of box support for comments on most static site solutions I'm aware of.

Quetions:

1. do we really want to do this?

2. what should we use here? There are a lot of static site generators out there. I use [hugo] for my blog; the [wasm WG](https://rustwasm.github.io/) uses [jekyll]; and there are generators written in Rust as well.

[hugo]: https://gohugo.io/
[jekyll]: https://jekyllrb.com

3. where do we put it? rust-embedded.github.io? blog.rust-embedded.org? rust-embedded.org?

cc ~@rust-embedded/resources~ @andre-richter @jamesmunns @therealprof

ryankurte, 2018-08-02:
I think a blog would be a neat idea! But maybe it'd be better to avoid having comments at all? Seems like the moderation burden could be significant, and that way the blog stuff could remain static.

Another option (that I've been kindof wishing for) could be to pair it with a forum ([Discourse](https://www.discourse.org/) or something) for discussion and questions and everything else in a more friendly / index-able way than IRC. Moderation overhead would probably be the same / worse, but it seems like it could become quite a useful resource, and I could spin up an instance if we wanted to try it out.

jamesmunns, 2018-08-03:
:+1: from me for having a blog. Right now, I use https://www.getgutenberg.io/ for my site at https://jamesmunns.com.

I don't have comments at the moment, but https://utteranc.es/ is something that I have looked at before (comments driven by GitHub issues). If we have a forum or something similar, we can always link out with "Click here to discuss on XYZ" (see below).

I think a discourse could be good, though this may be an attempt at "there are too many places to discuss, lets add another official one!".

I think the current discussion spaces are:

1. internals.rust-lang.org
2. users.rust-lang.org
3. reddit.com/r/rust
4. `#rust-embedded` IRC
5. `@rustembedded` twitter

None of the above are embedded focused, outside of the IRC channel and twitter, which is perhaps a little ephemeral for tracking "tribal knowledge"

I'd also like to suggest a wiki, for collecting knowledge. The book should cover all of our knowledge around embedded rust from a language perspective, however I think are lots of chip, family, tooling, sensor, device, etc domain knowledge that doesn't make sense to cover in the book, but would be valuable to the community.

I'm fine using the built in github wiki, perhaps in the `rust-embedded/wg` repo.

So I guess to summarize, I would suggest that our knowledge/communication strategy could be as follows:

### A forum

Set up at `https://forum.rust-embedded.org`. Perhaps powered by Discourse.

Meant for topics slightly larger than a single blog post or IRC communication, but slightly too ephemeral for a wiki post.

Content driven by the community

### A Chat Room

`#rust-embedded @ irc.mozilla.org`. Maintained as-is. Meant for direct questions, random chat, technical questions.

### A blog

A static site, set up at `https://blog.rust-embedded.org`. Static site generator TBD, likely hosted on github pages, probably at `rust-embedded/blog` github space. Covers posts such as the newsletter, calls to action, and technical posts by guest contributors (they can send in posts by filing a PR to the blog project, reviewed by the `resources` team.

Content driven by the resources team and the community

### A Wiki

A wiki page driven by github, probably residing in the `rust-embedded/wg` repository, which would locate it at `https://github.com/rust-embedded/wg/wiki`. We could add an alias at `https://wiki.rust-embedded.org`.

Content created and curated by the community. No guarantees on correctness.

### Our Bookshelf

These are static `mdbook` projects that will be hosted at `https://*.rust-embedded.org`. Currently we (will) have:

* The Discovery Book (learn embedded through rust)
* The Rust Embedded Book (learn embedded rust)
* The Embedonomicon (learn the low level details of embedded rust)

This content is written by anyone, and curated for completeness and correctness by the resources team. We should guarantee correctness, at least with a similar level of guarantee that "The Rust Programming Language" book does.

### Two Websites

We will have the following two pages accessible to us:

* `https://rust-embedded.org`
* The WIP new rust site, likely `https://rust-lang.org/what/embedded`

#### `https://rust-lang.org/what/embedded`

I see this as a promotional website. It should have examples and content, however it serves as a landing point for people not currently involved with rust, or embedded rust. We can contribute content here, but we do not "own" this page.

#### `https://rust-embedded.org`

I see this as the technical landing page. If someone asks "how do I do embedded rust", this becomes a singular place we can link them to. From here, we can link to all of the spaced I enumerated above, as well our official repositories, and content such as the "Awesome Embedded Rust" list.

This should be a static site, generated similarly with tools similar to above. Would probably live at `rust-embedded/website`. Hosted at `https://rust-embedded.org`.

Content here would be written and maintained by members of the whole embedded wg, and maintained by the resources team.

jamesmunns, 2018-08-03:
Sorry, that got a little more in detail than I planned. Let me know if you would like me to open a new RFC, or if you are okay with me hijacking this RFC as our whole content plan.

kjetilkjeka, 2018-08-03:
@jamesmunns Has gitter been considered for the chat room? If so, what was the reason IRC was chosen?

jamesmunns, 2018-08-03:
@kjetilkjeka IRC is chosen in my above proposal as that is what our most active chat is currently. For most of the other items, I am proposing new things that we have not had before.

I'm not 100% against changing our chat situation (to gitter, discord, rocket, etc.), but I have a feeling that there will be strong opinions around that, and would rather have that in a separate RFC.

levex, 2018-08-03:
I like the idea of a blog, it's an easily approachable medium that can help spread Rust's support for embedded. It would also help give the newsletter a more prominent place which is a welcome improvement.

Regarding the chat room, I _personally_ find Gitter too distracting (and RAM consuming!). IRC with a publicly available logging service is low-maintenance, and highly-effective IMO. On the other hand, IRC is less attractive to contributors I think, could we explore a bridge maybe?

japaric, 2018-08-03:
@kjetilkjeka 

We also have an "official" space in Rust discord server (rust-lang): #wg-embedded, which was created at the time the Rust team decided to experiment with alternatives for IRC, though there's less activity there than there is on IRC. I believe is planned to set up bridges between the IRC and discord servers but I don't know the timeline or the details.

---

I have always wanted to have a FAQ section somewhere. That way I can point all the people asking for "Rust on Arduino?", "Rust on ESP8266?", etc. there. I was thinking that it would make sense to put that in the forum as that's where I expect people will go for asking questions and looking for answers, but perhaps the FAQ should go in the wiki?

jamesmunns, 2018-08-03:
Unless anyone feels very strongly about chat, I'd like to avoid discussing it in this thread. We can always open up another RFC to discuss :)

@japaric I think the FAQ in a wiki would make sense

therealprof, 2018-08-03:
"Rust on Arduino?" makes me chuckle: Arduino what? The IDE? The ecosystem? "The" hardware (which one)? Funny enough Rust should even usable with on a number of Arduino boards, e.g. the 101 using Intel Atom and the ATSAMD ones...

kjetilkjeka, 2018-08-03:
@jamesmunns I do not have very strong feelings about the chat but I think it would still be nice to have an "Embedded Rust Communication Plan" issue where things like this could be discussed without the fear of going off-topic.

---

I think being able to navigate between newsletter more easily is a huge advantage. There are currently also a lot of semi-unrelated issues on this repo and moving the newsletter issues/PRs to a blog repo would be an advantage as well. As long as there are people to maintain it and they feel good about the blog medium it sounds like an excellent idea.

jamesmunns, 2018-08-04:
Hey, I talked with @japaric, and he gave me the :+1: to change this into an RFC covering our larger community plan. Here is his original RFC content, for historical reasons:

---

So far we have been posting our newsletter and other announcements to {users,internals}.rust-lang.org (because it was the easiest thing to start out with) and punted on the question of having a blog or not. As we approach the 2018 edition I think it's time to revisit that question.

Advantages of having a blog:

- Easier to navigate between newsletters and other announcements
- RSS support. People have asked for this before.
- It looks nicer. (subjective)

Disadvantages:

- It takes work to set it up.
- No out of box support for comments on most static site solutions I'm aware of.

Quetions:

1. do we really want to do this?

2. what should we use here? There are a lot of static site generators out there. I use [hugo] for my blog; the [wasm WG](https://rustwasm.github.io/) uses [jekyll]; and there are generators written in Rust as well.

[hugo]: https://gohugo.io/
[jekyll]: https://jekyllrb.com

3. where do we put it? rust-embedded.github.io? blog.rust-embedded.org? rust-embedded.org?

cc ~@rust-embedded/resources~ (at)andre-richter (at)jamesmunns (at)therealprof

jamesmunns, 2018-08-04:
Oh, and if we have a forum, I would suggest no comments on the blog to keep it simple, and instead link to a forum post discussing that article.

steveklabnik, 2018-08-04:
FWIW, I always thought that we were going to end up adding the various WG books to the bookshelf on doc.rust-lang.org; if you all would like that, we can do it, but if you don't, we don't have to.

jamesmunns, 2018-08-04:
Good call @steveklabnik. That was the original intent.

Discovery and the Embedonomicon are probably ready, or close to ready for inclusion. Any objections @japaric?

IMO, the repos would still live in our organization, but this would remove the need for DNS entries, as well as maybe even setting up GH pages/CI.

steveklabnik, 2018-08-04:
Yup! All you'd need to do is have a repo with and mdbook book in it; we submodule it in, and i take care of the rest. you don't even have to update it; I update all the books for each release.

jamesmunns, 2018-08-06:
Notes from the meeting:

```
Blog, generally +1, maybe consider setting up a "planet" for multiple blogs in this space
+1 to forum - perhaps discourse, might be able to get free hosted instance
-3-4 to wiki
+3-4 to no comments on a blog
japaric will ask the Rust team about the possibility of getting a free discourse forum
```

ryankurte, 2018-08-08:
I think it's also important that we have a place to compose [institutional knowledge](https://github.com/rust-embedded/wg/pull/136#issuecomment-411173888) and processes for us, it seems like a lot of things occur in issues/PRs that would be valuable to surface.

I think a wiki would do that reasonably well to compose everything, it's searchable and linking makes sense, and would allow us to slowly build it up as questions arise without creating a huge amount of noise on the wg repo / wherever else we would put it.

If we're dead set against a wiki we could maybe add another set of documentation somewhere for this?

nerdyvaishali, 2018-08-10:
Sorry, I'm probably late for the party but just wanted to add one more point. Loved the idea of blog and other tooling related things. Regarding social media platforms, I was wondering if there is an interest in using Instagram? I believe that along with twitter, Instagram can be used for embedded Rust demos [in the form of videos/pictures/GIFs, Outreaching about embedded projects [again content can be created accordingly] and last but not least embedded Rust Memes. :)

For the things mentioned above, Twitter can be used as well. But i'm not sure if we want to use different sort of social media platforms for different type of content or we want to focus on using one platform to outreach about all sort of embedded Rust related content.

japaric, 2018-08-20:
@nerdyvaishali Interesting proposal. I'm not an instagram user myself (I post
pics and videos of my embedded stuff on twitter) but my opinion is that we
should have a presence on social sites where there's a "significant" presence of
(potential) (embedded) Rust developers, if we have the people to run the
account. The Arduino instagram account has 180K followers; in contrast the Rust
twitter account only has 30K followers so it seems worthwhile.

Before we venture there though I would like to see a plan that details how we
are going to produce content: any existing embedded Rust user that's also
posting embedded stuff on instagram and that we can repost content from? Can we
get existing e.g. twitter users to also post on instagram? Can we repost twitter
content? etc.

thenewwazoo, 2018-08-28:
Per today's embedded wg discussion, we identified a need to a place to have discussions about things that aren't well-suited to github issues, or that are broader than a specific crate/repo's mandate.

Some solutions were proposed, along with some pending responses/actions:
- An embedded Discourse instance (as mentioned in the OP), pending a response to @japaric from someone on the Rust team
- A Category on users.rust-lang.org, pending an inquiry about adding a category there

japaric, 2018-09-04:
I had milestoned this because originally this was about only the blog but since that is done I'm going to remove it from the milestone.

---

@nerdyvaishali yes, opening a separate issue with a draft of social media guidelines or anything similar to start the discussion would be the next step here.

jamesmunns, 2019-02-04:
I think we have reached the majority of the goals set out here. I would like to nominate this to be closed.

@nerdyvaishali if you are still interested in opening additional social media accounts, please feel free to get in touch.

jamesmunns, 2019-02-14:
I am closing this issue, please feel free to open another issue if you would like this discussed further.

closed 2019-02-14 by jamesmunns