rust-embedded/wg #149: Embedded WG meeting, August 6

japaric, 2018-08-06:
[6-7 PM CEST][0] on #rust-embedded on irc.mozilla.org

[0]: https://everytimezone.com/#2018-8-06,240,b8jj

Backlog.

- "Could we briefly mention talking to chip companies about Rust?" -- @thejpster
- "Managing operator status on #rust-embedded (currently it's just @japaric)" -- @thejpster?

Agenda.

- r?
  - org/teams RFC https://github.com/rust-lang-nursery/embedded-wg/pull/136
  - MSP430 team https://github.com/rust-lang-nursery/embedded-wg/pull/137
  - RISCV team https://github.com/rust-lang-nursery/embedded-wg/pull/138
  - resources team https://github.com/rust-lang-nursery/embedded-wg/pull/140
  - tools team https://github.com/rust-lang-nursery/embedded-wg/pull/141
  - HAL team https://github.com/rust-lang-nursery/embedded-wg/pull/142
  - Cortex-M team https://github.com/rust-lang-nursery/embedded-wg/pull/143

- New meeting time to accommodate new members.

- Community plan https://github.com/rust-lang-nursery/embedded-wg/issues/148

- [RC1 milestone](https://github.com/rust-lang-nursery/embedded-wg/milestone/3)

- [Issues nominated for discussion, if any](https://github.com/rust-lang-nursery/embedded-wg/labels/nominated)

- [your topic goes here]

This meeting is open for anyone to attend. If you'd like to bring up any issue / topic related to embedded Rust leave a comment in this issue so we can add it to the agenda.

TeXitoi, 2018-08-06:
I had some friction begining with rust on the blue pill, maybe that's relevant:
 * stm32f103xx-hal is not on crates.io. As it must be one of the most used hal (I see a lot of examples using it), having a version on crates.io seems important. There is already an issue: https://github.com/japaric/stm32f103xx-hal/issues/52
 * stm32f103xx-hal doesn't support I2C out of the box. There is a PR waiting to be merged: https://github.com/japaric/stm32f103xx-hal/pull/50
 * cortex-m-rtfm doesn't work on nightly since a long time. There is several PR and issues about that: https://github.com/japaric/rtfm-syntax/pull/9 https://github.com/japaric/rtfm-syntax/pull/8 https://github.com/japaric/rtfm-syntax/issues/7 https://github.com/japaric/cortex-m-rtfm/issues/82 https://github.com/japaric/cortex-m-rtfm/pull/80
 * cortex-m-rtfm is not compatible with cortex-m-rt 0.5. I managed to make it working with 0.4, but the handler plumbing seems different with 0.5.

I think fixing all of that for the 2018 edition would be great. I can help with my competences (experienced rustacean, newbie in embedded dev).

jamesmunns, 2018-08-06:
I'll be there, but will be a few minutes late. I'd also like to discuss #148 regarding the community plan

japaric, 2018-08-06:
This meeting happened! Minutes are up [here](https://github.com/rust-lang-nursery/embedded-wg/blob/master/minutes/2018-08-06.md); logs are [here](https://github.com/rust-lang-nursery/embedded-wg/blob/master/minutes/2018-08-06.irc.log). The agenda issue for the next meeting is #150.

@TeXitoi I author / maintain those crates and they are not a blocker for the 2018 edition so they are not a priority for me at the moment. I will have some time for those projects once the teams are formed and they take some load (RC1 milestone) off from me.

closed 2018-08-06 by japaric